Ultimaker/Cura #1034: 2.3: Wall Speed Locked

CCS86, 2016-10-08:
Hi guys, any idea what's going on here? I'm locked out from changing wall speed.

[image]

fieldOfView, 2016-10-08:
I think that in the profile you have selected, a value is set for both the inner wall speed and the outer wall speed. Together these override the wall speed setting. So entering a number there would not have any effect

CCS86, 2016-10-08:
You can see here though, that only the outer wall speed child setting is enabled, and the values don't match. A hidden child setting shouldn't drive, right?

Tried enabling it. Something still not right:

[image]

fieldOfView, 2016-10-08:
Could you click on the "i" buttons for both inner and outer wall speed?

CCS86, 2016-10-08:
It doesn't affect the wall speed lockout

CCS86, 2016-10-08:
I've tried exporting the current profile, then modifying the values for those speeds, renaming, then importing; but I get an "invalid argument" error.

CCS86, 2016-10-08:
I think it was something about the existing profile.

I scrapped the whole Cura appdata folder, and started everything from scratch. Now the field is unlocked.

fieldOfView, 2016-10-08:
In 2.4 we're going to have to be more clear about why a value is disabled. Too late for 2.3 though.

CCS86, 2017-01-09:
I'm still struggling with this in 2.4:

[image]

Ghostkeeper, 2017-01-22:
The setting is disabled because it has no effect on the print any more. Both `Outer Wall Speed` and `Inner Wall Speed` have customised settings (as indicated by the rotating arrow). `Wall Speed` does nothing but to affect the default of `Outer Wall Speed` and `Inner Wall Speed`. So now that those settings are different from the default, the setting does nothing any more.

I believe we give a warning message on the bottom of the screen if there are hidden settings that are different from the default.

> A hidden child setting shouldn't drive, right?

No, hiding a setting does nothing to affect the slice. A hidden child setting still drives.

I can't find the Jira ticket related to this issue though. Did we make one? I don't think it is really clear yet why a setting is disabled if all its child settings are hidden.

ianpaschal, 2018-02-21:
No more work will be done on versions 2.x. If this is still occurring in versions 3.2 or newer, we can re-open this issue.

Ghostkeeper, 2018-03-13:
This is still a problem. It's a bit of a user experience problem that the setting is greyed out and nothing explains why.

Vandrasc, 2018-04-09:
Devs see CURA-5083

Ghostkeeper, 2018-05-15:
Our project manager decided to remove this from our planning. It's a problem but not important enough to spend time on now until we get to make some time for rethinking this part of the UI.

closed 2018-11-26 by Appesteijn