Ultimaker/Cura #1036: No way to remove the heating commands (M190, M104, and M109) from the beginning of generated code files

Pointedstick, 2016-10-10:
I edited my machine definition so that it has the following start.gcode section:

```
        "machine_start_gcode": {
            "default": "G21 ;metric values\nG90 ;absolute positioning\nM82 ;set extruder to absolute mode\nM107 ;start with the fan off\nM104 S210 ; set the extruder temperature to 210 and continue on\nM140 S60 ; set the bed temperature to 60 and continue on\nG28 ;home all axes\nG92 E0 ;zero the extruded length\nG1 F200 E3 ;extrude 3mm of feed stock\nG92 E0 ;zero the extruded length again\nG1 X5 F5000 ; move nozzle directly over build platform to prevent ooze\nM109 S210 ; wait for extruder to reach its temperature\nM116 ; wait for heated elements to reach their temperatures before starting\nG1 F9000"
        },
```

However, Cura appears to always prepend M190, M104, and M109 commands BEFORE the custom start.gcode. Here's what the happens:

```
;FLAVOR:RepRap
;TIME:1153
M190 S60.000000                <- These three are automatically prepended
M104 S210.000000
M109 S210.000000
G21 ;metric values             <- start of my custom start.gcode routine
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
M104 S210 ; set the extruder temperature to 210 and continue on
M140 S60 ; set the bed temperature to 60 and continue on
G28 ;home all axes
G92 E0 ;zero the extruded length
G1 F200 E3 ;extrude 3mm of feed stock
G92 E0 ;zero the extruded length again
G1 X5 F5000 ; move nozzle directly over build platform to prevent ooze
M109 S210 ; wait for extruder to reach its temperature
M116 ; wait for heated elements to reach their temperatures before starting
G1 F9000
;Generated with Cura_SteamEngine 2.1.3
```

I can't find a way to get it to stop doing this. It's problematic for my particular setup because I want to home the extruder and move the nozzle directly over the build plate _before_ heating up, to prevent ooze. If it first heats up the nozzle and bed before executing custom start.gcode commands, then there's already a big oozy mess coming out of the nozzle that wastes filament, requires even more wastage with extrusion priming, and then some kind of wipe operation to get rid of the mess so it doesn't interfere with the print.

Ghostkeeper, 2016-10-11:
They are put there to allow the start g-code to extrude something too, at the correct temperature.

Pointedstick, 2016-10-11:
Right, but as a technically knowledgeable user, I'd like to remove them and control 100% of the start gcode commands myself. I've even seen references to Cura automatically removing these commands if it sees them in your custom start.gcode sequence, and that isn't happening, so perhaps there's a bug.

fieldOfView, 2016-10-11:
If you put the temperatures into your gcode as variables instead of hardcoded values, Cura will not add them a second time.
```
M140 S{material_bed_temperature} ; set the bed temperature and continue on
M104 S{material_print_temperature}; set the extruder temperature and continue on
```
Note that it is generally a better idea to heat up the bed before the hotend; heating up the bed takes considerably longer than heating up the hotend, and you would keep the hotend hot while the bed heats up, without extruding material. This would mean the material in the hotend would start to degrade.

Pointedstick, 2016-10-11:
Oh, cool!

Pointedstick, 2016-10-12:
That worked!

closed 2016-10-12 by Pointedstick

alexwhittemore, 2017-11-09:
This appears to be a bug again in 3.0.4 - I'm trying to write a start sequence for my i3 Mk2 that deliberately waits to heat the bed until the height sensor is out of the way so as to avoid thermal error on that probe. This is getting frustrated by Cura auto-inserting `M190 S55` before my custom start code, despite said code containing `M190 S[first_layer_bed_temperature]` where 55 above is whatever my actual slicer setting is for first layer temp.

roycruse, 2018-01-02:
I have tried to replace the default inserted M190 M104 M109 codes with ones in my srtart gcode settings against the machine.

however - if i allow cura to insert these its self it picks up the correct on screen temperatures of my sliced 3d file of 55 bed and 245 hot end.

but when i insert the M104 or M109  commands myself into the start gcode sections using parameter "machine_print_temperature" i do not get the on screen selected temperature but i get 210 deg instead    is there a different parameter name i should be using for the print temperature thats selected at time of slicing my model. not some old default temp thats buried away in a profile xml file which is by the way the only place i can find a mention of 210 degrees - but i have no idea why its that temp thats making it into the produced gcode.

Any help greatly appreciated as this is driving me mad

fieldOfView, 2018-01-02:
That’s a bug in Cura 3.1. It worked correctly in Cura 3.0 and will wirk correctly in Cura 3.2 again.

Prunar, 2018-03-31:
Running into the same issue. Start Gcode:
```
G21 ;metric values
M140 S{print_bed_temperature} ; begin heating bed
M109 S{print_temperature} ; begin heating nozzle
G90 ;absolute positioning
M82 ;set extruder to absolute mode
```

Results in:
```
;Generated with Cura_SteamEngine 3.2.1
M190 S60
M104 S200
M109 S200
M82 ;absolute extrusion mode
G21 ;metric values
M140 S60 ; begin heating bed
M109 S200 ; begin heating nozzle
G90 ;absolute positioning
```

fieldOfView, 2018-03-31:
Read https://github.com/Ultimaker/Cura/issues/1036#comment-253020560

You are using the wrong replacement patterns. material_print_temperature vs print_temperature

Prunar, 2018-03-31:
@fieldOfView Doh, sorry about that. Now it works. Maybe deprecate those old variables (if they are old)?

ju-mpe-r, 2019-02-18:
Hi, @fieldOfView!
Can you please give me some advice about how exclude M107 g-code command from Cura generated .gcode file?

fieldOfView, 2019-02-19:
@ju-mpe-r, please elaborate. CuraEngine does not insert M107 before the start gcode (which is the topic of this issue).

If you don't want any M107 commands, you can use a search/replace post processing script.

ngrigoriev, 2024-06-27:
It seems to be broken again in 5.7.2 (at least). My custom g-code has this:

```
...
M104 S{material_print_temperature_layer_0}
M109 S{material_print_temperature_layer_0}
...
```

Yet, the generated resulting g-code ends up like this:

```
...
;TARGET_MACHINE.NAME:Creality Ender-3 Max
;Generated with Cura_SteamEngine 5.7.2
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
; Ender 3 Max Custom Start G-code
M117 Getting the bed up to temp!
M140 S65 ; Set Heat Bed temperature
M190 S65 ; Wait for Heat Bed temperature
M117 Pre-heating the extruder!
M104 S160; start warming extruder to 160
G28 ; Home all axes
M117 Auto bed-level GO!
G29 ; Auto bed-level (BL-Touch)
G92 E0 ; Reset Extruder
M117 Getting the extruder up to temp!
M104 S200
M109 S200
G1 Z2.0 F3000 ; move z up little to prevent scratching of surface
...

```

M104/109 commands are added regardless of what is in the custom g-code.

fieldOfView, 2024-06-28:
Can't help without a project file (File -> Save project as...)

ngrigoriev, 2024-06-28:
[plate.zip](https://github.com/user-attachments/files/16030312/plate.zip)

Thanks!

fieldOfView, 2024-06-28:
Upon cursory look it does not seem like you are doing anything wrong in your project. Unfortunately I do not have time for more than a cursory look.

Subcode, 2024-06-29:
> It seems to be broken again in 5.7.2 (at least). My custom g-code has this:
>
> ```
> ...
> M104 S{material_print_temperature_layer_0}
> M109 S{material_print_temperature_layer_0}
> ...
> ```
>
> Yet, the generated [...]

Same issue here, very annoying.

andyn, 2024-07-07:
5.7.2 is broken, but 5.7.1 does not output those extra commands. Don't have time to bisect the exact commit, sadly.

PingRen32, 2024-07-24:
I got a starter code 

```
; Ender 3 Custom Start G-code
M140 S{material_bed_temperature_layer_0} ;Start heating bed
M190 S{material_bed_temperature_layer_0} ;Wait for bed to reach temp before proceeding
M900 K0.11; PLA PRO Linear Advance
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 ; Bed leveling
M104 S{material_print_temperature_layer_0} ;Start heating extruder
M109 S{material_print_temperature_layer_0} ;Wait for extruder to reach temp before 
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
```

which should skip the `Generated with Cura_SteamEngine` (which did back from 5.3) so something must have been broken.