Ultimaker/Cura #1040: Final layer of print completely covered

JBatCCA, 2016-10-12:
I was told by tech support to come here with my issue.....

"your issue with the final layer is definitely on the software side as well, and could possibly be within the shell settings for Cura. I’d recommend opening a support ticket within the Cura Github, and they should be able to advise you upon the specific setting:"

Just as it sounds, I am printing a Lithophane from a jpeg and the final layer is completely covering my print with plastic in one last layer. I have had to put in a "pause at layer" g-code modifier, but can't figure out why it is doing this?
I attached my STL file as well as my g-code. Hopefully someone can tell me what I'm doing wrong.

Thank you.
[UM2E+_buck 5.zip](https://github.com/Ultimaker/Cura/files/525845/UM2E._buck.5.zip)

fieldOfView, 2016-10-13:
I don't think you are doing anything wrong. The image importer has not had very extensive testing since it was rewritten for Cura 2.x. I do see a lot of issues with the structure of the mesh that Cura creates when loading an image. I'll raise an issue about this on our internal tracker.

Having said that I cannot reproduce the issue when slicing your stl with the 2.3 beta. That could mean the issue has been fixed in the CuraEngine since 2.1. It won't be long now before the final version of 2.3 is released (you might get an idea of when it is released if you vists ultimaker.com, but I cannot comment on that).

JBatCCA, 2016-10-14:
Hi Aldo,

    Thanks for the response. I look forward to trying it with the new slicer than. (2.3)

Sincerely,

Jason Brown
Cad/Cam Advantage, Inc.
651-245-6471

The information in this email is confidential and may be legally privileged. It is intended solely for the addressee. Access or any use by any other person to this
Internet email is not authorized and may be unlawful. If you are not the intended recipient, please delete or destroy this email. If you do not wish to receive
future emails from the sender, please reply directly to this email requesting you be removed from any mailing list.

From: Aldo Hoeben 
Sent: Thursday, October 13, 2016 1:22 AM
To: Ultimaker/Cura 
Cc: JBatCCA ; Author 
Subject: Re: [Ultimaker/Cura] Final layer of print completely covered (#1040)

I don't think you are doing anything wrong. The image importer has not had very extensive testing since it was rewritten for Cura 2.x. I do see a lot of issues with the structure of the mesh that Cura creates when loading an image. I'll raise an issue about this on our internal tracker.

Having said that I cannot reproduce the issue when slicing your stl with the 2.3 beta. That could mean the issue has been fixed in the CuraEngine since 2.1. It won't be long now before the final version of 2.3 is released (you might get an idea of when it is released if you vists ultimaker.com, but I cannot comment on that).

BagelOrb, 2017-07-23:
outdated

closed 2017-07-23 by BagelOrb