Ultimaker/Cura #1041: Preparing to slice...

joostmarkerink, 2016-10-13:
Sometimes when I am changing Print Job settings it stays at status "Preparing to slice". It seems that this happens when I change settings while it is still slicing. When it hangs in "Preparing to slice" the CPU is not busy. So it looks like it is waiting for current changes to be made, while there aren't any. Something with update changes queue maybe?
(running on Mac OSX Sierra).
I restart Cura to be able to generate the G-Code again.

fieldOfView, 2016-10-13:
Which version of Cura is this? 2.1.3, or the 2.3 beta?

red-green, 2016-10-13:
I'm having this error as well on Cura 2.1.3, downloaded from the official website, running on Mac OS 10.11. I was able to slice one model successfully, but now I can't seem to slice a second model (same settings as the first but with supports enabled).

fieldOfView, 2016-10-13:
Try the Cura 2.3 beta, or wait a couple of days and try the Cura 2.3 final.

red-green, 2016-10-13:
Update: On mac, manually killing CuraEngine and restarting the whole app seems to fix it for me.

joostmarkerink, 2016-10-13:
Ah yes sorry. I forgot to mention. the version is 2.1.3

nallath, 2017-07-27:
This should no longer be an issue. Haven't seen this in a long time.

closed 2017-07-27 by nallath