Ultimaker/Cura #1045: 2.3 complete mix up of version naming

claus1009, 2016-10-13:
I downloaded a file called:  Cura-2.3.0-BETA-win64.exe

But after installation "help-> about2 tells me that I'm using a version 2.1.99
Also during the whole installation process the GUIs report about a verion 2.1.

Which version am I using now

This is really inconsistent and confusing?

fieldOfView, 2016-10-14:
It's a known issue caused by an internal build accidentally being uploaded to the public server. The issue will be solved in a couple of days.

ChrisRiddell, 2016-10-21:
@fieldOfView This issue is still half there in the stable release, in windows 10.

[image]

CCS86, 2016-10-21:
On a slightly related question, is the new "stable" 2.3.0 an actual update to the "beta" 2.3.0?

nallath, 2016-10-21:
@CCS86 A bit of both. There is one notable new feature in the stable release that was not in the beta; the network printing.

closed 2017-07-27 by nallath