Ultimaker/Cura #1047: 2.3 Beta Mac repeatable every time Crash on startup.

DavidPhillipOster, 2016-10-15:
where: /Volumes/Cura-2.1.99-master.20160930.signed-Darwin
copied Cura.app to my /Applications/3D/Cura2.3β directory.

Every time I try to run it, I get an immedaiate crash. Apple's crash report starts::

Process:               cura [7058]
Path:                  /Applications/3rd Party Apps/*/Cura.app/Contents/MacOS/cura
Identifier:            nl.ultimaker.cura
Version:               2.1 (2.1.99-master.20160930.signed)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           cura [7058]
User ID:               501

Date/Time:             2016-10-15 07:27:47.660 -0700
OS Version:            Mac OS X 10.11.6 (15G1004)
Report Version:        11
Anonymous UUID:        85F3817A-DD07-605E-C193-9D89F225C480

Time Awake Since Boot: 300000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff9d273f06 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff929a04ec pthread_kill + 90
2   libsystem_c.dylib             	0x00007fff990296df abort + 129
3   nl.ultimaker.cura             	0x0000000106f1c4be Py_FatalError + 462
4   nl.ultimaker.cura             	0x0000000106f1c298 _Py_InitializeEx_Private + 1336
5   nl.ultimaker.cura             	0x0000000106e23917 main + 967
6   nl.ultimaker.cura             	0x0000000106e23544 start + 52

fieldOfView, 2016-10-15:
Could it be that the β character in the pathname causes the crash? Have you tried it without that, or by writing out "beta" instead of "β"?

DavidPhillipOster, 2016-10-15:
Thank, you, yes  - it's the non-ASCII character in the path that's causing the trouble. I can live with the restriction to only ASCII characters in the path, and appname. Feel free to close this issue.

closed 2016-10-15 by fieldOfView