Ultimaker/Cura #1048: Can't change material information?

peteruithoven, 2016-10-16:
Cura 2.3.0 has materials preferences with information like filament cost. Is there any way to change this? All this information is shown in input fields which, indicate that they can be changed except for the fact that they are all disabled. 
I'm hoping to get a costs estimation of prints.

awhiemstra, 2016-10-17:
Not yet. They are there because we wanted to add that ability, but unfortunately that did not make it for 2.3.0 and no one thought about removing the input fields.

Ghostkeeper, 2016-10-17:
PrintCostCalculator doesn't work any more :cry:

peteruithoven, 2016-10-17:
Ah alright, good to know. I fully understand that it didn't have the highest priority. Is there an issue for this already or could I keep tracking this issue?

Ghostkeeper, 2016-10-17:
We have an internal issue for this (CURA-806, for our developers' reference). It is currently planned for version 2.6, which is like halfway towards the heat death of the universe. These things are fluid though. Perhaps a member of our CCB could bring that up in the board, soon?

fieldOfView, 2017-04-27:
This one can be closed

closed 2017-05-01 by nallath