Ultimaker/Cura #1050: info cura 2.11.99 lost function from 15.04.6

ininverno, 2016-10-17:
q?
I pass from cura 15 to cura 21
much better but I miss some very good from the previous version::
1-  I cannot see the origin in the platform any suggestion to see it like previews edition?
2- I cannot save the model, each time I need reload everything, any suggestion for that?

Other Q:
I see on the material have the cost field, but cannot edit adding new one, and I cannot see the cost per print including support.. any suggestion?

Thanks in advance for information.

Ghostkeeper, 2016-10-17:
1. Good suggestion. I'll propose it, see what others think.
2. Try going into File -> Save All -> Save to File, then save as a model file, such as .3mf. It will save the current scene as a .3mf file. The implementation in 2.1 is slightly buggy, and it will lose all per-object settings and such, but we're improving that for 2.3 and further for 2.4.
3. (Other Q) Currently being discussed here: https://github.com/Ultimaker/Cura/issues/1048

nickthetait, 2016-10-18:
Actually, we (Aleph Objects) have already done #1. Expect a pull request in the next few days or so...

ininverno, 2016-10-25:
thanks , amazing.. waitfor new revision... lets try more and find something to made it better

fieldOfView, 2016-11-09:
This has been implemented. Issue can be closed.

closed 2016-11-09 by nallath