Ultimaker/Cura #1059: Add .g to allowed gcode file extensions

themaninthesuitcase, 2016-10-20:
Enhancement Request:

With the release of 2.3 and the ability to use custom printers I would like the abilty to export gcode with a .g file extension.

Reasoning behind this on printrbot printers there are no on printer controls.  In order to print from the SD you need to have a file name auto0.g which will be printed on booting the printer.  Whilst I can rename the file after export, allowing for .g skips a step and removes the risk I'll forget to do this and re-print my last print.

I am willing to do the leg work on this one if someone would point me at the file(s) I need to look at.  So far I've not located the controller code for the filenaming.  

Application Version: 2.3
Platform: OS X 10.11

fieldOfView, 2016-10-20:
The extension is defined here:
https://github.com/Ultimaker/Cura/blob/master/plugins/GCodeWriter/__init__.py
I *think* it should be possible to just duplicate the contents of the output there. Make sure you add a comma, like so:
```
            "output": [{
                "extension": "gcode",
                "description": catalog.i18nc("@item:inlistbox", "GCode File"),
                "mime_type": "text/x-gcode",
                "mode": GCodeWriter.GCodeWriter.OutputMode.TextMode
            },{
                "extension": "g",
                "description": catalog.i18nc("@item:inlistbox", "GCode File for printrbot"),
                "mime_type": "text/x-gcode",
                "mode": GCodeWriter.GCodeWriter.OutputMode.TextMode
            }]
```

themaninthesuitcase, 2016-10-20:
Thanks.  I'll try find some time over the weekend to make the change and test it.

Ghostkeeper, 2016-10-21:
It's also done that way in the STL Writer, though that are actually two different file formats: https://github.com/Ultimaker/Uranium/blob/master/plugins/FileHandlers/STLWriter/__init__.py

Ghostkeeper, 2016-10-24:
Ultimaker just rejected this feature request. If it's going to be implemented, it'll have to be via the community.

themaninthesuitcase, 2016-12-05:
Spot of assistance if I may.  I thought I'd done all I needed as @fieldOfView has posted in a few issues but:
```
$ ./cura_app.py 
Traceback (most recent call last):
  File "./cura_app.py", line 10, in <module>
    from UM.Platform import Platform
ImportError: No module named 'UM'
```
My python path is set and accurate:
`PYTHONPATH=:~/Source/3rdParty/Ultimaker/Uranium:~/Source/3rdParty/Ultimaker/libArcus/python`

I also believe I have the required dependancies setup (i.e. engine is in a bin folder and I've installed PYQT5 etc).

Using OS X 10.11.6 and most deps came from homebrew.

fieldOfView, 2016-12-06:
Just so we know it is using the "right" python version, does this also happen if you start it with ```python3 cura_app.py```?

themaninthesuitcase, 2016-12-06:
```
$ python3 cura_app.py 
Traceback (most recent call last):
  File "cura_app.py", line 10, in <module>
    from UM.Platform import Platform
ImportError: No module named 'UM'
```

The shebang is `#!/usr/bin/env python3` though so the 2 statements *should* be equivalent, but you're right to question it and I had tried both. Seems that was the last one I tried.

Ghostkeeper, 2016-12-06:
You may need to expand the home directory manually? Python doesn't do this consistently, usually only when it goes via the operating system.

themaninthesuitcase, 2016-12-06:
🛎🛎🛎 I had to use a full path not `~/` . Well and install numpy as I forgot that dep. 

BUT:
```
$ python3 ./cura_app.py 
cura.CrashHandler.show [29]: An uncaught exception has occurred!
cura.CrashHandler.show [32]: Traceback (most recent call last):
cura.CrashHandler.show [32]:   File "./cura_app.py", line 49, in <module>
cura.CrashHandler.show [32]:     import cura.CuraApplication
cura.CrashHandler.show [32]:   File "/Users/themaninthesuitcase/Source/3rdParty/Ultimaker/cura/cura/CuraApplication.py", line 4, in <module>
cura.CrashHandler.show [32]:     from UM.Qt.QtApplication import QtApplication
cura.CrashHandler.show [32]:   File "/Users/themaninthesuitcase/Source/3rdParty/Ultimaker/Uranium/UM/Qt/QtApplication.py", line 17, in <module>
cura.CrashHandler.show [32]:     from UM.Qt.Bindings.Bindings import Bindings
cura.CrashHandler.show [32]:   File "/Users/themaninthesuitcase/Source/3rdParty/Ultimaker/Uranium/UM/Qt/Bindings/Bindings.py", line 13, in <module>
cura.CrashHandler.show [32]:     from . import BackendProxy
cura.CrashHandler.show [32]:   File "/Users/themaninthesuitcase/Source/3rdParty/Ultimaker/Uranium/UM/Qt/Bindings/BackendProxy.py", line 9, in <module>
cura.CrashHandler.show [32]:     from UM.Backend.Backend import BackendState
cura.CrashHandler.show [32]:   File "/Users/themaninthesuitcase/Source/3rdParty/Ultimaker/Uranium/UM/Backend/Backend.py", line 6, in <module>
cura.CrashHandler.show [32]:     from UM.Backend.SignalSocket import SignalSocket
cura.CrashHandler.show [32]:   File "/Users/themaninthesuitcase/Source/3rdParty/Ultimaker/Uranium/UM/Backend/SignalSocket.py", line 34, in <module>
cura.CrashHandler.show [32]:     class _SocketListener(Arcus.SocketListener):
cura.CrashHandler.show [32]: AttributeError: module 'Arcus' has no attribute 'SocketListener'
```
I've checked that libarcus and uranium are up to date. I've also nuked my master and re-pulled from the Ultimaker repo.  If I'm reading this right it's a bad import statement?

fieldOfView, 2016-12-06:
Perhaps see https://github.com/Ultimaker/libArcus/issues/19#issuecomment-193419915

themaninthesuitcase, 2016-12-06:
Got it, 30 mins faffing about with libarcus builds and python paths and I've got it.
For those that follow
```
git clone git@github.com:Ultimaker/libArcus.git
cd libArcus
mkdir build && cd build
make ..
make
```
Add the build folder (`/Users/themaninthesuitcase/Source/3rdParty/Ultimaker/libArcus/build` in my case) to the `PYTHONPATH` and it should pop into life.

themaninthesuitcase, 2016-12-06:
[image]
Well it works! Though there doesn't seem to be an obvious way to default to gcode which I am assuming is the preference from a UM point of view.  I've been digging about in the STL example posted above but I can't see anything obvious.

Ghostkeeper, 2016-12-07:
The default selection is decided based on the formats that the printer supports. See for example the Ultimaker 2 Extended definition: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/ultimaker2_extended.def.json

This is identified by MIME type. The MIME types are then defined by the __init__.py file of the writer plug-in. These files also define the extension for the file: https://github.com/Ultimaker/Cura/blob/master/plugins/GCodeWriter/__init__.py#L21

ianpaschal, 2018-02-27:
Closing due to inactivity (1 year+)

closed 2018-02-27 by ianpaschal