Ultimaker/Cura #1063: I can not switch the printer profile

landway, 2016-10-21:
Hello,

I have a fundamental problem with CURA 2.3.0. I can not switch the printer profile. Neither the menu nor the preferences window I can not switch the printer profile. In preferences window I use button activate, but It does nothing...

PS: in attachment [screenshot cura um3 to um2go buildplate] it can be seen changing build platform, but not build envelope...

Love new CURA, please help....

[image]
[image]
[image]
[image]

[cura.log.txt](https://github.com/Ultimaker/Cura/files/544387/cura.log.txt)

[DxDiag.txt](https://github.com/Ultimaker/Cura/files/544365/DxDiag.txt)

nallath, 2016-10-21:
Oh wow. That's a new one. I'll have a look at your logs.

nallath, 2016-10-21:
I think this is caused by an issue that was in cura 2.1 which created profiles that shouldn't be there. We can't identify those profiles correctly, so they get converted to 2.3 automatically (but cause issues).

Could you try clearing your local data (make a backup first)?

landway, 2016-10-21:
Yes I can... Can you tell me how, please? Btw, this problem I have from 2.3.0.rc1 so I uninstall it and install 2.3.0. and I have same problem. You are right with profiles.... i have old profiles in new instalation of CURA.

nallath, 2016-10-21:
Go to help->show configuration folder

That is the folder that you need to backup (and then delete & reboot Cura)

landway, 2016-10-21:
OK, I delete all files from c.f. and now it is ok. You are greate **nallath**! Everywhere I will praise you. Thank you.

closed 2016-10-21 by landway