Ultimaker/Cura #1066: CMakeLists.txt uses wrong python dir

greyltc, 2016-10-21:
Hey devs, I maintain Arch Linux's cura package. I want to say thanks for a smooth 2.3.0 release! This is the first time that this and all its related parts have been properly tagged and all in sync.

FYI, your CMakeLists.txt isn't using the correct python directory in Arch (for Uranium and Cura). That's the only thing I had to patch here to get this working. I fixed it with the following seds
```bash
  sed -i 's,DESTINATION lib/python${PYTHON_VERSION_MAJOR}/dist-packages,DESTINATION lib/python${PYTHON_VERSION_MAJOR}.${PYTHON_VERSION_MINOR}/site-packages,g' CMakeLists.txt
  sed -i 's,DESTINATION lib/python${PYTHON_VERSION_MAJOR}/dist-packages/cura),DESTINATION lib/python${PYTHON_VERSION_MAJOR}.${PYTHON_VERSION_MINOR}/site-packages/cura),g' CMakeLists.txt

```

nallath, 2016-10-24:
That's odd. I know that @awhiemstra develops on arch by default. I think he should have a look at this.

thopiekar, 2016-10-24:
In my point of view using python${PYTHON_VERSION_MAJOR}.${PYTHON_VERSION_MINOR} would be correct here. Debian's packaging tools are moving the files to python${PYTHON_VERSION_MAJOR} and manage everything automatically.
It only would need a set of changes to cura-build to support these paths.

awhiemstra, 2016-10-24:
The installation paths were set up for Debian packaging. Now that we are moving away from that and building AppImages we can also fix errors like these properly.

closed 2017-07-27 by nallath