Ultimaker/Cura #1070: "Machine center is zero" + "Z seam Allighment: BACK" cause curaengine.exe to crash

davidwyl, 2016-10-23:
if i tick **"Machine center is zero"** in custom fdm printer and set "**Z seam Allighment**" to **BACK** , this will cause curaengine.exe to crash and stopped. This Message will pop up **"CuraEngine.exe has stopped working".**.
the problem disappear if i untick the **Machine center is zero**.

Ghostkeeper, 2016-10-24:
I could reproduce this also. I also tried on other meshes though (a half-sphere and a tetrahedron), but couldn't reproduce this on those. Perhaps it only happens with meshes that are not convex?

davidwyl, 2016-10-24:
Working fine on square cube too but doesn't work on most of the stl files... Definitely some bug here.

smartavionics, 2016-11-20:
Bug confirmation - same behaviour with cura 2.3.1 on Linux.

smartavionics, 2016-11-20:
BTW - I made a discovery, moving the object away from the centre of the build plate (possibly so that none of the object overlaps the centre?) appears to avoid the crash.

BagelOrb, 2016-11-21:
In Cura 2.4 Z Seam Alignment Back will be replaced by User Specified, so that you can choose yourself close to which point you want the z-seam to be.

With that feature the bug also disappeared :)

closed 2016-11-21 by BagelOrb