Ultimaker/Cura #1072: Crash on startup

TheWalterWhite, 2016-10-24:
I was able to launch cura once but now even if I reinstall it crashes on launch.  Log:
2016-10-23 21:57:39,905 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\Bindings\Theme.py (load [118]): Loading theme file: D:\Program Files\Cura 2.3\resources\themes\cura\theme.json
2016-10-23 21:57:39,905 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\Bindings\Theme.py (load [118]): Loading theme file: D:\Program Files\Cura 2.3\resources\themes\cura\theme.json
2016-10-23 21:57:41,328 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\Bindings\Theme.py (load [118]): Loaded theme D:\Program Files\Cura 2.3\resources\themes\cura
2016-10-23 21:57:41,329 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\Bindings\Theme.py (load [118]): Loaded theme D:\Program Files\Cura 2.3\resources\themes\cura
2016-10-23 21:57:43,719 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]): An uncaught exception has occurred!
2016-10-23 21:57:43,720 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]): TypeError: unable to convert a Python 'str' object to a C++ 'bool' instance
2016-10-23 21:57:43,720 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]): An uncaught exception has occurred!
2016-10-23 21:57:43,720 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]): TypeError: unable to convert a Python 'str' object to a C++ 'bool' instance
2016-10-23 21:57:43,721 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\cura\Settings\ContainerManager.py (getContainerMetaDataEntry [171]): Could not get metadata of container  because it was not found.
2016-10-23 21:57:43,721 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]): An uncaught exception has occurred!
ld not get metadata of container  because it was not found.
2016-10-23 21:57:43,721 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]): TypeError: unable to convert a Python 'str' object to a C++ 'bool' instance
2016-10-23 21:57:43,721 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]): An uncaught exception has occurred!
2016-10-23 21:57:43,721 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]): TypeError: unable to convert a Python 'str' object to a C++ 'bool' instance
2016-10-23 21:57:45,508 - DEBUG - D:\Program Files\Cura 2.3\plugins\UM3NetworkPrinting\DiscoverUM3Action.py (_createAdditionalComponentsView [133]): Creating additional ui components for UM3.
2016-10-23 21:57:45,508 - DEBUG - D:\Program Files\Cura 2.3\plugins\UM3NetworkPrinting\DiscoverUM3Action.py (_createAdditionalComponentsView [133]): Creating additional ui components for UM3.
2016-10-23 21:57:45,825 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Mesh\MeshData.py (calculateNormalsFromVertices [339]): Calculating normals took 0.12100720405578613 seconds
2016-10-23 21:57:45,825 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Mesh\MeshData.py (calculateNormalsFromVertices [339]): Calculating normals took 0.12100720405578613 seconds
2016-10-23 21:57:45,826 - DEBUG - D:\Program Files\Cura 2.3\plugins\FileHandlers\STLReader\STLReader.py (read [35]): Loaded a mesh with 212304 vertices
2016-10-23 21:57:45,966 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Mesh\MeshData.py (approximateConvexHull [305]): approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.1390080451965332 seconds. 545 input vertices. 545 output vertices.
2016-10-23 21:57:46,451 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Mesh\MeshData.py (approximateConvexHull [305]): approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.05100297927856445 seconds. 545 input vertices. 545 output vertices.
2016-10-23 21:57:46,452 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Mesh\ReadMeshJob.py (run [30]): Loading mesh took 0.841048002243042 seconds
2016-10-23 21:57:46,452 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Mesh\MeshData.py (approximateConvexHull [305]): approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.052002906799316406 seconds. 545 input vertices. 545 output vertices.
2016-10-23 21:57:46,453 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Mesh\ReadMeshJob.py (run [30]): Loading mesh took 0.842048168182373 seconds
2016-10-23 22:08:52,780 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin FileLogger
2016-10-23 22:08:52,783 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeProfileReader
2016-10-23 22:08:52,786 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeWriter
2016-10-23 22:08:52,807 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin ImageReader
2016-10-23 22:08:52,811 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin LayerView
2016-10-23 22:08:52,814 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin LegacyProfileReader
2016-10-23 22:08:52,817 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin LocalFileOutputDevice
2016-10-23 22:08:52,820 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin MachineSettingsAction
2016-10-23 22:08:52,827 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Preferences.py (getValue [70]): Tried to get the value of non-existing setting cura/active_mode.
2016-10-23 22:08:52,827 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin PerObjectSettingsTool
2016-10-23 22:08:52,832 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin PostProcessingPlugin
2016-10-23 22:08:52,838 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin RemovableDriveOutputDevice
2016-10-23 22:08:52,843 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin SliceInfoPlugin
2016-10-23 22:08:52,846 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin SolidView
2016-10-23 22:08:52,852 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin CameraTool
2016-10-23 22:08:52,904 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin MirrorTool
2016-10-23 22:08:52,910 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin RotateTool
2016-10-23 22:08:52,957 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin ScaleTool
2016-10-23 22:08:53,314 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin SelectionTool
2016-10-23 22:08:53,452 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin TranslateTool
2016-10-23 22:08:53,513 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin UltimakerMachineActions
2016-10-23 22:08:53,607 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin UM3NetworkPrinting
2016-10-23 22:08:53,657 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin UpdateChecker
2016-10-23 22:08:53,657 - INFO - D:\Program Files\Cura 2.3\plugins\UpdateChecker\UpdateChecker.py (run [28]): Checking for new version of cura
2016-10-23 22:08:53,782 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin USBPrinting
2016-10-23 22:08:53,805 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin VersionUpgrade21to22
[... cut ...]
2016-10-23 22:08:59,702 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]): An uncaught exception has occurred!
2016-10-23 22:08:59,702 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]): TypeError: unable to convert a Python 'str' object to a C++ 'bool' instance
2016-10-23 22:08:59,907 - DEBUG - D:\Program Files\Cura 2.3\plugins\UM3NetworkPrinting\DiscoverUM3Action.py (_createAdditionalComponentsView [133]): Creating additional ui components for UM3.

Ghostkeeper, 2016-10-24:
Again that mysterious warning:

    2016-10-23 22:08:59,701 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\cura\Settings\ContainerManager.py (getContainerMetaDataEntry [171]): Could not get metadata of container  because it was not found.

It was encountered here, too: https://github.com/Ultimaker/Cura/issues/1028

swissgiant, 2016-10-26:
Same here.
guess its about loading the profile from windows and the path is locked wrong.
in german the install path is translated to "c:\programme"
gui is searched in the path "c:\program files\..."

log attached

2016-10-26 23:56:22,173 - DEBUG - C:\Program Files\Cura 2.3\plugins\MachineSettingsAction\MachineSettingsAction.py (_onContainerAdded [40]): Not attaching MachineSettingsAction to ultimaker_original_dual; Multi-extrusion printers are not supported
2016-10-26 23:56:22,248 - DEBUG - C:\Program Files\Cura 2.3\plugins\MachineSettingsAction\MachineSettingsAction.py (_onContainerAdded [40]): Not attaching MachineSettingsAction to ultimaker3_extended; Multi-extrusion printers are not supported
2016-10-26 23:56:22,362 - DEBUG - C:\Program Files\Cura 2.3\plugins\MachineSettingsAction\MachineSettingsAction.py (_onContainerAdded [40]): Not attaching MachineSettingsAction to ultimaker2_plus; Machines that use variants are not supported
2016-10-26 23:56:22,507 - DEBUG - C:\Program Files\Cura 2.3\plugins\MachineSettingsAction\MachineSettingsAction.py (_onContainerAdded [40]): Not attaching MachineSettingsAction to ultimaker3; Multi-extrusion printers are not supported
2016-10-26 23:56:27,168 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_logSocketState [165]): Socket state changed to Connected
2016-10-26 23:56:27,169 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_onSocketStateChanged [155]): Backend connected on port 49674
2016-10-26 23:56:27,169 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Controller.py (setActiveView [80]): Setting active view to SolidView
2016-10-26 23:56:27,858 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\Bindings\Theme.py (load [118]): Loading theme file: C:\Program Files\Cura 2.3\resources\themes\cura\theme.json
2016-10-26 23:56:27,915 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\Bindings\Theme.py (load [118]): Loaded theme C:\Program Files\Cura 2.3\resources\themes\cura
2016-10-26 23:56:27,957 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]): An uncaught exception has occurred!
2016-10-26 23:56:27,957 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]): TypeError: unable to convert a Python 'str' object to a C++ 'bool' instance
2016-10-26 23:56:27,958 - WARNING - D:\2.3\build\inst\lib\python3.4\site-packages\cura\Settings\ContainerManager.py (getContainerMetaDataEntry [171]): Could not get metadata of container  because it was not found.
2016-10-26 23:56:27,958 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]): An uncaught exception has occurred!
2016-10-26 23:56:27,958 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]): TypeError: unable to convert a Python 'str' object to a C++ 'bool' instance
2016-10-26 23:56:28,144 - DEBUG - C:\Program Files\Cura 2.3\plugins\UM3NetworkPrinting\DiscoverUM3Action.py (_createAdditionalComponentsView [133]): Creating additional ui components for UM3.

zsserg, 2016-11-24:
Same here - crash on start after update from 2.1 to 2.3.1 =(
2016-11-24 09:40:29,231 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin ConsoleLogger
2016-11-24 09:40:29,237 - INFO - C:\Program Files\Cura 2\plugins\CuraEngineBackend\CuraEngineBackend.py (__init__ [40]): Found CuraEngine at: C:\Program Files\Cura 2\CuraEngine.exe
2016-11-24 09:40:29,238 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraEngineBackend
2016-11-24 09:40:29,240 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileReader
2016-11-24 09:40:29,243 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileWriter
2016-11-24 09:40:29,267 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Plugin Doodle3D uses an incompatible API version, ignoring
2016-11-24 09:40:29,269 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFWriter
2016-11-24 09:40:29,272 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJReader
2016-11-24 09:40:29,274 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJWriter
2016-11-24 09:40:29,279 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin STLReader
2016-11-24 09:40:29,282 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin STLWriter
2016-11-24 09:40:29,285 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin FileLogger
2016-11-24 09:40:29,288 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeProfileReader
2016-11-24 09:40:29,291 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeWriter
2016-11-24 09:40:29,295 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin ImageReader
2016-11-24 09:40:29,300 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin LayerView
2016-11-24 09:40:29,304 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin LegacyProfileReader
2016-11-24 09:40:29,309 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin LocalFileOutputDevice
2016-11-24 09:40:29,312 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin MachineSettingsAction
2016-11-24 09:40:29,316 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin PerObjectSettingsTool
2016-11-24 09:40:29,321 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin PostProcessingPlugin
2016-11-24 09:40:29,326 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin RemovableDriveOutputDevice
2016-11-24 09:40:29,334 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin SliceInfoPlugin
2016-11-24 09:40:29,338 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin SolidView
2016-11-24 09:40:29,341 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin CameraTool
2016-11-24 09:40:29,349 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin MirrorTool
2016-11-24 09:40:29,357 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin RotateTool
2016-11-24 09:40:29,362 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin ScaleTool
2016-11-24 09:40:29,367 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin SelectionTool
2016-11-24 09:40:29,371 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin TranslateTool
2016-11-24 09:40:29,378 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin UltimakerMachineActions
2016-11-24 09:40:29,408 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin UM3NetworkPrinting
2016-11-24 09:40:29,435 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin UpdateChecker
2016-11-24 09:40:29,436 - INFO - C:\Program Files\Cura 2\plugins\UpdateChecker\UpdateChecker.py (run [28]): Checking for new version of cura
2016-11-24 09:40:29,447 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin USBPrinting
2016-11-24 09:40:29,453 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin VersionUpgrade21to22
[... cut ...]
2016-11-24 09:40:29,549 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]):   File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 259, in _storeOldFile
2016-11-24 09:40:29,549 - CRITICAL - D:\2.3\build\inst\lib\python3.4\site-packages\cura\CrashHandler.py (show [24]): FileNotFoundError: [WinError 3] Системе не удается найти указанный путь: 'C:\\Program Files\\Cura 2\\resources\\./profiles\\general\\High+Quality.cfg' -> 'C:\\Program Files\\Cura 2\\resources\\./profiles\\old\\1\\general\\High+Quality.cfg'

Ghostkeeper, 2016-11-24:
In zsserg's case, it seems to be crashing because it tries to upgrade files inside his Cura installation itself, in the Program Files folder. Why it is doing that is a mystery to me. It seems to have been added to the list of configuration directories?

zsserg, 2016-11-25:
I have no idea why it did it either. Iwas able to resolve it only via doing clean install & then adding all the materials/settings/printers manually.

stuff-stuff, 2017-01-24:
Hey!
I'm not sur if it will help but I try anyway ...

I am running the last Ubuntu (LTS) version, and I got that error/crash went I try so start Cura

`WARNING - cura.Settings.ContainerManager.getContainerMetaDataEntry [175]: Could not get metadata of container  because it was not found.`

After looking around to install dependencies or any other packages that could resolve the problem without any success I realise MySQL was not install ...
So I `sudo apt-get install parallel` and now Cura boot up and did not crash!

Ghostkeeper, 2017-01-25:
Eh... Cura doesn't use any database whatsoever. I doubt that installing MySQL would have any effect.

stuff-stuff, 2017-01-25:
Yeah I was trying stuff ...

CreshZmine, 2017-01-31:
This issue also happens to me on Arch.

> 2017-01-31 11:16:26,242 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ConsoleLogger
> 2017-01-31 11:16:26,255 - INFO - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (__init__ [40]): Found CuraEngine at: /usr/bin/CuraEngine
> 2017-01-31 11:16:26,255 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraEngineBackend
> 2017-01-31 11:16:26,256 - WARNING - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Plugin ConsoleLogger was already loaded
> 2017-01-31 11:16:26,261 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFWriter
> 2017-01-31 11:16:26,262 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJWriter
> 2017-01-31 11:16:26,263 - WARNING - /usr/lib/uranium/plugins/FileHandlers/STLReader/STLReader.py (<module> [5]): Could not find numpy-stl, falling back to slower code.
> 2017-01-31 11:16:26,264 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLReader
> 2017-01-31 11:16:26,266 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLWriter
> 2017-01-31 11:16:26,268 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJReader
> 2017-01-31 11:16:26,271 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LocalFileOutputDevice
> 2017-01-31 11:16:26,272 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin FileLogger
> 2017-01-31 11:16:26,293 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin UpdateChecker
> 2017-01-31 11:16:26,294 - INFO - /usr/lib/uranium/plugins/UpdateChecker/UpdateChecker.py (run [28]): Checking for new version of cura
> 2017-01-31 11:16:26,298 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SimpleView
> 2017-01-31 11:16:26,299 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin WireframeView
> 2017-01-31 11:16:26,306 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RotateTool
> 2017-01-31 11:16:26,308 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SelectionTool
> 2017-01-31 11:16:26,313 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ScaleTool
> 2017-01-31 11:16:26,316 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin TranslateTool
> 2017-01-31 11:16:26,318 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CameraTool
> 2017-01-31 11:16:26,322 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MirrorTool
> 2017-01-31 11:16:26,323 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileWriter
> 2017-01-31 11:16:26,325 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileReader
> 2017-01-31 11:16:26,327 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin AutoSave
> 2017-01-31 11:16:26,332 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin VersionUpgrade21to22
> 2017-01-31 11:16:26,334 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeWriter
> 2017-01-31 11:16:26,334 - WARNING - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Plugin ConsoleLogger was already loaded
> 2017-01-31 11:16:26,338 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ImageReader
> 2017-01-31 11:16:26,340 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ChangeLogPlugin
> 2017-01-31 11:16:26,344 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RemovableDriveOutputDevice
> 2017-01-31 11:16:26,344 - WARNING - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Plugin 3MFWriter was already loaded
> 2017-01-31 11:16:26,344 - WARNING - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Plugin OBJWriter was already loaded
> 2017-01-31 11:16:26,344 - WARNING - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Plugin STLReader was already loaded
> 2017-01-31 11:16:26,344 - WARNING - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Plugin STLWriter was already loaded
> 2017-01-31 11:16:26,345 - WARNING - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Plugin OBJReader was already loaded
> 2017-01-31 11:16:26,347 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LegacyProfileReader
> 2017-01-31 11:16:26,353 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin X3DReader
> 2017-01-31 11:16:26,353 - WARNING - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Plugin LocalFileOutputDevice was already loaded
> 2017-01-31 11:16:26,356 - WARNING - .../site-packages/UM/Preferences.py (getValue [70]): Tried to get the value of non-existing setting cura/active_mode.
> 2017-01-31 11:16:26,356 - INFO - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin PerObjectSettingsTool
> 2017-01-31 11:16:26,356 - WARNING - .../site-packages/UM/PluginRegistry.py (loadPlugin [63]): Plugin FileLogger was already loaded
[... cut ...]
> 2017-01-31 11:16:29,042 - DEBUG - .../site-packages/UM/Qt/Bindings/Theme.py (load [118]): Loaded theme /usr/share/cura/resources/themes/cura
> QObject::connect: Parentheses expected, signal MainWindow::
> QObject::connect: Parentheses expected, signal MainWindow::
> QObject::connect: Parentheses expected, signal MainWindow::
> 2017-01-31 11:16:29,083 - WARNING - .../site-packages/cura/Settings/ContainerManager.py (getContainerMetaDataEntry [171]): Could not get metadata of container  because it was not found.
> Errore di segmentazione (core dump creato)

Also with sudo permission crash.

awhiemstra, 2017-01-31:
Arch updated to Qt 5.8 which was released a few days ago. PyQt does not yet officially support Qt 5.8. Currently, there is a crash in PyQt when trying to access properties of the Window. This causes Cura to crash. The only known workaround I have for that is to comment out everything that tries to use the window properties. I hope this will be fixed once PyQt properly supports Qt 5.8.

awhiemstra, 2017-01-31:
Note that I found all of that out because I run Arch myself and thus already ran into the same issue. :)

greyltc, 2017-02-07:
@awhiemstra are you aware of any related upstream PyQt bug report for this issue?

Geend, 2017-02-07:
For everyone using Arch (or everyone using Qt 5.8): You can use the AppImage version of Cura 2.4 (beta) available [here](https://ultimaker.com/en/products/cura-software) until PyQt supports Qt 5.8.

awhiemstra, 2017-02-07:
@greyltc I have not seen any mention of it. I have not had the time yet to produce a minimal test case to report to them. Then again, there is also no official release of PyQt 5.7.2 yet.

FirefighterBlu3, 2017-02-17:
PyQt has been updated to 5.8 but the segfault crash still occurs

FirefighterBlu3, 2017-02-17:
@Geend can this be run alongside my installed version or do i need to wipe it out?

Ghostkeeper, 2017-02-17:
The AppImage will run alongside your installed version.

They share the same setting directory though. If you're mixing version 2.3 with 2.4 or 2.5 you could expect to lose your saved profiles. If you're mixing the same minor version that shouldn't happen much.

awhiemstra, 2017-02-20:
As mentioned in the other thread, I have now reported the crash upstream, waiting to see what the reaction is.

nallath, 2017-07-27:
As this is an upstream, i'm closing this.

closed 2017-07-27 by nallath