Ultimaker/Cura #1073: 2.3 beta infill flow too low

aricblunk, 2016-10-24:
I've noticed my prints with 2.3 have very anemic infill flow, it almost seems like infill density (line distance) is mistakenly being used as the amount of flow during infill moves. On that note, it would be awesome to have settings for separate flow values for infill/wall/skin/supports, making it easier to print stronger models with weaker support, without compromising on complexity of support structure.

Ghostkeeper, 2016-10-25:
While Cura does expose a "Flow" setting, the flow as you know it is typically adjusted by adjusting the line width. You can change the line width of infill/wall/skin/support separately. Changing the line width also changes the spacing of lines, leaving more room for a thicker line. If there is a general overextrusion or underextrusion problem you can adjust that with the single Flow setting, but it's uncommon to have an underextrusion problem that only occurs on infill lines.

The flow for infill is not determined on line distance. It's the other way around: The line distance is dependant on the line width. Or at least, it should be.

aricblunk, 2016-10-25:
I see, thanks for the info. I hadn't thought to try adjusting line width, but that makes perfect sense. I just ran a quick test print with 0.5mm infill line width instead of 0.4 and it looks much better, the infill lines actually stay solid instead of fraying now.

closed 2016-10-25 by Ghostkeeper