Ultimaker/Cura #1077: Multiple object slicing problem

MrLadin97, 2016-10-25:
Hello, I have problem with slicing more than one object, when I slice just one stl, then everything is ok, but when I try to add second object, then the inner structure is missing. First object is always ok, but the second or third one always fail. Here is screenshot of my problem: (https://cl.ly/1w1D1Q1x3Z2G)
(https://cl.ly/3U0V442a0845)
Does someone please know, where can be the problem?
Thank you

Cura version 2.3
MacOS version: 10.12

Ghostkeeper, 2016-10-26:
My first thought is that the model might not be water-tight or has extra faces inside. If you look at the mesh in X-Ray View (right above Layer View in the menu), is there anything red? If there is, strange things can happen, including that some pieces may disappear or reappear if there are other pieces on the same height. You would then need to repair the mesh in a CAD application and load it back in.

If that is not the case, then I'd wager it's a bug in CuraEngine. I can't reproduce it outright though, so we'd need to find a model where it happens before we can fix it.

MrLadin97, 2016-10-26:
Thank you very much for your reply, I think that model is ok, because when I slice just one object, it's ok, but when I duplicate it, then the duplicated model fails. The object I am trying to slice is 3D Lab Print Messerschmitt. In attachment, there are my Cura profile and stl test files.
[cura_profile_and_stl_file.zip](https://github.com/Ultimaker/Cura/files/552874/cura_profile_and_stl_file.zip)

Ghostkeeper, 2016-10-26:
The model doesn't seem to have a top and bottom face. X-Ray view shows quite a lot of red: 
[image]

I think this model was spliced to make it fit in the build volume by your Messerschmitt, but it didn't make the mesh water tight by putting a closing face on the top and bottom.

MrLadin97, 2016-10-26:
No, sorry but I don't think that you understand correctly my problem, i don't have any problem with printing just  one object. I already printed a lot of them, but I cannot slice more than one object at once. When you try to multiple this object and then view it in layer view, then you will see missing structure on one model. Missing top and bottom is wished. In profile I attached above, I have defined 0mm top/bottom layer thickness.
[image]

Here is photo of this plane printed with older cura 15.04.6 There wasn't any problem with printing for example 5 object at once.
[image]

fieldOfView, 2016-10-26:
The problem is that the model has only single walls without thickness. This would normally not be printable, but CuraEngine does some magic to determine what is "inside" and what is "outside". This magic comes at no guarantee of it working; it is a best guess made by CuraEngine. 

Apparently the magic just so happens to work when there is only the model on the buildplate, but the geometry becomes too tricky to figure out for CuraEngine when you have multiple objects on the buildplate.

chrismrutherford, 2017-03-15:
I have the exact same problem.  The first model slices fine, any additional loaded models have random internal structure removed.  Interestingly if I duplicate the same model then the problem doesn't occur.  I'm going to revert cura (possibly bisect revisions) until I find a version that works.

fieldOfView, 2017-07-23:
Try slicing with these two settings from the Mesh Fixes category:

Union overlapping values: unchecked
Merged meshes overlap: 0

You can use the "Search" box on the top of the custom settings to search for these settings.

nallath, 2017-07-27:
Closing this due to inactivity.

closed 2017-07-27 by nallath