Ultimaker/Cura #1082: Absolutely insane retracts

claus1009, 2016-10-26:
In 2.3 it's still the same as in 2.3.-beta. (at least for UM2+)

Cura adds **absolutely insane retracts** of more than 20cm, 
 - after having placed the printhead in the initial print-positions
-  after having finished the brim
- at the end of print

G10 is added at those positions in the gcode.
It seems that the default value for G10 is somwhere set wrong.

Ghostkeeper, 2016-10-27:
I just tried this, but couldn't find any G207 commands in the g-code. Thus the retraction length is determined by the UM2+ itself, rather than by Cura.

On the UM2(+), you can see the retraction amounts it uses for G10 commands using the display on the printer: Material -> Settings -> Customize -> Retraction.

claus1009, 2016-10-27:
Checked on my UM2+. (Material -> Settings -> Customize -> Retraction.)
Retraction Length is set to 6.5mm ==> so quite OK

But theses insane ratrctions of more than 20cm happen exactly where G10 is placed in the gcode.
Those G10 commands were not place in gcode from 15.04.5.

Just for comparison the first few lines of code 15.04.5 vs. 2.3.

**Cura15.02.5:**

> ;FLAVOR:UltiGCode
> ;TIME:2587
> ;MATERIAL:18920
> ;MATERIAL2:0
> ;NOZZLE_DIAMETER:0.800000
> ;NOZZLE_DIAMETER2:0.800000
> 
> ;Layer count: 16
> ;LAYER:0
> M107
> G0 F6000 X66.864 Y74.189 Z0.250
> ;TYPE:SKIRT
> G1 F1800 X178.069 Y44.391 E23.02561

**Cura 2.3:**

> ;FLAVOR:UltiGCode
> ;TIME:1220
> ;MATERIAL:4563
> ;MATERIAL2:0
> ;NOZZLE_DIAMETER:0.8
> ;Generated with Cura_SteamEngine 2.3.0
> 
> ;LAYER_COUNT:23
> ;LAYER:0
> M107
> G10
> G1 Z21
> G0 F3600 X78.588 Y95.721 Z1.53
> ;TYPE:SKIRT
> G1 Z.53
> G11
> G1 F1200 X79.098 Y95.374 E0.22885

BagelOrb, 2016-10-27:
Cura 15 didn't do a retraction from the start to the first thing printed,
while Cura 2.x does.

That's unlikely to be the cause of the problem here.

I bet that further on in the gcode from 15.04 you will find G10 commands,
because retraction was also a feature of Cura 15.04.

Perhaps your firmware is outdated. Try updating your firmware.

In the mean time I will try and get someone from the firmware team to have
a look at your issue.

BagelOrb, 2016-10-28:
I've had a better look at the gcode you posted and tried to recreate.

The weird thing is that in your Cura 15.04 gcode the first E-move is already 20mm ahead, while the Cura 2.3 gcode neatly starts at E0.
When I generate a gcode using 15.04.6 the first E move isn't 20mm ahead.

Perhaps you are using an old version of Cura 15.x and an old version of the firmware.
Try updating to Cura 15.04.6 and update the firmware on your printer. ( Machine > Install default firmware... )

claus1009, 2016-10-28:
In my case:
Printer firmware is always exactly the one released with the cura version.
I always take meticulous care that they are in sync.
So a mixup can be absolutely excluded.

Regarding your feedback on 15.4.6's first E-move I can only say .. I never faced any problems with 15,.4.6! I would call it pretty mature- regarding my expectations.

This ISSUE #1082 really is only about those extremely long retracts.
I would like to understand why they only happen in 2.3.
And the crazy thing is that those insane retracts only happen at the beginning and at the end of the print.

Just to make one thing absolutely clear:
In my setup everything is really out of he box.
No messing/tweaking around with code or stuff like this!

Installed 2.3. --> Connected Printer via USB --> update firmware  --> re-sliced old project -> re-printed

Nothing else!!!

The only thing which I re-used from 15.04.6 is the MATERIAL.TXT file.
After updating firmware to 2.3 I did
Maertrial-> Settings-> Import from SD 

And MATERIAL.txt looks like this: it contains only one material, (I only use PETG from a single prodecer)

[material]
name=PETG_220
temperature=220
temperature_0.40=220
temperature_0.25=210
temperature_0.60=225
temperature_0.80=230
temperature_1.00=235
bed_temperature=90
fan_speed=100
flow=105
diameter=2.85 

Can this cause any problems?

BagelOrb, 2016-10-28:
Just to buy clear: I tested with Cura 15.04.6 and couldn't reproduce the
gcode you generated with Cura 15.02.

Maybe there is a difference in G92 being called at the start in one of the
Cura versions involved here....

BagelOrb, 2016-10-28:
What happens when you upgrade from Cura 15.02 to 15.04.6?

claus1009, 2016-10-28:
Just rechecked!
I'm comparing 15.04.5 with 2.3 

Do you expect any diffeernces 15.04.5 vs. 15.04.6 regarding this issue!

claus1009, 2016-10-28:
How come you think I'm using 15.02 ??

fieldOfView, 2016-10-28:
Your post https://github.com/Ultimaker/Cura/issues/1082#issuecomment-256599556 suggests you are using 15.02.5 (but I am fairly sure there never was a 15.02.5, so you probably meant 15.04.5 there)

claus1009, 2016-10-28:
ah .. ok ... the comment in front of the GCODE-listing 6 post ago  caused the disconnect
Oh sorry ..
that was a typo ..

mentioned that I'm using  15.04.5 twice just in the two lines befor that ,..

Sorry for the inconvenience ...

Again ..
want to come back to your second post in this thread ...

_Cura 15 didn't do a retraction from the start to the first thing printed,
while Cura 2.x does._

So here we have a common understanding.
This comment fits to my observation.

And I obsereve ...

Those strange retracts only happen where 2.3 places a retract while 15.04.5 doesn't.
All others turn out normally!

I think we should for the moment focus on that symptom?

Is there anything in the firmware which causes those retracts to be handeled different to all otheres?

ChrisTerBeke, 2017-11-14:
Is this still an issue?

ianpaschal, 2018-02-26:
Closed for inactivity and the fact that we will no longer work on 2.x.

closed 2018-02-26 by ianpaschal