Ultimaker/Cura #1085: VersionUpgrade does not ensure unique container names

fieldOfView, 2016-10-27:
In Cura 2.1 it was possible to have a profile with the same name as a printer. Eg if the user names his UM2 "Ultimaker 2" (the default choice), he can create a quality profile named "Ultimaker 2" as well.

When upgrading Cura to 2.3, the VersionUpgrade plugin blindly creates a ContainerStack with the id "Ultimaker 2" and a InstanceContainer with the id "Ultimaker 2". This leads to an error where either the InstanceContainer is found as a ContainerStack, or in a more positive case an error that a container with the id "Ultimaker 2" already exists. In either case, Cura does not start.

Note that when trying to create a profile named "Ultimaker 2"  in Cura 2.3, it is correctly renamed "Ultimaker 2 #2".

ChrisTerBeke, 2017-11-14:
Is this still an issue?

Ghostkeeper, 2017-11-23:
Yes, and this causes Cura to brick itself. The version upgrade system upgrades file-by-file without looking at whether a file already exists.

Vandrasc, 2018-04-09:
I think we fixed this, @Ghostkeeper can you confirm (again) that it's still an issue? ^^

Ghostkeeper, 2018-05-02:
Yes, it's still an issue. Renames occurred in the upgrade from 2.7 to 3.0. Also, if we ever want to rename a file again in a future version upgrade then this will again occur.

closed 2018-11-26 by Appesteijn