Ultimaker/Cura #1086: 3mf meta data support?

peteruithoven, 2016-10-27:
The 3mf file format should allow giving more meta data about models, how it should be sliced / printed. What's the current status of that?

nallath, 2016-10-27:
I've attended the 3mf meeting last week, which helped me get a much better overview on what we need (and can) do with 3mf. 

We're going to focus on getting more complete 3mf support for 2.4. I've been spending most of my time not spent on fixing stuff for the 2.3.1 release on improving the god-mode plugin so it's easier to debug SceneNodes (3mf allows for pretty complex scene loading / saving, which we don't fully support)

We hope to get at least full workspace saving / loading possible by means of 3mf (which also implies that Cura is going to save certain data that only makes sense to Cura in those 3mf files). This is still a step from getting certain meta data about models (such as "design intents" that indicate that certain models should be flexible or certain parts of a model should be flexible). Generating that kind of data is something that should (in my opinion) be done by "3mf producers" (aka CAD/CAM software).

peteruithoven, 2016-10-27:
Awesome!

> We hope to get at least full workspace saving / loading possible by means of 3mf (which also implies that Cura is going to save certain data that only makes sense to Cura in those 3mf files). T

Could this include profile information, like layer height? (I understand the downside, compared to intents, that this is low level and probably only usable for Cura).

nallath, 2016-10-27:
That is exactly what we want to do by that.

meghan81, 2016-10-28:
It would be great to see the same export of profile information saved into X3D format, since that is also XML-based. Cura now imports X3D, and I'm checking on progress to save as X3D.

nallath, 2017-07-27:
We now support full project saving, so this is done.

closed 2017-07-27 by nallath