Ultimaker/Cura #1087: Wipe and Prime tower could use a revisit, especially in the Ultimaker 3

MTKelly, 2016-10-27:
Wipe and Prime tower has been a source of frustration for me for years. Even with a brim it seems to fall over at a very high failure rate. 

My belief is that this happens because the oozing bits protrude up which get caught on the nozzle and then enough force is generated to snap a layer bond or the bond to the bed. 

On all of our prints we've done on our UM3 the wipe and prime tower failed, but luckily didn't kill the print with it. Increasing the area of the wipe and prime tower can help with this, but wastes a lot of material. 

My proposal is instead to print a small dump bucket on each print before starting the main print.

My vision is a small-ish bucket, around 20mmx20mm with a 1-2 layer bottom. Then print a 1-2mm perimeter with maybe a single X inside (~5% infill etc) to increase strength between the 4 corners. Since we know the "print one at a time" dimensions of the extruder we can easily build a relatively tall tower knowing that it won't be an impact to the print. Once the purge tower is sufficiently tall you can then purge into the bucket and print another layer for the tower, then move back to the print.

I realize this isn't easy, but my least favorite aspect of the 3 right now is the W&P

Ghostkeeper, 2016-10-28:
How would we prevent the bucket from falling over when you wipe against its border at each layer?

Or are you proposing to move the build plate up and down for this wipe, at each layer?

We have plans to make the wipe tower hollow and wipe on the inside, which might also help a bit since the build plate adhesion has more surface area then.

MTKelly, 2016-10-28:
My vision is basically the latter. Lower the print bucket and drop the nozzle in so that the oozing bit it below the top layer line, Purge a little so it's more molten then print the next layer on the tower. Then resume back to the print. Takes a bit longer but you could fine tune just how much you need to print above bed surface so you're not purging into existing plastic. 

Have you seen how Makerware does their wipe and prime shields? I believe they purge into the end diamonds before printing the oscillating bit.  Those can fall over if they don't get good adhesion, but far less frequently than W&P towers

maukcc, 2017-01-12:
Another good thing would be to not have a fixed position for the tower, but to have it near the object/brim/raft/skirt.

guysoft, 2017-01-22:
Hey,
I am trying to go over the code, and figure out how to change the prime tower shape (so the base would be larger than the top).

(I found this bit of ``_computeDisallowedAreasPrinted``)[https://github.com/Ultimaker/Cura/blob/master/cura/BuildVolume.py#L547]. Is changing the polygon there enough?

guysoft, 2017-01-22:
@Ghostkeeper The git blame is code you wrote, is there a chance you could confirm this is what I need to change?
Is there a way to modify it with a plugin?

Ghostkeeper, 2017-01-22:
You can't change the prime tower shape there, @guysoft. That is the collision area of the prime tower as we have it in the front-end. It is used to draw a shadow on the build plate and to prevent objects from intersecting with the prime tower.

The actual prime tower is generated here: https://github.com/Ultimaker/CuraEngine/blob/152f6e89a8f0c9851911aee1e12d80dff53e14f6/src/PrimeTower.cpp#L61

The two places where the prime tower polygon is generated should coincide, lest the user is able to place meshes intersecting with the prime tower.

Ghostkeeper, 2017-01-22:
Also, a pyramid-shaped prime tower requires more extensive modifications, as that CuraEngine code I linked there is computed once for the base and then copied to all layers that require a prime tower.

guysoft, 2017-01-23:
@Ghostkeeper Ok, so the way the base is copied to all layers is from ``void PrimeTower::addToGcode``?
Because if so we could just regenerate it each time it goes up a layer?
I'm having a heard time following it.

Ghostkeeper, 2017-01-24:
Yes, you understood it correctly. It generates the `PolygonRef` instance once, stores it in the `ground_poly` variable, and then uses that same polygon for every layer.

To have a pyramid-shaped prime tower you'd need to generate a new `PolygonRef` at each layer, for instance by scaling `ground_poly` to make it smaller as the layer number increases.

BagelOrb, 2017-07-23:
How often do you experience the prime tower falling over these days?

guysoft, 2017-07-23:
I haven't done it in a whole, but since I commented I sorta gave up on it, It kept failing.

Ghostkeeper, 2017-08-06:
According to our Materials and Processing team, it still happens a lot with materials that don't adhere well to each other, such as PVA and CPE. They're hoping for us to implement something like concentric infill for the prime tower.

Ghostkeeper, 2017-08-06:
Thinking about this a bit further, I've come to realise that a "prime bucket" would only be feasible for objects lower than the height of the bucket + the gantry height (55mm on UM3). If you print anything higher than that, there's chance of the gantry hitting your print. So at heights larger than ~6cm you won't be able to move the build plate up to wipe in the bucket. At heights smaller than ~6cm the chance of the prime tower toppling should be quite slim...

closed 2018-11-26 by Appesteijn

guysoft, 2018-11-26:
I assume this is closed because it was one in Cura 3 (commenting for completeness)