Ultimaker/Cura #1089: Weight estimation non showing with UM2

zippy1978, 2016-10-28:
Hi,

I noticed that Cura 2.3 weight estimation does not work when using an Ultimaker 2.
Is there anything to fix that ?

Regards,
Gilles

Ghostkeeper, 2016-10-29:
I could reproduce this. It looks like a new bug!

fieldOfView, 2016-10-29:
There is some logic to the bug though. For the UM2, the material is set on the printer instead of in Cura. So Cura does not know the material, or the density of the material, and consequently it can not calculate the weight of the determined length of material.

I *think* 15.04 just assumed PLA for the UM2 weight calculation.

nallath, 2016-10-31:
Yea it did. It used PLA as defaults.

Ghostkeeper, 2016-11-04:
This was just classified by Ultimaker as a Won't Fix. So it's going to still say 0g, but please ignore it.

closed 2016-11-04 by Ghostkeeper

Dilbert1, 2017-04-30:
Classified by Ultimaker as a Won't Fix?
...?
Is it sooo difficult to multiply the filament length with the weight per length?
Plugins like "Print Cost Calculator" dont work any longer since new api.
Or ist there an other alternative to get the weight without to calculate it myself?

fieldOfView, 2017-04-30:
> Is it sooo difficult to multiply the filament length with the weight per length?

Yes, it is. Because you select the material on the printer, Cura does not know the density of the material used. Without the density it cannot calculate the weight of the used material.

Ghostkeeper, 2017-05-05:
We could assume PLA again, but that is somewhat misleading.