Ultimaker/Cura #1090: Save files as X3D

meghan81, 2016-10-28:
Cura can now import X3D, but I'd like to be able to save as an X3D file, and also save profile metadata, similar to [this request](https://github.com/Ultimaker/Cura/issues/1086) for 3MF, since X3D is also XML-based.
@nickthetait

nallath, 2016-10-31:
We're not going to also put time in making x3d be able to save & load profile data. I personally also feel that 3mf is way more suited for doing this. I think it's best to go for one file type that does this well, instead of two halfhearted implementations

awhiemstra, 2016-10-31:
I think we should set it up in a generic enough way that it is not only tied to 3MF but can also support other formats writing the data. We ourselves can then focus on getting it properly implemented in 3MF, while someone else can add the support to X3D.

nickthetait, 2016-11-01:
Agree with awhiemstra, the core functionality can be extracted and made to work with more than one output format. X3D export is a feature that Aleph Objects is interested in implementing further down the road.

Scene exporting would seriously improve reproducibility and ease testing for both users and software developers. Basically it will allow you to save a print job just before it is sliced. Then when your engine/settings/design model have improvements you can test these independently and worry free.

nallath, 2016-11-01:
I'm currently working on fixing the 3MF loading & saving, which is the stepping stone to complete workspace saving.

Glad to see that everyone has the same priorities regarding this :)

meghan81, 2016-11-01:
Thanks @nallath, I'm looking forward to seeing that. X3D has added value beyond reproducibility and in comparison to 3MF, so I think there will be a good return on the time investment to set profile exporting to support more than one format, as suggested by @awhiemstra and @nickthetait.

fieldOfView, 2016-11-01:
>  X3D has added value beyond reproducibility and in comparison to 3MF

Out of curiosity, could you describe the added value of saving profile data in comparison to 3MF?

meghan81, 2016-11-01:
Saving profile data in X3D isn't more valuable than saving profile data in 3MF. However, X3D files aren't zipped up like 3MF, so they don't need to be extracted in order to use/view them. They can also be displayed in a web-browser easily, without a plugin (just HTML+JS), and can be used in VR/AR devices. The ability to have *one* file that includes geometry and metadata, and which can be 3D printed and visualized in various applications would be a helpful to the 3D community.

FWIW, 3MF is based on X3D.

ianpaschal, 2018-02-27:
This hasn't been touched in 1 year+ and isn't broken so I'm going to consider it closed.

closed 2018-02-27 by ianpaschal