Ultimaker/Cura #1091: Cura 2.3.0 change path

BondKing, 2016-10-29:
I would like to change the path of the configuration folder.But I can't find out how to change it! It's not in the settings.

thopiekar, 2016-10-29:
As far as I know there is neither a system variable nor a option you can pass to Cura to change the config path. If you have experiences in Python programming, you can add such an option or variable for this.

Ghostkeeper, 2016-10-30:
Well, not really, since the path to the configuration folder would be stored in the configuration folder if you use Uranium's normal way of creating configuration.

You would need to make a registry entry to change that, but that has all sorts of problems for other platforms than Windows.

The place where we store configuration is hardcoded here: https://github.com/Ultimaker/Uranium/blob/master/UM/Resources.py#L262

BondKing, 2016-10-30:
Appreciate for all your replies! But I can find neither a folder named 'UM' nor a file named 'Resources.py' in my computer.
I do remember there was a 'resources.py' when I was using version 15.04.3.
It's really important for me, cuz Cura always create the configuration folder in the path which I have already deleted.
I tried to uninstall Cura, but after rebooting and reinstalling Cura, the configuration folder shows up in the same path!
 It's driving me crazy!!!

fieldOfView, 2016-10-30:
Please first tell us why you want to change the path of the configuration? Perhaps we can give you a better solution to the problem you are trying to solve.

BondKing, 2016-10-30:
The configuration folder is in the path which I used to install OrCAD, but now I have already unstall OrCAD. I would like to keep my computer in order.For example, 'D:\software\AD' is where I install Altium Designer, so, it's really strange that there is a folder named 'OrCAD' but actually it only contains Cura configuration folder. And it's really annoying me!
Perhaps uninstall Cura clearly then reinstall is a good way.But I don't know how to uninstall clearly. Maybe I need to unstall Visual C++ at the same time? Who knows?

fieldOfView, 2016-10-30:
Are you saying that there Cura creates a folder named "cura" inside D:\software\OrCAD ? In that case, your Windows installation has much bigger issues than just that cura folder. Cura creates that folder in what Windows tells it is the user profile directory.

BondKing, 2016-10-30:
Yes. Every time I install Crua(no matter version 2.3 or 15.04.3), it sure installs in the path I ask(D:\Softerware\Cura), but when I run the Cura for the first time, it will create it's configuration folder at 'D:\Code\OrCAD\AppData\Local\cura'. What kind of issues I may face, and how can I solve it?

fieldOfView, 2016-10-30:
Hmm, I don't know how to say this... your Windows installation is messed up beyond what I can help you with remotely. Cura is doing the right thing, your Windows is just giving it weird information. Something must have gone wrong when you configured it to write certain bits to your D: drive instead of to its default location on C:.

Cura would normally create its configuration folder in C:\Users\[your username]\AppData\Local\cura. The C:\Users\[your username]\AppData\Local folder is the "proper" place for an application to store their configuration according to Microsoft. Somehow your system instructs Cura to use D:\Code\OrCAD instead. The problem is that it doesn't just tell Cura to do this; it probably tells all applications to store their data there.

Either way, this is an issue with your OS, beyond the control of Cura. I cannot really help you with that.

BondKing, 2016-10-30:
Hmm..I need to say,I tried to change the AppData path using regedit just now and I succeed delete a few of my software by accident...It's not the first time I gave F words to my computer...At last, I decided to reinstall my Win10...And maybe I need to be more carefully :P 
Anyway, Thanks a lot! You guys are the best!

Ghostkeeper, 2016-10-31:
It appears that something changed your Windows "home" directory to that OrCAD...

You might be able to fix this using a [symbolic link](http://www.howtogeek.com/howto/16226/complete-guide-to-symbolic-links-symlinks-on-windows-or-linux/)...?

BondKing, 2016-10-31:
Many thanks! But it seems that I have to reinstall my windows…Hope everything will go right after the reinstalling. I changed the appdata path through regedit…Never do that again

fieldOfView, 2016-10-31:
> I changed the appdata path through regedit…Never do that again

Good plan. Don't touch regedit, unless you know exactly what you are doing

closed 2017-07-27 by nallath