Ultimaker/Cura #1092: Questions while working on a new plugin

thopiekar, 2016-10-30:
While having access on the campus to use CATIA, I decided to begin to code a plugin for it.
Inspired by a script I found on the net I found a way to convert files via COM API from format "XY" to stl.

So my idea here in this plugin is to provide all file types which are supported by CATIA and export them as an STL file and open it with our STLreader.

1. Is there a way to call the STLreader from my plugin?
2. I opened an CATpart files but don't see any debug message in the logs. Eg. neither https://github.com/thopiekar/CuraCatiaConnectorPlugin/blob/master/CATpartReader.py#L39 nor https://github.com/thopiekar/CuraCatiaConnectorPlugin/blob/master/CATpartReader.py#L46.

Additionally, I added test code which works great. Running CATIA before running the script is not needed, it just takes a little but more time. After the first run the code works much faster.

* https://github.com/thopiekar/CuraCatiaConnectorPlugin

fieldOfView, 2016-10-30:
Couldn't your plugin simply import and inherit from the STLReader class instead of MeshReader?

fieldOfView, 2016-10-30:
See https://github.com/Ultimaker/Uranium/blob/master/plugins/FileHandlers/STLWriter/__init__.py for handling multiple extensions

thopiekar, 2016-10-30:
@fieldOfView Well, the STLReader is a plugin and thus I personally think import won't work here...
... except I ask our plugin system for the paths where plugins are installed, add it to sys.path and then import the STLReader.

Ah! Great, thanks!

fieldOfView, 2016-10-30:
```import plugins.FileHandlers.STLReader as STLReader``` does not work?

thopiekar, 2016-10-30:
Not here. I'm sure it is because the "plugin" directory is no package.
But I took a look at UM.PluginRegistry in the meantime. Looks like I can the loaded class from there..

thopiekar, 2016-10-30:
I'm still coding but I guess I can do something like that:
```
if "STLReader" in PluginRegistry.getActivePlugins():
            print("STLReader not loaded!")
            return None
```

Then use `UM.PluginRegistry.getPluginObject("STLReader")`..

fieldOfView, 2016-10-30:
Strange, this works fine for me:
```
>>> from plugins.FileHandlers.STLReader import STLReader
>>> stl = STLReader.STLReader()
>>> stl
<plugins.FileHandlers.STLReader.STLReader.STLReader object at 0x0000000002350080>
```
I have my Uranium checkout in my PYTHONPATH, but that's all.

thopiekar, 2016-10-30:
Maybe I mistyped before 🙄 
Will give it a retry in short...

By the way, is that really the correct way? What if someone decides (don't ask for the reasons) not to use the STLReader? Because it is buggy, whatever..
Wouldn't it be better if I look for STLReader first and then use it? And if it is not loaded tell the user to enable it?
Probably it isn't possible at the moment to disable the STLReader, but I just want to code the plugin the "correct" way. 😉

thopiekar, 2016-10-30:
@fieldOfView And about your tip on STLWriter.. Multiple export is supported but no import it seems 😕

thopiekar, 2016-10-30:
Ah! They are, but in a different way than done in *Writer plugins..

thopiekar, 2016-10-30:
Wondered the whole time why `read()` was never called..
Looked into the Uranium code and guess what: It expects the file extension in lower case 😭 
```self._supported_extensions = [".CATPart".lower()]```

thopiekar, 2016-10-30:
@fieldOfView Tried to import STLReader like you've mentioned above, but using a Cura installation on Windows I get:
```
2016-10-30 19:31:04,672 - ERROR - C:/Users/thopi/Documents/GitHub/cura-build/build/inst/lib/python3.5/site-packages/UM\PluginRegistry.py (_findPlugin [268]): Import error loading module CATpartReader: No module named 'plugins.FileHandlers'
2016-10-30 19:31:04,672 - ERROR - C:/Users/thopi/Documents/GitHub/cura-build/build/inst/lib/python3.5/site-packages/UM\PluginRegistry.py (_populateMetaData [238]): Could not find plugin CATpartReader
2016-10-30 19:31:04,678 - ERROR - C:/Users/thopi/Documents/GitHub/cura-build/build/inst/lib/python3.5/site-packages/UM\PluginRegistry.py (_findPlugin [268]): Import error loading module CATpartReader: No module named 'plugins.FileHandlers'
```

thopiekar, 2016-10-30:
Ok, except of CATProducts everything works.
Don't have more details about the API, so I can just experiment on this using macros on the net.

Thank you @fieldOfView for your help 👍

closed 2016-10-30 by thopiekar

awhiemstra, 2016-10-31:
> import plugins.FileHandlers.STLReader as STLReader does not work?

Note that you should never do this, as this will cause dependencies on file locations that are not true on all platforms. For example, when using an installed package on Linux the import will fail because the plugin directory is not in PYTHONPATH. Likewise on Windows as @thopiekar already mentioned.

The right way to use the STL reader is to ask MeshFileHandler either to get the STLReader directly or to just ask it to import a file. Its API could do with a little extension to match that of the writer plugins, but other than that it is the right class to use.

thopiekar, 2016-10-31:
I've done the trick by asking the PluginRegistry whether STLReader is loaded and then asked it again for the plugins object, so I can call .read() on it.

Works great so far.

awhiemstra, 2016-10-31:
That works too, it's just a little more low-level than using MeshFileHandler.

Note that we do not really support dependencies in plugins at this time, so depending on when you do this you run the risk of your plugin being loaded before the STLReader and then failing.

thopiekar, 2016-10-31:
That's why I'm asking for STLReader when read() is called. At least I assume that STLReader should be loaded at this time...

As a reference: 
* https://github.com/thopiekar/CuraCatiaConnectorPlugin/blob/9da62711d6741e0511868e8fb3ae06388a4c24df/CATpartReader.py#L24
* https://github.com/thopiekar/CuraCatiaConnectorPlugin/blob/9da62711d6741e0511868e8fb3ae06388a4c24df/CATpartReader.py#L69

awhiemstra, 2016-10-31:
Yeah, that should be fine. You want to make sure you call it after the plugin loading process is done.

thopiekar, 2016-10-31:
:ok_hand:

fieldOfView, 2016-10-31:
> Note that you should never do this, as this will cause dependencies on file locations that are not true on all platforms.

Point taken, thanks for setting me straight.