Ultimaker/Cura #1094: Crash on Startup using Ubuntu 16.10

argo2445, 2016-10-31:
I use the ppa version from 'http://ppa.launchpad.net/thopiekar/cura/ubuntu'. 
It does not start any more after upgrade. 

I already deleted all cura folders in .cache, .config and .local.

Output on Startup:
`2016-10-31 12:20:49,850 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ConsoleLogger
2016-10-31 12:20:49,857 - INFO - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (__init__ [40]): Found CuraEngine at: /usr/bin/CuraEngine
2016-10-31 12:20:49,858 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraEngineBackend
2016-10-31 12:20:49,861 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin WireframeView
2016-10-31 12:20:49,863 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SimpleView
2016-10-31 12:20:49,873 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RotateTool
2016-10-31 12:20:49,876 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SelectionTool
2016-10-31 12:20:49,885 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ScaleTool
2016-10-31 12:20:49,889 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin TranslateTool
2016-10-31 12:20:49,893 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CameraTool
2016-10-31 12:20:49,898 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MirrorTool
2016-10-31 12:20:49,902 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LocalFileOutputDevice
2016-10-31 12:20:49,902 - WARNING - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin ConsoleLogger was already loaded
2016-10-31 12:20:49,904 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin FileLogger
2016-10-31 12:20:49,907 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLWriter
2016-10-31 12:20:49,909 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJReader
2016-10-31 12:20:49,912 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJWriter
2016-10-31 12:20:49,915 - WARNING - /usr/lib/uranium/plugins/FileHandlers/STLReader/STLReader.py (<module> [5]): Could not find numpy-stl, falling back to slower code.
2016-10-31 12:20:49,916 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLReader
2016-10-31 12:20:49,922 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFWriter
2016-10-31 12:20:49,924 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileWriter
2016-10-31 12:20:49,953 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin Doodle3D uses an incompatible API version, ignoring
2016-10-31 12:20:49,956 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LayerView
2016-10-31 12:20:49,957 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LegacyProfileReader
2016-10-31 12:20:49,959 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ImageReader
2016-10-31 12:20:49,961 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeProfileReader
2016-10-31 12:20:49,964 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MachineSettingsAction
2016-10-31 12:20:49,966 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin X3DReader
2016-10-31 12:20:49,967 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin AutoSave
2016-10-31 12:20:49,969 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SolidView
2016-10-31 12:20:49,973 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin VersionUpgrade21to22
2016-10-31 12:20:49,975 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XRayView
2016-10-31 12:20:49,982 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin USBPrinting
2016-10-31 12:20:49,985 - WARNING - .../UM/Preferences.py (getValue [70]): Tried to get the value of non-existing setting cura/active_mode.
2016-10-31 12:20:49,986 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin PerObjectSettingsTool
2016-10-31 12:20:49,995 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XmlMaterialProfile
2016-10-31 12:20:49,997 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ChangeLogPlugin
2016-10-31 12:20:49,997 - WARNING - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin CuraEngineBackend was already loaded
2016-10-31 12:20:49,999 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFReader
2016-10-31 12:20:50,002 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RemovableDriveOutputDevice
2016-10-31 12:20:50,005 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileReader
2016-10-31 12:20:50,010 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin UltimakerMachineActions
2016-10-31 12:20:50,014 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SliceInfoPlugin
2016-10-31 12:20:50,016 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeWriter
2016-10-31 12:20:50,017 - INFO - .../UM/Qt/QtApplication.py (__init__ [39]): Command line arguments: {'disable-textures': False, 'debug-mode': False, 'external-backend': False, 'file': []}
2016-10-31 12:20:50,021 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Attempting to connect to /dev/ttyACM2
2016-10-31 12:20:50,022 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Attempting to connect to /dev/ttyACM1
2016-10-31 12:20:50,023 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Attempting to connect to /dev/ttyACM0
2016-10-31 12:20:50,024 - INFO - .../UM/VersionUpgradeManager.py (upgrade [98]): Looking for old configuration files to upgrade.
2016-10-31 12:20:52,211 - DEBUG - .../UM/Backend/Backend.py (_logSocketState [165]): Socket state changed to Listening
2016-10-31 12:20:52,216 - INFO - .../UM/Backend/Backend.py (startEngine [55]): Started engine process: /usr/bin/CuraEngine
2016-10-31 12:20:52,216 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] Calling engine with: ['/usr/bin/CuraEngine', 'connect', '127.0.0.1:49674', '-j', '/usr/share/cura/resources/definitions/fdmprinter.def.json', '']
2016-10-31 12:20:52,218 - WARNING - .../UM/OutputDevice/OutputDeviceManager.py (removeOutputDevice [123]): Could not find output device with id /dev/ttyACM2 to remove
2016-10-31 12:20:52,218 - WARNING - .../UM/OutputDevice/OutputDeviceManager.py (removeOutputDevice [123]): Could not find output device with id /dev/ttyACM1 to remove
2016-10-31 12:20:52,218 - WARNING - .../UM/OutputDevice/OutputDeviceManager.py (removeOutputDevice [123]): Could not find output device with id /dev/ttyACM0 to remove
2016-10-31 12:20:52,218 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] 
2016-10-31 12:20:52,219 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] Cura_SteamEngine version DEV
2016-10-31 12:20:52,220 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] Copyright (C) 2014 David Braam
2016-10-31 12:20:52,221 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] 
2016-10-31 12:20:52,222 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] This program is free software: you can redistribute it and/or modify
2016-10-31 12:20:52,223 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] it under the terms of the GNU Affero General Public License as published by
2016-10-31 12:20:52,227 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] the Free Software Foundation, either version 3 of the License, or
2016-10-31 12:20:52,228 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] (at your option) any later version.
2016-10-31 12:20:52,228 - DEBUG - .../UM/Backend/Backend.py (_backendLog [88]): [Backend] 
[... 8 lines cut ...]
2016-10-31 12:20:52,517 - DEBUG - /usr/lib/cura/plugins/MachineSettingsAction/MachineSettingsAction.py (_onContainerAdded [40]): Not attaching MachineSettingsAction to ultimaker2_extended_plus; Machines that use variants are not supported
2016-10-31 12:20:52,734 - INFO - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Could not establish connection on /dev/ttyACM1: 'Timeout'. Device is not arduino based.
[... cut ...]
2016-10-31 12:20:58,303 - DEBUG - .../UM/Qt/Bindings/Theme.py (load [118]): Loaded theme /usr/share/cura/resources/themes/cura
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
2016-10-31 12:20:58,406 - WARNING - .../cura/Settings/ContainerManager.py (getContainerMetaDataEntry [171]): Could not get metadata of container  because it was not found.
Speicherzugriffsfehler (Speicherabzug geschrieben)
`

thopiekar, 2016-11-01:
@argo2445 Could you please do an system update? I just want to be sure we both use the same packages here.

argo2445, 2016-11-01:
@thopiekar I just updated the system. The result semms to be the same. 
Output:
`2016-11-01 11:28:19,828 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ConsoleLogger
2016-11-01 11:28:19,956 - INFO - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (__init__ [40]): Found CuraEngine at: /usr/bin/CuraEngine
2016-11-01 11:28:19,958 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraEngineBackend
2016-11-01 11:28:19,982 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin WireframeView
2016-11-01 11:28:19,985 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SimpleView
2016-11-01 11:28:20,141 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RotateTool
2016-11-01 11:28:20,144 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SelectionTool
2016-11-01 11:28:20,152 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ScaleTool
2016-11-01 11:28:20,158 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin TranslateTool
2016-11-01 11:28:20,164 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CameraTool
2016-11-01 11:28:20,170 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MirrorTool
2016-11-01 11:28:20,186 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LocalFileOutputDevice
2016-11-01 11:28:20,186 - WARNING - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin ConsoleLogger was already loaded
2016-11-01 11:28:20,188 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin FileLogger
2016-11-01 11:28:20,229 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLWriter
2016-11-01 11:28:20,232 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJReader
2016-11-01 11:28:20,237 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJWriter
2016-11-01 11:28:20,239 - WARNING - /usr/lib/uranium/plugins/FileHandlers/STLReader/STLReader.py (<module> [5]): Could not find numpy-stl, falling back to slower code.
2016-11-01 11:28:20,241 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLReader
2016-11-01 11:28:20,247 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFWriter
2016-11-01 11:28:20,256 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileWriter
2016-11-01 11:28:20,301 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin Doodle3D uses an incompatible API version, ignoring
2016-11-01 11:28:20,337 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LayerView
2016-11-01 11:28:20,339 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LegacyProfileReader
2016-11-01 11:28:20,342 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ImageReader
2016-11-01 11:28:20,343 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeProfileReader
2016-11-01 11:28:20,360 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MachineSettingsAction
2016-11-01 11:28:20,386 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin X3DReader
2016-11-01 11:28:20,388 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin AutoSave
2016-11-01 11:28:20,390 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SolidView
2016-11-01 11:28:20,408 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin VersionUpgrade21to22
2016-11-01 11:28:20,410 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XRayView
2016-11-01 11:28:20,475 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin USBPrinting
2016-11-01 11:28:20,498 - WARNING - .../UM/Preferences.py (getValue [70]): Tried to get the value of non-existing setting cura/active_mode.
2016-11-01 11:28:20,498 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin PerObjectSettingsTool
2016-11-01 11:28:20,510 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XmlMaterialProfile
2016-11-01 11:28:20,520 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ChangeLogPlugin
2016-11-01 11:28:20,520 - WARNING - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin CuraEngineBackend was already loaded
2016-11-01 11:28:20,534 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFReader
2016-11-01 11:28:20,538 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RemovableDriveOutputDevice
2016-11-01 11:28:20,542 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileReader
2016-11-01 11:28:20,548 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin UltimakerMachineActions
2016-11-01 11:28:20,553 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SliceInfoPlugin
2016-11-01 11:28:20,555 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeWriter
2016-11-01 11:28:20,556 - INFO - .../UM/Qt/QtApplication.py (__init__ [39]): Command line arguments: {'file': [], 'debug-mode': False, 'external-backend': False, 'disable-textures': False}
2016-11-01 11:28:20,593 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Attempting to connect to /dev/ttyACM2
2016-11-01 11:28:20,597 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Attempting to connect to /dev/ttyACM1
[... cut ...]
2016-11-01 11:28:31,929 - DEBUG - .../UM/Qt/Bindings/Theme.py (load [118]): Loaded theme /usr/share/cura/resources/themes/cura
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
2016-11-01 11:28:32,074 - WARNING - .../cura/Settings/ContainerManager.py (getContainerMetaDataEntry [171]): Could not get metadata of container  because it was not found.
Speicherzugriffsfehler (Speicherabzug geschrieben)`

Ghostkeeper, 2016-11-01:
Speicherzugriffsfehler means it's a memory error. A segfault, it seems.

thopiekar, 2016-11-01:
I always had this message here since we've started to work on 2.0.x.
At the beginning it wasn't appearing every time, but now it appears every time I close Cura.
So in my opinion this seg. fault isn't the problem here.

Prepared my master PPA to work with the debug version of Python. I'll try to experiment on this later.

But back to the problem here. What I personally miss here is the message about OpenGL being loaded..
@argo2445 Could you please take a terminal and type in: ```glxinfo | grep -i "OpenGL version string:"```

argo2445, 2016-11-01:
$ glxinfo | grep -i "OpenGL version string:"
OpenGL version string: 3.0 Mesa 12.0.3

thopiekar, 2016-11-01:
And ```glxinfo | grep -i "opengl"```?

argo2445, 2016-11-01:
$ glxinfo | grep -i "opengl"
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) Ivybridge Mobile 
OpenGL core profile version string: 3.3 (Core Profile) Mesa 12.0.3
OpenGL core profile shading language version string: 3.30
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 12.0.3
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 12.0.3
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00
OpenGL ES profile extensions:

thopiekar, 2016-11-01:
Great, thanks! Intel GPU and opensource drivers :+1: 

So let's run Cura without your drivers, but using software rendering by: ```LIBGL_ALWAYS_SOFTWARE=1 cura```

argo2445, 2016-11-01:
It is crashing again. Seems to be the same :(. 
Output:
`$ LIBGL_ALWAYS_SOFTWARE=1 cura
2016-11-01 12:01:28,745 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ConsoleLogger
2016-11-01 12:01:28,755 - INFO - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (__init__ [40]): Found CuraEngine at: /usr/bin/CuraEngine
2016-11-01 12:01:28,755 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraEngineBackend
2016-11-01 12:01:28,760 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin WireframeView
2016-11-01 12:01:28,762 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SimpleView
2016-11-01 12:01:28,772 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RotateTool
2016-11-01 12:01:28,777 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SelectionTool
2016-11-01 12:01:28,785 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ScaleTool
2016-11-01 12:01:28,791 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin TranslateTool
2016-11-01 12:01:28,795 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CameraTool
2016-11-01 12:01:28,800 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MirrorTool
2016-11-01 12:01:28,803 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LocalFileOutputDevice
2016-11-01 12:01:28,804 - WARNING - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin ConsoleLogger was already loaded
2016-11-01 12:01:28,806 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin FileLogger
2016-11-01 12:01:28,810 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLWriter
2016-11-01 12:01:28,813 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJReader
2016-11-01 12:01:28,816 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJWriter
2016-11-01 12:01:28,818 - WARNING - /usr/lib/uranium/plugins/FileHandlers/STLReader/STLReader.py (<module> [5]): Could not find numpy-stl, falling back to slower code.
2016-11-01 12:01:28,819 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLReader
2016-11-01 12:01:28,826 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFWriter
2016-11-01 12:01:28,828 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileWriter
2016-11-01 12:01:28,857 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin Doodle3D uses an incompatible API version, ignoring
2016-11-01 12:01:28,860 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LayerView
2016-11-01 12:01:28,861 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LegacyProfileReader
2016-11-01 12:01:28,863 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ImageReader
2016-11-01 12:01:28,865 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeProfileReader
2016-11-01 12:01:28,868 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MachineSettingsAction
2016-11-01 12:01:28,871 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin X3DReader
2016-11-01 12:01:28,873 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin AutoSave
2016-11-01 12:01:28,875 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SolidView
2016-11-01 12:01:28,877 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin VersionUpgrade21to22
2016-11-01 12:01:28,879 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XRayView
2016-11-01 12:01:28,886 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin USBPrinting
2016-11-01 12:01:28,889 - WARNING - .../UM/Preferences.py (getValue [70]): Tried to get the value of non-existing setting cura/active_mode.
2016-11-01 12:01:28,889 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin PerObjectSettingsTool
2016-11-01 12:01:28,899 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XmlMaterialProfile
2016-11-01 12:01:28,902 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ChangeLogPlugin
2016-11-01 12:01:28,903 - WARNING - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin CuraEngineBackend was already loaded
2016-11-01 12:01:28,905 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFReader
2016-11-01 12:01:28,907 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RemovableDriveOutputDevice
2016-11-01 12:01:28,910 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileReader
2016-11-01 12:01:28,914 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin UltimakerMachineActions
2016-11-01 12:01:28,919 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SliceInfoPlugin
2016-11-01 12:01:28,921 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeWriter
2016-11-01 12:01:28,922 - INFO - .../UM/Qt/QtApplication.py (__init__ [39]): Command line arguments: {'disable-textures': False, 'file': [], 'external-backend': False, 'debug-mode': False}
2016-11-01 12:01:28,925 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Attempting to connect to /dev/ttyACM2
2016-11-01 12:01:28,927 - DEBUG - /usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDevice.py (_connect [298]): Attempting to connect to /dev/ttyACM1
[... cut ...]
2016-11-01 12:01:37,219 - DEBUG - .../UM/Qt/Bindings/Theme.py (load [118]): Loaded theme /usr/share/cura/resources/themes/cura
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
2016-11-01 12:01:37,330 - WARNING - .../cura/Settings/ContainerManager.py (getContainerMetaDataEntry [171]): Could not get metadata of container  because it was not found.
Speicherzugriffsfehler (Speicherabzug geschrieben)`

thopiekar, 2016-11-01:
@Ghostkeeper, just a little update. Now the segmentation fault is away again..
```
thopiekar@hp:~$ LIBGL_ALWAYS_SOFTWARE=1 cura
2016-11-01 12:05:16,560 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ConsoleLogger
2016-11-01 12:05:16,696 - INFO - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (__init__ [40]): Found CuraEngine at: /usr/bin/CuraEngine
2016-11-01 12:05:16,711 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraEngineBackend
2016-11-01 12:05:16,752 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LocalFileOutputDevice
2016-11-01 12:05:16,753 - WARNING - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin ConsoleLogger was already loaded
2016-11-01 12:05:16,758 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin FileLogger
2016-11-01 12:05:16,824 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin TranslateTool
2016-11-01 12:05:16,849 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ScaleTool
2016-11-01 12:05:16,855 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CameraTool
2016-11-01 12:05:16,898 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MirrorTool
2016-11-01 12:05:16,912 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RotateTool
2016-11-01 12:05:16,917 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SelectionTool
2016-11-01 12:05:16,945 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJWriter
2016-11-01 12:05:17,021 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFWriter                                                                                                                                                                   
2016-11-01 12:05:17,024 - WARNING - /usr/lib/uranium/plugins/FileHandlers/STLReader/STLReader.py (<module> [5]): Could not find numpy-stl, falling back to slower code.                                                                                                
2016-11-01 12:05:17,026 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLReader                                                                                                                                                                   
2016-11-01 12:05:17,033 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLWriter                                                                                                                                                                   
2016-11-01 12:05:17,040 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJReader                                                                                                                                                                   
2016-11-01 12:05:17,045 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin WireframeView                                                                                                                                                               
2016-11-01 12:05:17,050 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SimpleView                                                                                                                                                                  
2016-11-01 12:05:17,079 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin PostProcessingPlugin                                                                                                                                                        
2016-11-01 12:05:17,083 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XRayView                                                                                                                                                                    
2016-11-01 12:05:17,086 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SolidView                                                                                                                                                                   
2016-11-01 12:05:17,101 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin UltimakerMachineActions                                                                                                                                                     
2016-11-01 12:05:17,106 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LegacyProfileReader                                                                                                                                                         
2016-11-01 12:05:17,110 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeWriter                                                                                                                                                                 
2016-11-01 12:05:17,114 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileReader                                                                                                                                                           
2016-11-01 12:05:17,263 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin UM3NetworkPrinting                                                                                                                                                     [... cut ...]
2016-11-01 12:06:24,595 - DEBUG - .../UM/Qt/QtApplication.py (windowClosed [197]): Shutting down cura
2016-11-01 12:06:25,525 - DEBUG - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (_terminate [186]): Attempting to kill the engine process
2016-11-01 12:06:25,526 - DEBUG - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (_terminate [186]): Killing engine process
2016-11-01 12:06:25,527 - DEBUG - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (_terminate [186]): Engine process is killed. Received return code -15
```

thopiekar, 2016-11-01:
@argo2445 I also resetted my configs and I can't reproduce it here.

Could you run `dmesg` and show us `mount`? Just want to be sure there is everything ok with your system.

thopiekar, 2016-11-01:
Additionally, are there any modifications on your Xorg's settings?

argo2445, 2016-11-01:
I did not modify anything about Xorg. I just upgraded the System from Ubuntu 16.04 to 16.10 two weeks ago.

$ dmesg
[    0.000000] microcode: microcode updated early to revision 0x1c, date = 2015-02-26
[    0.000000] Linux version 4.8.0-26-generic (buildd@lgw01-58) (gcc version 6.2.0 20161005 (Ubuntu 6.2.0-5ubuntu12) ) #28-Ubuntu SMP Tue Oct 18 14:39:52 UTC 2016 (Ubuntu 4.8.0-26.28-generic 4.8.0)
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.8.0-26-generic root=UUID=e032e18e-7911-4b53-a3c5-adf2221bfd7d ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'standard' format.
[    0.000000] x86/fpu: Using 'eager' FPU context switches.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009cfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009d000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000020000000-0x00000000201fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020200000-0x0000000040003fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040004000-0x0000000040004fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040005000-0x00000000c8298fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000c8299000-0x00000000dae5efff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000dae5f000-0x00000000daf7efff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000daf7f000-0x00000000daffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000dafff000-0x00000000df9fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f8000000-0x00000000fbffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe800000-0x00000000fe80ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed08000-0x00000000fed08fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffc20000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000021e5fffff] usable
[    0.000000] BIOS-e820: [mem 0x000000021e600000-0x000000021e7fffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: LENOVO 3354DSG/3354DSG, BIOS H3ET72WW(1.09) 02/25/2013
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x21e600 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-BFFFF uncachable
[    0.000000]   C0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 0FFC00000 mask FFFC00000 write-protect
[    0.000000]   1 base 000000000 mask F80000000 write-back
[    0.000000]   2 base 080000000 mask FC0000000 write-back
[    0.000000]   3 base 0C0000000 mask FE0000000 write-back
[    0.000000]   4 base 0DC000000 mask FFC000000 uncachable
[    0.000000]   5 base 0DB000000 mask FFF000000 uncachable
[    0.000000]   6 base 100000000 mask F00000000 write-back
[    0.000000]   7 base 200000000 mask FE0000000 write-back
[    0.000000]   8 base 21F000000 mask FFF000000 uncachable
[    0.000000]   9 base 21E800000 mask FFF800000 uncachable
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT  
[    0.000000] e820: last_pfn = 0xc8299 max_arch_pfn = 0x400000000
[    0.000000] found SMP MP-table at [mem 0x000f0100-0x000f010f] mapped at [ffff9df9c00f0100]
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Base memory trampoline at [ffff9df9c0097000] 97000 size 24576
[    0.000000] BRK [0x57651000, 0x57651fff] PGTABLE
[    0.000000] BRK [0x57652000, 0x57652fff] PGTABLE
[    0.000000] BRK [0x57653000, 0x57653fff] PGTABLE
[    0.000000] BRK [0x57654000, 0x57654fff] PGTABLE
[... 955 lines cut ...]
/dev/sda1 on / type ext4 (rw,relatime,errors=remount-ro,data=ordered)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (rw,mode=755)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,release_agent=/lib/systemd/systemd-cgroups-agent,name=systemd)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
[... cut ...]
mqueue on /dev/mqueue type mqueue (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
hugetlbfs on /dev/hugepages type hugetlbfs (rw,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,relatime)
binfmt_misc on /proc/sys/fs/binfmt_misc type binfmt_misc (rw,relatime)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=775884k,mode=700,uid=1000,gid=1000)
gvfsd-fuse on /run/user/1000/gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)

thopiekar, 2016-11-01:
Ok, your systems seems to be fine. Your dmesg says that your problem here actually takes place in libQt5XcbQpa.so.5.6.1, which is a Qt5 library..

thopiekar, 2016-11-01:
@argo2445 Please run: `gdb -ex r --args python3 /usr/bin/cura`, wait until it crashes and type in "bt".
Finally please send the complete output here.

argo2445, 2016-11-01:
$ gdb -ex r --args python3 /usr/bin/cura
GNU gdb (Ubuntu 7.11.90.20161005-0ubuntu1) 7.11.90.20161005-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from python3...(no debugging symbols found)...done.
Starting program: /usr/bin/python3 /usr/bin/cura
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe1c0e700 (LWP 9750)]
[New Thread 0x7fffe0ff9700 (LWP 9751)]
[New Thread 0x7fffcf2d8700 (LWP 9753)]
[New Thread 0x7fffcead7700 (LWP 9754)]
[New Thread 0x7fffce2d6700 (LWP 9755)]
[New Thread 0x7fffcd1f6700 (LWP 9756)]
[New Thread 0x7fffcc9f5700 (LWP 9757)]
2016-11-01 12:42:37,147 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ConsoleLogger
2016-11-01 12:42:37,157 - INFO - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (__init__ [40]): Found CuraEngine at: /usr/bin/CuraEngine
2016-11-01 12:42:37,158 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraEngineBackend
2016-11-01 12:42:37,162 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin WireframeView
2016-11-01 12:42:37,164 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SimpleView
2016-11-01 12:42:37,175 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RotateTool
2016-11-01 12:42:37,180 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SelectionTool
2016-11-01 12:42:37,189 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ScaleTool
2016-11-01 12:42:37,194 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin TranslateTool
2016-11-01 12:42:37,198 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CameraTool
2016-11-01 12:42:37,204 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MirrorTool
2016-11-01 12:42:37,210 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LocalFileOutputDevice
2016-11-01 12:42:37,210 - WARNING - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin ConsoleLogger was already loaded
2016-11-01 12:42:37,212 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin FileLogger
2016-11-01 12:42:37,216 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLWriter
2016-11-01 12:42:37,219 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJReader
2016-11-01 12:42:37,222 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJWriter
2016-11-01 12:42:37,224 - WARNING - /usr/lib/uranium/plugins/FileHandlers/STLReader/STLReader.py (<module> [5]): Could not find numpy-stl, falling back to slower code.
2016-11-01 12:42:37,226 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLReader
2016-11-01 12:42:37,233 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFWriter
2016-11-01 12:42:37,236 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileWriter
2016-11-01 12:42:37,282 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin Doodle3D uses an incompatible API version, ignoring
2016-11-01 12:42:37,285 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LayerView
2016-11-01 12:42:37,288 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LegacyProfileReader
2016-11-01 12:42:37,291 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ImageReader
2016-11-01 12:42:37,294 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeProfileReader
2016-11-01 12:42:37,298 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MachineSettingsAction
2016-11-01 12:42:37,301 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin X3DReader
2016-11-01 12:42:37,303 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin AutoSave
2016-11-01 12:42:37,305 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SolidView
2016-11-01 12:42:37,308 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin VersionUpgrade21to22
2016-11-01 12:42:37,310 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XRayView
[New Thread 0x7fffb5a98700 (LWP 9760)]
2016-11-01 12:42:37,323 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin USBPrinting
2016-11-01 12:42:37,331 - WARNING - .../UM/Preferences.py (getValue [70]): Tried to get the value of non-existing setting cura/active_mode.
2016-11-01 12:42:37,331 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin PerObjectSettingsTool
2016-11-01 12:42:37,340 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XmlMaterialProfile
2016-11-01 12:42:37,344 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ChangeLogPlugin
[... cut ...]
#37 0x00007fffe8b2f6ae in ?? () from /usr/lib/python3/dist-packages/PyQt5/QtQml.cpython-35m-x86_64-linux-gnu.so
#38 0x00005555556720ff in PyCFunction_Call ()
#39 0x00005555556c58ba in PyEval_EvalFrameEx ()
#40 0x00005555556c5b0f in PyEval_EvalFrameEx ()
#41 0x00005555556c5b0f in PyEval_EvalFrameEx ()
#42 0x00005555556ca3f6 in ?? ()
#43 0x00005555556cb0af in PyEval_EvalCode ()
#44 0x00005555557a4532 in ?? ()
#45 0x00005555557a6a5d in PyRun_FileExFlags ()
#46 0x00005555557a71fe in PyRun_SimpleFileExFlags ()
#47 0x00005555557d7647 in Py_Main ()
#48 0x00005555556677e1 in main ()
(gdb)

thopiekar, 2016-11-01:
Thanks! I see that some debug packages are missing here. Will tell you which are needed here in short.

thopiekar, 2016-11-01:
Please install `qtbase5-dbg` and `python3-pyqt5-dbg`. It also should be wise to install `python3-dbg`, I think.

awhiemstra, 2016-11-01:
Also, qtdeclarative5-dbg

thopiekar, 2016-11-01:
@argo2445 Oh, and please rerun GDB as described before.

argo2445, 2016-11-01:
$ gdb -ex r --args python3 /usr/bin/cura
GNU gdb (Ubuntu 7.11.90.20161005-0ubuntu1) 7.11.90.20161005-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from python3...Reading symbols from /usr/lib/debug/.build-id/b0/38c0c0f8d605ffb7d8fc0f48509284140ff3ae.debug...done.
done.
Starting program: /usr/bin/python3 /usr/bin/cura
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe1c0e700 (LWP 11800)]
[New Thread 0x7fffe0ff9700 (LWP 11801)]
[New Thread 0x7fffcf2d8700 (LWP 11803)]
[New Thread 0x7fffcead7700 (LWP 11804)]
[New Thread 0x7fffce2d6700 (LWP 11805)]
[New Thread 0x7fffcd1f6700 (LWP 11806)]
[New Thread 0x7fffcc9f5700 (LWP 11807)]
2016-11-01 13:06:05,579 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ConsoleLogger
2016-11-01 13:06:05,587 - INFO - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (__init__ [40]): Found CuraEngine at: /usr/bin/CuraEngine
2016-11-01 13:06:05,588 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraEngineBackend
2016-11-01 13:06:05,592 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin WireframeView
2016-11-01 13:06:05,594 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SimpleView
2016-11-01 13:06:05,604 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RotateTool
2016-11-01 13:06:05,607 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SelectionTool
2016-11-01 13:06:05,615 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ScaleTool
2016-11-01 13:06:05,621 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin TranslateTool
2016-11-01 13:06:05,626 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CameraTool
2016-11-01 13:06:05,631 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MirrorTool
2016-11-01 13:06:05,636 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LocalFileOutputDevice
2016-11-01 13:06:05,636 - WARNING - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin ConsoleLogger was already loaded
2016-11-01 13:06:05,639 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin FileLogger
2016-11-01 13:06:05,644 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLWriter
2016-11-01 13:06:05,647 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJReader
2016-11-01 13:06:05,650 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJWriter
2016-11-01 13:06:05,652 - WARNING - /usr/lib/uranium/plugins/FileHandlers/STLReader/STLReader.py (<module> [5]): Could not find numpy-stl, falling back to slower code.
2016-11-01 13:06:05,653 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLReader
2016-11-01 13:06:05,662 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFWriter
2016-11-01 13:06:05,664 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileWriter
2016-11-01 13:06:05,712 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin Doodle3D uses an incompatible API version, ignoring
2016-11-01 13:06:05,717 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LayerView
2016-11-01 13:06:05,719 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LegacyProfileReader
2016-11-01 13:06:05,722 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ImageReader
2016-11-01 13:06:05,724 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeProfileReader
2016-11-01 13:06:05,727 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MachineSettingsAction
2016-11-01 13:06:05,730 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin X3DReader
2016-11-01 13:06:05,731 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin AutoSave
2016-11-01 13:06:05,733 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SolidView
2016-11-01 13:06:05,737 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin VersionUpgrade21to22
2016-11-01 13:06:05,741 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XRayView
[New Thread 0x7fffb5a98700 (LWP 11810)]
2016-11-01 13:06:05,757 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin USBPrinting
2016-11-01 13:06:05,761 - WARNING - .../UM/Preferences.py (getValue [70]): Tried to get the value of non-existing setting cura/active_mode.
2016-11-01 13:06:05,762 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin PerObjectSettingsTool
2016-11-01 13:06:05,773 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XmlMaterialProfile
2016-11-01 13:06:05,777 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ChangeLogPlugin
[... cut ...]
#57 0x00005555557d7647 in run_file (p_cf=0x7fffffffdc90, filename=0x555555c0b280 L"/usr/bin/cura", fp=0x555555d03a60) at ../Modules/main.c:318
#58 Py_Main () at ../Modules/main.c:768
#59 0x00005555556677e1 in main () at ../Programs/python.c:65
#60 0x00007ffff78133f1 in __libc_start_main (main=0x555555667700 <main>, argc=2, argv=0x7fffffffdea8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffde98) at ../csu/libc-start.c:291
#61 0x000055555576f1ba in _start ()
(gdb)

thopiekar, 2016-11-01:
@argo2445 Thanks! Just added the missing ``` at the beginning and end for you :wink:

awhiemstra, 2016-11-01:
> #0 QDBusPlatformMenu::insertMenuItem (this=0x555559a3d430, menuItem=0x0, before=) at dbusmenu/qdbusplatformmenu.cpp:187
> #1 0x00007fffe4e6ec2d in QDBusMenuBar::insertMenu (this=0x555559a3d530, menu=, before=0x0) at dbusmenu/qdbusmenubar.cpp:102

Well there's your issue... sounds like this is an upstream issue.

argo2445, 2016-11-01:
Thank you very much @thopiekar and @awhiemstra! What to do now? 
Can I fix this on my own? Do I need to wait for a fix?

thopiekar, 2016-11-01:
Sadly, yes. I don't see any proposed updates on the Qt5 package in Ubuntu's repository.
Can't say when it will land in their repo at all :disappointed:

argo2445, 2016-11-01:
Okay, I will wait for a fix :(.
Thank you very much for your help!

awhiemstra, 2016-11-01:
In the meantime, it might be possible to find a workaround. If you don't mind breaking global menu bar on some applications, uninstalling libdbusmenu-qt5 might work for example.

thopiekar, 2016-11-01:
@argo2445 Be careful, it might uninstall your desktop. My KDE desktop depends on it for example:
```
thopiekar@home:~$ sudo apt remove libdbusmenu-qt5
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Die folgenden Pakete wurden automatisch installiert und werden nicht mehr benötigt:
  akonadi-backend-mysql akonadi-server dvgrab ieee-data kaccounts-providers kactivities-bin kactivitymanagerd kamera kate5-data kde-cli-tools-data kde-style-oxygen-qt5 kde-telepathy-data kde-telepathy-kaccounts kdeconnect kded5
  kdeedu-data kdegraphics-thumbnailers kdenlive-data kdepim-doc kdepimlibs-data kdepimlibs-kio-plugins kdeplasma-addons-data kdoctools5 kf5-kdepim-apps-libs-data kf5-messagelib-data khotkeys-data kinit konversation-data kross
  ksysguard-data ksysguardd ktexteditor-data ktexteditor-katepart kwin-data libcln6 libdolphinvcs5 libfakekey0 libgps22 libgrantlee-templates5 libgrantlee-textdocument5 libkaccounts1 libkf5activities5 libkf5activitiesstats1
  libkf5akonadiagentbase5 libkf5akonadicontact5 libkf5akonadicore-bin libkf5akonadicore5 libkf5akonadimime5 libkf5akonadinotes5 libkf5akonadiprivate5 libkf5akonadisearchdebug5 libkf5akonadisearchpim5 libkf5akonadiwidgets5
  libkf5baloowidgets-bin libkf5baloowidgets5 libkf5bluezqt-data libkf5bluezqt6 libkf5calendarevents5 libkf5calendarsupport-data libkf5contacts-data libkf5contacts5 libkf5dnssd-data libkf5dnssd5 libkf5emoticons-bin libkf5emoticons-data
  libkf5emoticons5 libkf5eventviews-data libkf5followupreminder5 libkf5gapi-data libkf5gapicontacts5 libkf5gapicore5 libkf5gpgmepp-pthread5 libkf5gpgmepp5 libkf5grantleetheme-plugins libkf5holidays-data libkf5holidays5
  libkf5identitymanagement5abi1 libkf5imap5 libkf5itemmodels5 libkf5js5 libkf5jsembed-data libkf5jsembed5 libkf5kdcraw5 libkf5kdelibs4support-data libkf5kdepimdbusinterfaces5 libkf5kdgantt2-5 libkf5kexiv2-15.0.0 libkf5khtml-data
  libkf5kiogui5 libkf5kiontlm5 libkf5kipi-bin libkf5kipi31.0.0 libkf5kmanagesieve5 libkf5kontactinterface-data libkf5kontactinterface5 libkf5krosscore5 libkf5krossui5 libkf5ksieve-data libkf5ksieve5 libkf5ldap5 libkf5libkdepim-data
  libkf5libkleo5 libkf5mailtransport-data libkf5mbox5 libkf5messagecore5 libkf5mime5 libkf5modemmanagerqt6 libkf5networkmanagerqt6 libkf5newstuff-data libkf5notifications-data libkf5notifyconfig-data libkf5notifyconfig5
  libkf5people-data libkf5people5 libkf5peoplebackend5 libkf5peoplewidgets5 libkf5pimtextedit5abi1 libkf5plotting5 libkf5qgpgme5 libkf5screen-bin libkf5screen7 libkf5sendlater5 libkf5su-bin libkf5su-data libkf5su5 libkf5syndication5
  libkf5sysguard-bin libkf5sysguard-data libkf5texteditor5 libkf5texteditor5-libjs-underscore libkf5threadweaver5 libkf5unitconversion-data libkf5unitconversion5 libkf5wallet-data libkf5waylandserver5 libkf5xmlrpcclient-data
  libkf5xmlrpcclient5 libkfontinst5 libkfontinstui5 libkolabxml1v5 libksgrd7 libksignalplotter7 libkubuntu1 libkwineffects8 libkwinglutils8 libkwinxrenderutils8 libkworkspace5-5 liblegacytaskmanager5 liboxygenstyle5-5
  liboxygenstyleconfig5-5 libplasma-geolocation-interface5 libpowerdevilui5 libprocesscore7 libprocessui7 libqalculate5-data libqalculate5v5 libqca-qt5-2 libqca-qt5-2-plugins libqt5multimedia5-plugins libscim8v5 libtaskmanager6
  libtelepathy-logger-qt5 libtelepathy-qt4-2 libtelepathy-qt5-0 plasma-desktop-data powerdevil-data qml-module-org-kde-activities qml-module-org-kde-bluezqt qml-module-org-kde-draganddrop qml-module-org-kde-extensionplugin
  qml-module-org-kde-kcoreaddons qml-module-org-kde-kio qml-module-org-kde-kquickcontrols qml-module-org-kde-kquickcontrolsaddons qml-module-org-kde-kwindowsystem qml-module-org-kde-solid qml-module-qtmultimedia qml-module-qtwebkit
  quassel-data recordmydesktop signon-plugin-password sni-qt swh-plugins telepathy-accounts-signon
Verwenden Sie »sudo apt autoremove«, um sie zu entfernen.
Die folgenden Pakete werden ENTFERNT:
  accountwizard akregator bluedevil dolphin frameworkintegration gwenview kaccounts-integration kaddressbook kate kcalc kde-cli-tools kde-config-gtk-style kde-config-mailtransport kde-config-screenlocker kde-config-telepathy-accounts
  kde-spectacle kde-style-qtcurve-qt5 kde-telepathy kde-telepathy-approver kde-telepathy-auth-handler kde-telepathy-contact-list kde-telepathy-desktop-applets kde-telepathy-filetransfer-handler kde-telepathy-integration-module
  kde-telepathy-kpeople kde-telepathy-minimal kde-telepathy-send-file kde-telepathy-text-ui kdeconnect-plasma kdenlive kdepim-addons kdepim-runtime kdepim-themeeditors kgamma5 khelpcenter khotkeys kinfocenter kio kio-extras kmail
  kmenuedit kmplot knotes konsole konsole-kpart kontact konversation korganizer krdc kscreen ksshaskpass ksysguard ktnef kubuntu-desktop kubuntu-full kubuntu-notification-helper kwalletmanager kwin kwin-addons kwin-common kwin-x11
  kwrited libdbusmenu-qt5 libkf5akonadicalendar5 libkf5alarmcalendar5 libkf5calendarcore5 libkf5calendarsupport5 libkf5calendarutils5 libkf5eventviews5 libkf5gapicalendar5 libkf5gapitasks5 libkf5grantleetheme5 libkf5gravatar5
  libkf5incidenceeditor-bin libkf5incidenceeditor5 libkf5kaddressbookgrantlee5 libkf5kdelibs4support5 libkf5kdelibs4support5-bin libkf5khtml-bin libkf5khtml5 libkf5ksieveui5 libkf5libkdepim-plugins libkf5libkdepim5
[... cut ...]
  libktpwidgets9 libkwalletbackend5-5 libkwin4-effect-builtins1 libpowerdevilcore2 libweather-ion7 milou plasma-dataengines-addons plasma-desktop plasma-framework plasma-integration plasma-nm plasma-runners-addons
  plasma-wallpapers-addons plasma-widgets-addons plasma-workspace powerdevil print-manager qml-module-org-kde-runnermodel qml-module-org-kde-telepathy quassel sddm sddm-theme-breeze signon-kwallet-extension skanlite systemsettings
  user-manager
0 aktualisiert, 0 neu installiert, 135 zu entfernen und 0 nicht aktualisiert.
Nach dieser Operation werden 234 MB Plattenplatz freigegeben.
Möchten Sie fortfahren? [J/n]
```

Ghostkeeper, 2016-11-01:
This same issue was also reported in https://github.com/Ultimaker/Cura/issues/1028 . So it also occurs in OSX. In OSX, updating Qt is not an option since it loads the library from the app package.

thopiekar, 2016-11-01:
Maybe there will be a chance to get this fixed by Canonical ;)

awhiemstra, 2016-11-02:
> This same issue was also reported in #1028 . So it also occurs in OSX. In OSX, updating Qt is not an option since it loads the library from the app package.

That is impossible. The Qt XCB plugin does not even get built on OSX. The reported crash on OSX is unrelated to this one.

awhiemstra, 2016-11-02:
@argo2445 One thing you can try that is not so risky as removing dbusmenu is using the AppImage we built: http://software.ultimaker.com/current/Cura-2.3.0.AppImage . Download it and make it executable, then you can start it like a normal program. The AppImage contains a different Qt that (hopefully) does not run into the same problem.

awhiemstra, 2016-11-02:
Also, I just looked at the source package for Qt in 16.10 (http://packages.ubuntu.com/source/yakkety/qtbase-opensource-src). There are a bunch of dbusmenu patches there that are the likely culprit. This suggests to me the AppImage is going to work.

argo2445, 2016-11-02:
@awhiemstra The AppImage works! Thank you very much!

Freefly13, 2016-11-02:
I have installed Cura from ppa
if i lunch in a terminal "sudo cura" the program start but not work if i try "cura" non start and i have this:
An exception occured while trying to read preferences file: 'general'
2016-11-02 23:55:18,335 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ConsoleLogger
2016-11-02 23:55:18,343 - INFO - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (__init__ [40]): Found CuraEngine at: /usr/bin/CuraEngine
2016-11-02 23:55:18,344 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraEngineBackend
2016-11-02 23:55:18,347 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LocalFileOutputDevice
2016-11-02 23:55:18,347 - WARNING - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin ConsoleLogger was already loaded
2016-11-02 23:55:18,349 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SimpleView
2016-11-02 23:55:18,350 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin WireframeView
2016-11-02 23:55:18,354 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MirrorTool
2016-11-02 23:55:18,360 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RotateTool
2016-11-02 23:55:18,362 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CameraTool
2016-11-02 23:55:18,364 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin TranslateTool
2016-11-02 23:55:18,366 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SelectionTool
2016-11-02 23:55:18,370 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ScaleTool
2016-11-02 23:55:18,372 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin FileLogger
2016-11-02 23:55:18,373 - WARNING - /usr/lib/uranium/plugins/FileHandlers/STLReader/STLReader.py (<module> [5]): Could not find numpy-stl, falling back to slower code.
2016-11-02 23:55:18,374 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLReader
2016-11-02 23:55:18,375 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJWriter
2016-11-02 23:55:18,377 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJReader
2016-11-02 23:55:18,382 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFWriter
2016-11-02 23:55:18,384 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLWriter
2016-11-02 23:55:18,386 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XRayView
2016-11-02 23:55:18,387 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileWriter
2016-11-02 23:55:18,389 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin AutoSave
2016-11-02 23:55:18,392 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LayerView
2016-11-02 23:55:18,393 - WARNING - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin CuraEngineBackend was already loaded
2016-11-02 23:55:18,395 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin X3DReader
2016-11-02 23:55:18,397 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFReader
2016-11-02 23:55:18,402 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin UltimakerMachineActions
2016-11-02 23:55:18,404 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileReader
2016-11-02 23:55:18,433 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SliceInfoPlugin
2016-11-02 23:55:18,435 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MachineSettingsAction
2016-11-02 23:55:18,438 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SolidView
2016-11-02 23:55:18,440 - WARNING - .../UM/Preferences.py (getValue [70]): Tried to get the value of non-existing setting cura/active_mode.
2016-11-02 23:55:18,441 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin PerObjectSettingsTool
2016-11-02 23:55:18,457 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XmlMaterialProfile
2016-11-02 23:55:18,470 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin USBPrinting
2016-11-02 23:55:18,479 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ChangeLogPlugin
2016-11-02 23:55:18,483 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin VersionUpgrade21to22
2016-11-02 23:55:18,487 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeWriter
2016-11-02 23:55:18,491 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RemovableDriveOutputDevice
2016-11-02 23:55:18,495 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeProfileReader
2016-11-02 23:55:18,497 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LegacyProfileReader
2016-11-02 23:55:18,501 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin Doodle3D uses an incompatible API version, ignoring
2016-11-02 23:55:18,504 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ImageReader
2016-11-02 23:55:18,505 - INFO - .../UM/Qt/QtApplication.py (__init__ [39]): Command line arguments: {'disable-textures': False, 'file': [], 'debug-mode': False, 'external-backend': False}
2016-11-02 23:55:18,509 - INFO - .../UM/VersionUpgradeManager.py (upgrade [98]): Looking for old configuration files to upgrade.
[... cut ...]
2016-11-02 23:55:29,178 - DEBUG - .../UM/Qt/Bindings/Theme.py (load [118]): Loaded theme /usr/share/cura/resources/themes/cura
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
2016-11-02 23:55:29,270 - WARNING - .../cura/Settings/ContainerManager.py (getContainerMetaDataEntry [171]): Could not get metadata of container  because it was not found.
Errore di segmentazione (core dump creato)

awhiemstra, 2016-11-03:
@Freefly13 It is failing because it cannot write to ~/.local/share/cura because you ran it as root before. You should either delete the ~/.local/share/cura and ~/.config/cura directories and restart as non-root or fix the permissions of those directories.

Freefly13, 2016-11-03:
I have delete ~/.local/share/cura and ~/.config/cura and restart cura without sudo but the aren't fix the problem
this is the log:
 cura
An exception occured while trying to read preferences file: 'general'
2016-11-03 15:45:51,714 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ConsoleLogger
2016-11-03 15:45:51,726 - INFO - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (__init__ [40]): Found CuraEngine at: /usr/bin/CuraEngine
2016-11-03 15:45:51,727 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraEngineBackend
2016-11-03 15:45:51,730 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LocalFileOutputDevice
2016-11-03 15:45:51,730 - WARNING - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin ConsoleLogger was already loaded
2016-11-03 15:45:51,732 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SimpleView
2016-11-03 15:45:51,733 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin WireframeView
2016-11-03 15:45:51,736 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MirrorTool
2016-11-03 15:45:51,742 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RotateTool
2016-11-03 15:45:51,744 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CameraTool
2016-11-03 15:45:51,746 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin TranslateTool
2016-11-03 15:45:51,747 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SelectionTool
2016-11-03 15:45:51,752 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ScaleTool
2016-11-03 15:45:51,753 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin FileLogger
2016-11-03 15:45:51,754 - WARNING - /usr/lib/uranium/plugins/FileHandlers/STLReader/STLReader.py (<module> [5]): Could not find numpy-stl, falling back to slower code.
2016-11-03 15:45:51,755 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLReader
2016-11-03 15:45:51,756 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJWriter
2016-11-03 15:45:51,758 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin OBJReader
2016-11-03 15:45:51,763 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFWriter
2016-11-03 15:45:51,764 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin STLWriter
2016-11-03 15:45:51,766 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XRayView
2016-11-03 15:45:51,768 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileWriter
2016-11-03 15:45:51,770 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin AutoSave
2016-11-03 15:45:51,773 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LayerView
2016-11-03 15:45:51,773 - WARNING - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin CuraEngineBackend was already loaded
2016-11-03 15:45:51,775 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin X3DReader
2016-11-03 15:45:51,777 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin 3MFReader
2016-11-03 15:45:51,781 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin UltimakerMachineActions
2016-11-03 15:45:51,783 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin CuraProfileReader
2016-11-03 15:45:51,810 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SliceInfoPlugin
2016-11-03 15:45:51,813 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin MachineSettingsAction
2016-11-03 15:45:51,815 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin SolidView
2016-11-03 15:45:51,818 - WARNING - .../UM/Preferences.py (getValue [70]): Tried to get the value of non-existing setting cura/active_mode.
2016-11-03 15:45:51,818 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin PerObjectSettingsTool
2016-11-03 15:45:51,832 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin XmlMaterialProfile
2016-11-03 15:45:51,841 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin USBPrinting
2016-11-03 15:45:51,848 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ChangeLogPlugin
2016-11-03 15:45:51,853 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin VersionUpgrade21to22
2016-11-03 15:45:51,856 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeWriter
2016-11-03 15:45:51,859 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin RemovableDriveOutputDevice
2016-11-03 15:45:51,863 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin GCodeProfileReader
2016-11-03 15:45:51,866 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin LegacyProfileReader
2016-11-03 15:45:51,870 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Plugin Doodle3D uses an incompatible API version, ignoring
2016-11-03 15:45:51,873 - INFO - .../UM/PluginRegistry.py (loadPlugin [63]): Loaded plugin ImageReader
2016-11-03 15:45:51,874 - INFO - .../UM/Qt/QtApplication.py (__init__ [39]): Command line arguments: {'debug-mode': False, 'external-backend': False, 'file': [], 'disable-textures': False}
2016-11-03 15:45:51,888 - INFO - .../UM/VersionUpgradeManager.py (upgrade [98]): Looking for old configuration files to upgrade.
[... cut ...]
2016-11-03 15:46:02,114 - DEBUG - .../UM/Qt/Bindings/Theme.py (load [118]): Loaded theme /usr/share/cura/resources/themes/cura
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
2016-11-03 15:46:02,206 - WARNING - .../cura/Settings/ContainerManager.py (getContainerMetaDataEntry [171]): Could not get metadata of container  because it was not found.
Segmentation fault (core dump created)

adomasven, 2016-12-05:
Hey, any updates on this at all? The AppImage for Cura 2.3.1 segfaults too.

thopiekar, 2016-12-15:
@adomasven AppImage or PPA?

adomasven, 2016-12-15:
Right, there's actually a separate issue regarding 2.3.1 AppImage segfaulting [here](https://github.com/Ultimaker/Cura/issues/1146) (in case anyone wonders).

Regarding PPA, same problem persists. I saw cura being upgraded the other day as part of a system update, but no changes - still segfaulting. So the question remains - any updates on this?

thopiekar, 2016-12-15:
Regarding my PPA: If you are using Unity: No, the problem is known and comes from Unity in some way. You should report the problem at launchpad or even look for issues, which are already open in Launchpad, and mark there that this problem "affects you", too. By doing this the issue might get an higher priority and Canonical will probably take care about it sooner.
My only solution now is simply not to use Unity. Since I use KDE daily and MATE on slow systems, I don't face this problem.

Ghostkeeper, 2016-12-15:
I've been using Unity for years with Cura. The last problem I had with it (blanco menus at random) was fixed by upgrading to Ubuntu 16.10. But that's from source.

claell, 2016-12-25:
I think I have the same problem on Ubuntu 16.10 with Cura 2.4 (beta).

@thopiekar Do you know what exactly causes the issue on Unity? If there is an issue on Launchpad, can you post a link?

thopiekar, 2016-12-25:
Hey @claell ,
I remember that I debugged the problem in the past with an user and @awhiemstra here on Github.
It happens somewhere between Qt5 and DBus. If I'm not wrong happens in Unity as it is using Qt5+DBus for the external menu bar. The strange thing is that we are already disabling this menubar, but Unity seems to force the usage of it (?).
I'm don't know of a issue ticket on this, but since Qt is a commonly used toolkit, another project should also already ran into it.

awhiemstra, 2016-12-27:
Not necessarily, this issue is specifically about the QML Menu component. Most other applications will just be using a QWidgets based menu bar, which probably works fine. The QML version however triggers the crash.

claell, 2016-12-30:
Could someone with knowledge please file a bug report or point to an existing one? I think I am not experienced enough to create or find one.

nickthetait, 2016-12-30:
Here is a direct link to creating a new issue: https://github.com/Ultimaker/Cura/issues/new
Just need to write in a title and detailed description of the problem

claell, 2016-12-30:
What I meant is to create an issue on Launchpad so the Ubuntu developers can fix this.

thopiekar, 2016-12-30:
@claell Ah, you are talking about the problem with Unity?

claell, 2016-12-30:
Yes

thopiekar, 2016-12-30:
I would report the bug here: https://bugs.launchpad.net/ubuntu/+source/unity
Maybe also attach the Qt5 package related to this issue, so the Qt5 maintainers also get notified about this problem.

claell, 2016-12-31:
Alright, I created a bug report: https://bugs.launchpad.net/ubuntu/+source/unity/+bug/1653345

Since I don't know much about this I hope it is enough. Else feel free to add more information.

thopiekar, 2016-12-31:
Thanks. I also added some info to it.

To the others with the problem: If you have a launchpad account, please visit the ticket and mark yourself there that this issue 'affects' you. This will make the issue more important in Canonicals issue tracker.

Regards and happy new year!

skorokithakis, 2017-01-05:
That's very unfortunate. Is there a work around for now, short of changing desktop environment completely? I would quite like to keep using Cura.

thopiekar, 2017-01-05:
Sadly no. I installed Unity here on my own and compared the environment variables between the both desktop sessions.
While doing this I removed and changed some of them in order to simulate my KDE session and I still couldn't get it work.

I just can hope that someone at Canonical will find a solution or even workaround soon, so people won't need to install another desktop in order to use Cura.

thopiekar, 2017-01-05:
@awhiemstra Do you remember whether we located which patch file causes this problem? I can only remember that we found out that an unpatched Qt5 installation (as we are using it in our AppImages) just works fine.
If it is really related to a Ubuntu patch and not included here (https://anonscm.debian.org/cgit/pkg-kde/qt/qtbase.git/tree/debian/patches), then I can try to set up a separate PPA with nightly builds of Qt5 using the packaging files from Debian.

@skorokithakis Alternatively you can try our AppImage. You can find it in our beta section on our download page.

skorokithakis, 2017-01-05:
Do you have a link to an AppImage for 2.4.0? All I found (after considerable searching) was one for 2.3.1.

skorokithakis, 2017-01-05:
Ah, thanks. That really should say "AppImage" somewhere.

skorokithakis, 2017-01-05:
The AppImage works perfectly, thank you.

awhiemstra, 2017-01-05:
@thopiekar It is one of the dbusmenu related patches in the ubuntu source package. I can't recall which one precisely. In any case, they are not in the debian packages as far as I can tell.

[1 more reply]

closed 2017-07-27 by nallath