Ultimaker/Cura #1095: Cura 2.3.0 UI Bug: "Print One at a time" allows to save even if not all objects can be printed.

Cyberman-tM, 2016-11-01:
I'm trying to print two objects with "print one at a time" - so far I failed to get Cura to allow both to print, but that's OK.

However, if just one of the objects is printable, the "save" buttons becomes active.

Saving that file will get me the same result as if the second object were not there.

I'm using a custom profile for a Wanhao i3 on Win7 64Bit. Not sure where to find a better ID for Cura, or is it really just 2.3.0? Probably not the latest beta anymore?

thopiekar, 2016-11-01:
@Cyberman-tM Currently 2.3.1 is the version we are focusing on.
Please retry on 2.3.1 and feel free to update your info here. 😉

Ghostkeeper, 2016-11-01:
2.3.0 is the latest public release, so Cyberman is using the latest version available to him.

Ghostkeeper, 2016-11-01:
The behaviour that objects outside of the build volume are excluded from the print is intentional, and it also happens when not using "One at a Time". There's also some argument to be had for just refusing to slice altogether when something is out of the build volume. But it is quite handy to just be able to keep some stuff floating around in space while you slice multiple build plates full of stuff if not everything fits in one print.

nickthetait, 2016-11-01:
Would it be worth creating a preference called something like "Allow slicing with objects outside build volume" to let users decide? It may be overkill and just add complication... not sure.

nallath, 2016-11-01:
I'm not sure. Although i'm generally in favor of adding more preferences, we should be carefull with adding too much of those settings without having a good interace to manage all of them.

Ghostkeeper, 2016-11-01:
> Stupid idea: is it possible to color the "save" button? As in "green - everythings fine", "red - can't slice", "yellow - slicing possible, but see tooltip"?

Interesting idea. It is certainly technically possible. We have to be a bit careful with the colour scheme though. You could add very light blue diagonal stripes as alternative, sort of like that black-and-yellow warning tape. Or thicker borders on the button, just to indicate that something is off. And then a tooltip, of course.

closed 2018-11-26 by Appesteijn