Ultimaker/Cura #1096: Crashes when closing Cura and using NVidia drivers

thopiekar, 2016-11-01:
As discussed in another issue report, here the info by GDB:

```
thopiekar@home:~$ gdb -ex r --args python3 /usr/bin/cura
GNU gdb (Ubuntu 7.11.90.20161005-0ubuntu1) 7.11.90.20161005-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from python3...Reading symbols from /usr/lib/debug/.build-id/b0/38c0c0f8d605ffb7d8fc0f48509284140ff3ae.debug...done.
done.
Starting program: /usr/bin/python3 /usr/bin/cura
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff38a2700 (LWP 12271)]
[New Thread 0x7ffff30a1700 (LWP 12272)]
[New Thread 0x7ffff08a0700 (LWP 12273)]
[New Thread 0x7fffe5165700 (LWP 12274)]
[New Thread 0x7fffe2964700 (LWP 12275)]
[New Thread 0x7fffe0163700 (LWP 12276)]
[Thread 0x7fffe0163700 (LWP 12276) exited]
[Thread 0x7fffe2964700 (LWP 12275) exited]
[Thread 0x7fffe5165700 (LWP 12274) exited]
[Thread 0x7ffff08a0700 (LWP 12273) exited]
[Thread 0x7ffff30a1700 (LWP 12272) exited]
[Thread 0x7ffff38a2700 (LWP 12271) exited]
[New Thread 0x7ffff08a0700 (LWP 12283)]
[New Thread 0x7ffff30a1700 (LWP 12285)]
2016-11-01 12:40:13,556 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin ConsoleLogger
2016-11-01 12:40:13,565 - INFO - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (__init__ [40]): Found CuraEngine at: /usr/bin/CuraEngine
2016-11-01 12:40:13,566 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin CuraEngineBackend
2016-11-01 12:40:13,566 - WARNING - .../UM/PluginRegistry.py (loadPlugin [69]): Plugin ConsoleLogger was already loaded
2016-11-01 12:40:13,570 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin SelectionTool
2016-11-01 12:40:13,580 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin RotateTool
2016-11-01 12:40:13,585 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin MirrorTool
2016-11-01 12:40:13,589 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin CameraTool
2016-11-01 12:40:13,595 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin TranslateTool
2016-11-01 12:40:13,604 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin ScaleTool
2016-11-01 12:40:13,606 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin FileLogger
2016-11-01 12:40:13,609 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin OBJWriter
2016-11-01 12:40:13,614 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin STLWriter
2016-11-01 12:40:13,617 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin OBJReader
2016-11-01 12:40:13,625 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin 3MFWriter
2016-11-01 12:40:13,631 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin STLReader
2016-11-01 12:40:13,636 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin LocalFileOutputDevice
2016-11-01 12:40:13,639 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin SimpleView
2016-11-01 12:40:13,641 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin WireframeView
2016-11-01 12:40:13,643 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin XRayView
2016-11-01 12:40:13,644 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin CuraProfileReader
2016-11-01 12:40:13,647 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin ChangeLogPlugin
2016-11-01 12:40:13,649 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin X3DReader
[New Thread 0x7ffff38a2700 (LWP 12291)]
[New Thread 0x7fffe0163700 (LWP 12292)]
[New Thread 0x7fffb92b4700 (LWP 12293)]
[New Thread 0x7fffb8ab3700 (LWP 12294)]
2016-11-01 12:40:13,796 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin UM3NetworkPrinting
2016-11-01 12:40:13,801 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin VersionUpgrade21to22
[New Thread 0x7fffabfff700 (LWP 12295)]
2016-11-01 12:40:13,808 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin RemovableDriveOutputDevice
2016-11-01 12:40:13,879 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin SliceInfoPlugin
2016-11-01 12:40:13,883 - INFO - .../UM/PluginRegistry.py (loadPlugin [69]): Loaded plugin PerObjectSettingsTool
[... 103 lines cut ...]
#4  0x00005555565ea240 in  ()
#5  0x00007fffffffc7bf in  ()
#6  0x00007fffdba41c68 in  () at /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
(gdb) py-bt
Traceback (most recent call first):
(gdb) info threads
  Id   Target Id         Frame 
  1    Thread 0x7ffff7f9a700 (LWP 12267) "python3" _dl_close_worker (map=map@entry=0x5555566dfd40, force=force@entry=false) at dl-close.c:792
* 8    Thread 0x7ffff08a0700 (LWP 12283) "QXcbEventReader" 0x00007fffd35b4299 in ?? ()
  10   Thread 0x7ffff38a2700 (LWP 12291) "Qt bearer threa" 0x00007ffff78ef0bd in poll () at ../sysdeps/unix/syscall-template.S:84
  11   Thread 0x7fffe0163700 (LWP 12292) "python3" 0x00007ffff78f0ea3 in select () at ../sysdeps/unix/syscall-template.S:84
  12   Thread 0x7fffb92b4700 (LWP 12293) "python3" 0x00007ffff7bc9bf3 in futex_abstimed_wait_cancelable (private=0, abstime=0x7fffb92b34f0, expected=0, futex_word=0x7fffb0000e60) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
  13   Thread 0x7fffb8ab3700 (LWP 12294) "python3" 0x00007ffff7bc9bf3 in futex_abstimed_wait_cancelable (private=0, abstime=0x7fffb8ab24f0, expected=0, futex_word=0x7fffa4001110) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
  14   Thread 0x7fffabfff700 (LWP 12295) "python3" 0x00007ffff78f0ea3 in select () at ../sysdeps/unix/syscall-template.S:84
  15   Thread 0x7fffab379700 (LWP 12296) "python3" 0x00007ffff78f0ea3 in select () at ../sysdeps/unix/syscall-template.S:84
  16   Thread 0x7fffaab78700 (LWP 12297) "python3" 0x00007ffff7bc9bf3 in futex_abstimed_wait_cancelable (private=0, abstime=0x7fffaab774f0, expected=0, futex_word=0x7fff9c000e60) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
  17   Thread 0x7fffaa377700 (LWP 12298) "python3" 0x00007ffff78f0ea3 in select () at ../sysdeps/unix/syscall-template.S:84
  18   Thread 0x7fffa9b76700 (LWP 12299) "python3" 0x00007ffff7bc9bf3 in futex_abstimed_wait_cancelable (private=0, abstime=0x7fffa9b754f0, expected=0, futex_word=0x7fff94000e60) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
  19   Thread 0x7fffa9375700 (LWP 12300) "python3" 0x00007ffff7bc9bf3 in futex_abstimed_wait_cancelable (private=0, abstime=0x7fffa93744f0, expected=0, futex_word=0x7fff88000e80) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
  24   Thread 0x7fff777bf700 (LWP 12361) "python3" 0x00007ffff7bc9a07 in futex_abstimed_wait_cancelable (private=0, abstime=0x0, expected=0, futex_word=0x7fff700021c0) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
  25   Thread 0x7fff76fbe700 (LWP 12362) "python3" 0x00007ffff7bc9a07 in futex_abstimed_wait_cancelable (private=0, abstime=0x0, expected=0, futex_word=0x7fff680b3a20) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
  26   Thread 0x7fff767bd700 (LWP 12363) "python3" 0x00007ffff7bc9a07 in futex_abstimed_wait_cancelable (private=0, abstime=0x0, expected=0, futex_word=0x7fff6c001b70) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
  27   Thread 0x7fff75fbc700 (LWP 12364) "python3" 0x00007ffff7bc9a07 in futex_abstimed_wait_cancelable (private=0, abstime=0x0, expected=0, futex_word=0x7fff60001310) at ../sysdeps/unix/sysv/linux/futex-internal.h:205
  28   Thread 0x7fff7577b700 (LWP 12365) "QQmlThread" 0x00007ffff78ef0bd in poll () at ../sysdeps/unix/syscall-template.S:84
```

awhiemstra, 2016-11-01:
What version of Qt is this with?

thopiekar, 2016-11-01:
Qt5: `5.6.1+dfsg-3ubuntu6~4`
PyQt5: `5.7+dfsg-2`

awhiemstra, 2016-11-01:
It's probably related to the other bug, since both seem to crash in the QXcb code.

thopiekar, 2016-11-01:
Do you really think The Qt Company will care about this? I mean we are using closed source drivers here. I can image they will blame it on Nvidia - the usual game when it comes to proprietary binaries..

closed 2018-02-21 by ianpaschal