Ultimaker/Cura #1097: Feature Request: Explicit options for Firmware Retraction and Vol. Extrusion

landodragon141, 2016-11-01:
Feature Request: Explicit options for Firmware Retraction and Vol. Extrusion

Not much else to say. Can we please have an explicit option to enable FR and Vol Extrusion? This would be the final nail in the coffin for me using Slic3r.

Ghostkeeper, 2016-11-01:
I can see how that would be useful. Should be a fairly simple addition to the engine. At the moment it decides these things based on the g-code flavour. For UltiGCode it creates G10 and G11 and volume-based extrusion.

We could even let these settings "inherit" from the g-code flavour in the front-end. I'll add a ticket.

landodragon141, 2016-11-04:
Yeah my problem is that when I create a custom machine profile I can't select Marlin (volumetric). Honestly that's probably a bug but I think it would still be a useful feature.

dintid, 2017-08-04:
What is the status of this? 
Using the Diamond Hotend we must have firmware retraction which can be achieved (As far as I know) by using Ulticode or RepRap volumetric. Ulticode is not good for non-ultimakers while RepRap Volumetric doesn't use the G10 and G11? Or am I misinformed?
Regardless of the codes used, it would be usefull to have a description in cura to show that firmware retraction is actually turned on.

dintid, 2017-11-06:
I see Relative Extrusion has been added to latest Cura version! That's awesome! The Duet community rejoices.

Now we just need a tick to allow Firmware Retraction. Pretty Please. This is extremely important for Multi extrusions mixing hotends setups.

Does any of the Custom Useable Gcode flavors support Firmware Retraction? Aside from Volutmetric, as Volumetric is causing all kinds of problems (lots of confusion) since we have to define filament type as Volumetric rather than diameter in RepRapFirmware and other places.

Ghostkeeper, 2017-11-10:
Bits From Bytes, Marlin Volumetric and UltiGCode are the flavours that currently trigger firmware retractions.

Ghostkeeper, 2017-11-17:
I've made a PR that implements it. Let's see if my colleagues agree.

Ghostkeeper, 2017-11-21:
I created a ticket for the generic breaking the flavour into all of its component decisions, but that was rejected by our project manager since it's older than 12 weeks (the PR still stands).

PR #2797 (2017-11-28): Add setting to determine whether to use firmware retraction
This allows printers to determine this for themselves. By default it's based on the g-code flavour in the same way as it is now implemented in the engine.

See also https://github.com/Ultimaker/CuraEngine/pull/656

Fixes #1097.
commit 94dd5247 (2017-11-28): Add setting to determine whether to use firmware retraction

This allows printers to determine this for themselves. By default it's based on the g-code flavour in the same way as it is now implemented in the engine.

Fixes #1097.
closed 2017-11-28 by ChrisTerBeke