Ultimaker/Cura #1098: Ubuntu 16.04 - PPA: After last update Cura behave like a fresh install

BerndJM, 2016-11-01:
@thopiekar 
Since the todays update Cura "greets" after startup with the "Add printer" dialog.
Even in this dialog my custom printer isn't present.
Had a look in .local/share/cura - all my definitions are there. Seems like cura ignores the definitions in this place ?!? Any idea what's going wrong here?

thopiekar, 2016-11-01:
Which printer do you own?

BerndJM, 2016-11-01:
It's a Velleman K8200 - and worked fine with my definitions all the time 'til today

thopiekar, 2016-11-01:
@Ghostkeeper has this printer been removed? 🤔 
@BerndJM I can find your printer neither in the "2.3" branch nor in the "master" branch. Was this definition you own? One you've created and copied somewhere?

BerndJM, 2016-11-01:
I've created the definition on base of the custom FDM printer

thopiekar, 2016-11-01:
Where was it copied to? Is it still there?

BerndJM, 2016-11-01:
As I wrote above:
Had a look in .local/share/cura - all my definitions are there

BerndJM, 2016-11-01:
[local-share-cura.zip](https://github.com/Ultimaker/Cura/files/564677/local-share-cura.zip)

thopiekar, 2016-11-01:
`k8200.def.json` ? Never added a custom definition this way, but I'm sure @Ghostkeeper can help with that.
Alternatively you can try moving your file to /usr/share/cura/resources/definitions/ as root.
This directory really has to work as all the other defintions are there..
(the path might be incorrect, as I'm on Windows at the moment to work on something...)

Good luck 😉
.. Ah, and don't forget to share your definition if it works. I'm sure other Velleman K8200 users might find it useful 😃

thopiekar, 2016-11-01:
Btw. the log might be useful here, too. Maybe Cura already says what is hurting here..

BerndJM, 2016-11-01:
O.k. the strange thing: it worked 'til yesterday without any problems.
Here is the log from the last start:
[cura.log.zip](https://github.com/Ultimaker/Cura/files/564695/cura.log.zip)

mgroene, 2016-11-01:
I was just running into this problem, too. Windows 7, Cura 2.3 setting up from scratch.
I can't reproduce reliable, but if I use german umlauts in the material profile e.g. for the name/color, everything seems to be gone at the next restart.

BerndJM, 2016-11-01:
In addition:
I played around a bit more
- placing my printer definition in /usr/share/cura/resources/definitions/ gives me the possibility to select it under "Others"
- but Cura didn't remember this - after closing and reopening, there is again the "Add printer" dialog
- even if I choose a different printer (to be sure it's not caused by my def file) - the same behavior.

It looks like Cura didn't read the config file in ".config/cura" and/or the files in ".local/share/cura" at startup. The files have full read/write access. And Cura writes also to the files e.g. if I choose another printer ... strange ?!?

mgroene, 2016-11-01:
okay, indeed there seems to be problems with utf8 or some unicode-things. see logs:
```
2016-11-01 22:08:00,850 - INFO - C:\Program Files\Cura 2.3\plugins\UpdateChecker\UpdateChecker.py (run [28]): Checking for new version of cura
2016-11-01 22:08:00,866 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin USBPrinting
2016-11-01 22:08:00,881 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin VersionUpgrade21to22
2016-11-01 22:08:00,897 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin SimpleView
2016-11-01 22:08:00,897 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin WireframeView
2016-11-01 22:08:00,913 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin X3DReader
2016-11-01 22:08:00,928 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin XmlMaterialProfile
2016-11-01 22:08:00,944 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\PluginRegistry.py (loadPlugin [63]): Loaded plugin XRayView
2016-11-01 22:08:00,944 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py (__init__ [39]): Command line arguments: {'debug-mode': False, 'external-backend': False, 'disable-textures': False, 'file': []}
2016-11-01 22:08:00,960 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py (upgrade [98]): Looking for old configuration files to upgrade.
2016-11-01 22:08:01,022 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_logSocketState [165]): Socket state changed to Listening
2016-11-01 22:08:01,022 - INFO - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (startEngine [55]): Started engine process: C:\Program Files\Cura 2.3\CuraEngine.exe
2016-11-01 22:08:01,022 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] Calling engine with: ['C:\\Program Files\\Cura 2.3\\CuraEngine.exe', 'connect', '127.0.0.1:49674', '-j', 'C:\\Program Files\\Cura 2.3\\resources\\definitions\\fdmprinter.def.json', '']
2016-11-01 22:08:01,038 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] 
2016-11-01 22:08:01,038 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] Cura_SteamEngine version DEV
2016-11-01 22:08:01,038 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] Copyright (C) 2014 David Braam
2016-11-01 22:08:01,038 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] 
2016-11-01 22:08:01,038 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] This program is free software: you can redistribute it and/or modify
2016-11-01 22:08:01,038 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] it under the terms of the GNU Affero General Public License as published by
2016-11-01 22:08:01,038 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] the Free Software Foundation, either version 3 of the License, or
2016-11-01 22:08:01,038 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] (at your option) any later version.
2016-11-01 22:08:01,038 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] 
2016-11-01 22:08:01,038 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] This program is distributed in the hope that it will be useful,
2016-11-01 22:08:01,038 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2016-11-01 22:08:01,038 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2016-11-01 22:08:01,038 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] GNU Affero General Public License for more details.
2016-11-01 22:08:01,038 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] 
2016-11-01 22:08:01,038 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] You should have received a copy of the GNU Affero General Public License
2016-11-01 22:08:01,038 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2016-11-01 22:08:01,053 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_logSocketState [165]): Socket state changed to Connected
2016-11-01 22:08:01,053 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_onSocketStateChanged [155]): Backend connected on port 49674
[... cut ...]
2016-11-01 22:13:29,965 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py (windowClosed [197]): Shutting down cura
2016-11-01 22:13:31,224 - DEBUG - C:\Program Files\Cura 2.3\plugins\CuraEngineBackend\CuraEngineBackend.py (_terminate [186]): Attempting to kill the engine process
2016-11-01 22:13:31,225 - DEBUG - C:\Program Files\Cura 2.3\plugins\CuraEngineBackend\CuraEngineBackend.py (_terminate [186]): Killing engine process
2016-11-01 22:13:31,233 - DEBUG - C:\Program Files\Cura 2.3\plugins\CuraEngineBackend\CuraEngineBackend.py (_terminate [186]): Engine process is killed. Received return code 1
```

Ghostkeeper, 2016-11-01:
I've been running into the same problem the past few days too. It just seemed to delete my printers. I couldn't pinpoint it though, and maybe it was just because of me switching between master and 2.3. Nobody else seemed to run into it. So I held off with creating an issue about it until I could find reliable reproduce steps.

There was a Velleman definition coming in via a pull request: https://github.com/Ultimaker/Cura/pull/1079
It isn't accepted yet though, and it seems to be a different printer. The pull request isn't accepted yet mainly because of a syntax error in the definition.

Ghostkeeper, 2016-11-01:
As for the other question, if you put definitions in `~/.local/share/cura/definitions`, it should load just like any other definition. But the `definitions` subfolder doesn't exist by default, so you might have to create it.

Ghostkeeper, 2016-11-01:
> UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf6 in position 238: invalid start byte

Highlighting this. We recently changed some things in the saving to more reliably use UTF-8 everywhere regardless of system preferences. The series of commits to do this was these:
* https://github.com/Ultimaker/Cura/commit/bfec96f58425517cb89fa91d4c41a085fa515c70
* https://github.com/Ultimaker/Uranium/commit/c976a275f4dc7daaf0cbed69a2f890029048372d
* https://github.com/Ultimaker/Uranium/commit/5e45e969fe0d14c3553e58e7b7b178b14af5a5d9
* https://github.com/Ultimaker/Uranium/commit/0f7ecc1b5bb576bf9c9feee649a7c781d2b88f01
* https://github.com/Ultimaker/Uranium/commit/49e6e67dd4e430e834c250af25cdf25c4c40d2c3
* https://github.com/Ultimaker/Uranium/commit/5502a4cb418f4846eb68cf89e060ece963541817

We might have missed a place to use UTF-8.

BerndJM, 2016-11-01:
Yes the Velleman Vertex K8400 is a different printer.
And about the `~/.local/share/cura/definitions` - I know this and it worked the whole time fine 'til today.
Out of courisity just switched between "master" and "stable" but the behavior is the same. All time on startup comes the "Add printer" dialog - as said before it looks like Cura didn't read the config folder at startup ...

thopiekar, 2016-11-01:
@Ghostkeeper Can confirm the same behaviour here on master. :disappointed:

mgroene, 2016-11-02:
Additionally I want to mention, that filenames of Definitions in the config path contain some utf8codes. Dont know whether this is intended or should ne striped off, so i just leave the Note.

awhiemstra, 2016-11-02:
I pulled master yesterday and suddenly saw the not-saving behaviour. It seems something recent broke things.

BerndJM, 2016-11-02:
Looked a bit around in the sources for changes in the last days, and think I found the reason let Cura ignoring the configuration:
https://github.com/Ultimaker/Uranium/commit/ba1a3d7b66ed46a89b6045e635bf2ce44c59d221
Reverted the changes there and all seems to work fine.
Can someone please check and confirm this?

@mgroene:
I'm seriously wondering why you hijack a Linux/Ubuntu related thread with your windows/utf-8 issues ?!?

thopiekar, 2016-11-06:
I'm not sure, but I guess that is has been solved now.
At least I don't see that problem here anymore, neither on [2.3] nor in [master].

closed 2016-11-06 by thopiekar

awhiemstra, 2016-11-07:
Yes, I fixed it with https://github.com/Ultimaker/Uranium/commit/f8594982b720d33eac206b0be1f5d7b5ef6bccf8 . Forgot to mention it here.

thopiekar, 2016-11-07:
@awhiemstra Thanks :smile: