Ultimaker/Cura #1099: 2.3.0: Gap filling?

CCS86, 2016-11-02:
Hey guys, did the gap filling setting get taken out of the front end?

CCS86, 2016-11-02:
[image]

Ghostkeeper, 2016-11-02:
Also out of the back end. It was sharing most of its code with the old Remove Overlapping Walls feature, which got replaced by Compensate Wall Overlaps. Gap filling didn't get replaced. We plan to add it back for version 2.4.

closed 2016-11-02 by Ghostkeeper

CCS86, 2016-11-02:
Thanks @Ghostkeeper . Will it have a better chance of filling areas like this?

[image]

Ghostkeeper, 2016-11-03:
Compensate Wall Overlaps should fill those if they are wider than one line width, but it is limited in the sense that it needs to fit normal lines first before it can detect that there's wall overlap (and then compensate for the overlap). And it won't fit lines if there is less than one line width's worth of space.

Gap filling would have a better chance of filling such areas, though it is meant for filling in the insides of corners.

CCS86, 2016-11-03:
Is there any chance that instead of forcing the gap filling algorithm to fill long continuous gaps like these, the compensate wall overlaps could be leveraged instead? It would try to lay down a full width bead here, but compensate wall overlaps would reduce the extrusion amount to match?

Ghostkeeper, 2016-11-04:
No. Compensate Wall Overlap only kicks in after the walls are laid down initially. And the walls aren't laid down if the space is less than one line width. Also, Compensate Wall Overlaps only reduces the line width of one of the walls and only looks at the current polygon.

amoose136, 2017-03-14:
Just a thought, what if the behavior for thin lines less the nozzle diameter was to make the pass anyway but temporarily change the flow modifier so the correct amount of plastic is extruded to fill the gap? Sure it might not work perfectly but maybe if the pass is fast enough so as to not heat the plastic on the surrounding walls significantly, the melted plastic would flow into the right place. Some models need something like this or it just tries to print two disconnected shells for the inside and outside walls and this of course results in a weak print.

Ghostkeeper, 2017-03-14:
It lays down the wall at half the line width away from the actual edge, so that normally the wall reaches (in theory) exactly to the edge of the polygon. If you were to adjust the line width afterwards, where should the wall be placed?