Ultimaker/Cura #1100: 2.3.0: No "support placement" setting in per-object

CCS86, 2016-11-02:
Like the title says, when enabling per-object settings, the "support placement" setting is not available, and assumes the global value.

Ghostkeeper, 2016-11-03:
Intentional. Support Placement (touching build plate vs. everywhere) is not settable per extruder or per object. It's currently not supported by our engine.

It might be possible to change this. `support_type` is currently only used in [generateSupportAreas](https://github.com/Ultimaker/CuraEngine/blob/master/src/support.cpp#L116), where there is a mesh available. It does some tricky things though with a second pass over all support layers to subtract the areas which don't touch the build plate.

Ghostkeeper, 2017-02-18:
The new product manager for Cura at Ultimaker rejected this issue for being too unimportant. This means that the main developers can't spend time on it.

ianpaschal, 2018-02-26:
Since Cura 2.3 is no longer supported, this issue will be considered `Won't Fix`.

Ghostkeeper, 2018-03-16:
This is still not implemented in 3.2 though.

fieldOfView, 2018-03-16:
If one at a time printing is enabled, it the support placement option should be available in Cura 3.3 (#3154)

closed 2018-04-16 by Ghostkeeper