Ultimaker/Cura #1101: 2.3.0: Set Object on Buildplate?

CCS86, 2016-11-02:
I know there is a global program setting for this (which I have enabled). 

There seems to be a slight issue with it. I had a part, which was a plate of sorts. I added it to the buildplate, then duplicated it. Then took that duplicate, and using snapped rotation, gave it a full 180* to put the other side down. 

Everything looked fine, and the start layers all looked normal (full sized footprint on both). But the flipped object had one less 0.2mm layer at the top.

Ghostkeeper, 2016-11-03:
When you flip an object upside down, the alignment of the slices is different. If your plate had small details on one side, those details might disappear if they happen to fall between two layers. The engine will discard any layers at the bottom that are empty.

But if it's really just a plate then I wouldn't expect something like that to happen. I tried reproducing this with a flattened cube (various thicknesses ranging from 1mm to 2.1mm), but no luck trying to reproduce it. How should I go about trying to reproduce this?

CCS86, 2016-11-03:
Here's the model I was working with:

[Plate n.stl.txt](https://github.com/Ultimaker/Cura/files/569094/Plate.n.stl.txt)

Ghostkeeper, 2016-11-04:
Huh. You can actually see the mesh drop down to the build plate in the front end.

In normal orientation, Cura thinks that the mesh is 3.11mm high. Rotated upside-down, it becomes 3.19mm high. And you can see the same effect if you flip it instead of rotating it upside-down!

I think this is an inaccuracy in the transformation matrices.

CCS86, 2016-11-04:
Interesting!

Ghostkeeper, 2017-04-24:
Our product owner has just shot this issue down because it has been inactive for too long.

Ghostkeeper, 2017-12-15:
This is fixed for Cura 3.2.

closed 2017-12-15 by Ghostkeeper