Ultimaker/Cura #1102: Import or set cfg quality profiles

marcopiraccini, 2016-11-03:
I'm using cura-2.3.1-0 on Linux.
I'm glad that Prusa i3 MK2 has been added :), however I have some "old" quality profile files (`inst.cfg`) that I'd like to use, but I don't know how to make them in "profiles" menu (nor I can "import" them). 

Here the files: https://github.com/Apsu/cura-i3-mk2/tree/master/quality

Any idea in how to do that? I'd like also to "connect" those definitions with the printer definition...

Ghostkeeper, 2016-11-03:
Those profiles look fine to me, actually. We should have a feature to import .inst.cfg files some day...

But anyway, I suggest for now that you move those quality profiles into this folder: `~/.local/share/cura/quality`. They should then appear as quality profiles for your Prusa i3 MK2. If they are already there but don't show up for the Prusa printers, I think that's a bug.

marcopiraccini, 2016-11-03:
"If they are already there but don't show up for the Prusa printers, I think that's a bug"
Yes, indeed I copied those 3 files in `~/.local/share/cura/quality` but I cannot see them in Cura. 
Maybe I have to follow same naming convention to make them loaded?

fieldOfView, 2016-11-03:
In your profiles, quality=... should be quality_type=...

marcopiraccini, 2016-11-03:
Yes! you are right, changing to `quality_type` worked.

awhiemstra, 2016-11-03:
Hah, I was looking at the differences between those files and another custom profile and completely overlooked that they were using `quality =` and not `quality_type =`

closed 2017-11-14 by ChrisTerBeke