Ultimaker/Cura #1104: Custom materials are not saved

marcopiraccini, 2016-11-03:
Cura 2.3.1 (PPA) on Ubuntu 16.04
I duplicate a generic material, then I edit it, changing name and stuff.
When I close and reopen Cura, the material I created is disappeared. 

Also, if I export materials, the import seems not to work

nallath, 2016-11-03:
Could you add your logs?

marcopiraccini, 2016-11-04:
Sure. Actions done: 
- Started Cura
- Materials -> Duplicate PLA
- Edited the duplicated PLA, called "PLA - NEW" and set the diameter to 1.75mm
- Closed Cura => `
[log_cura_1.txt](https://github.com/Ultimaker/Cura/files/570827/log_cura_1.txt)

- Reopened Cura. Reopened Materials, "PLA - NEW" NOT was present =>
[log_cura_2.txt](https://github.com/Ultimaker/Cura/files/570828/log_cura_2.txt)

Ghostkeeper, 2016-11-04:
The logs make no mention of any "PLA - NEW" :(

BerndJM, 2016-11-04:
@Ghostkeeper: The log didn't refer to the given name but to a kind of filename and counter `....#2`.
I can confirm what @marcopiraccini wrote, trying to help to clearify things I made a "commented logfile" where you can see the steps I've done at the different moments in the log, maybe it helps.
[cura.log.commented.txt](https://github.com/Ultimaker/Cura/files/572352/cura.log.commented.txt)

BerndJM, 2016-11-04:
o.k. got it (I think) - compared the generated file with the material-files shipped with cura
@marcopiraccini could you please crosscheck this:
in your new generated material file in ~.local/share/cura/materials should be a line:
`<compatible>True</compatible>`
just delete this line and save the file and it should be visible within curas material-select

marcopiraccini, 2016-11-04:
Yes! removing `compatible` the material is now visible. 
I guess that this is a bug on saving...

BerndJM, 2016-11-04:
thanks for confirmation.
Yeah a bug in saving ... or in reading ... that's the question ;-))
@Ghostkeeper it's your turn now ...

Ghostkeeper, 2016-11-07:
That's some nice debugging, guys! We'll take a shot at this then.

MabuClou, 2016-11-12:
Ok. It is anoying. This line is always readded. So i always have to delete this line `<compatible>True</compatible>` 
before starting Cura.

Also, this line seems to be added when duplicating a material. So the new material will not even show up in the manager, while there is actually a file created in the directory.

And also, it is not possible to edit the fields about filamentcost and spoolweight. Is there a documentation about the XML-Structure of this file?

And when starting the Material Manager, then it does not show all my custom materials. I have used brand "ColorFabb", so the should stay on the top of the list. But there is only one line. I have to scroll down the list, before all the materials are available on the top of the listing.

Ghostkeeper, 2016-11-14:
The filament cost and weight editing was a feature that didn't get done in time for 2.3. It'll be available later. For now the field only displays the cost and weight information (if available).

dWoolridge, 2016-11-29:
Please note this problem is also active on Windows 7 64-bit.

dWoolridge, 2016-12-04:
Any idea when this will be fixed?  I really would like to switch to 2.3.1 (from 15.02.1), but this is preventing me from making the upgrade.  I'm really looking forward to seeing how coasting and retraction priming can improve print quality.

carlosjln, 2016-12-04:
I didn't know about this issue until now that I decided to use Process Monitor and do my own debugging, I guess I'm on the waiting list as well :)

Ghostkeeper, 2016-12-05:
We've had some changes last week that were aimed at other things, but also seems to have fixed this issue. I can't reproduce it any more with these reproduce steps (which used to reproduce it):

* Add an Ultimaker 2+ machine.
* Go into Manage Materials.
* Duplicate the TPU material.
* Rename the TPU material to TUTURU-TPU.
* Close Cura and restart it.
* Check if the TUTURU-TPU material is present.

So I hope it's fixed, but I'd like to keep this one open at least until the beta of 2.4 to see if it still occurs with other materials or something.

StefanKi, 2016-12-17:
Unfortunately the problem isn't fixed for me.
I am using ppa:thopiekar/cura-master.
Should it already be fixed there?

BerndJM, 2016-12-17:
No, it's not (fully?) fixed at the moment.
You can delete the line mentioned above from the file and the definition is visible in Cura.
But a soon as you edit the material definition (in Cura) and it's saved, this line will be written again in the file :-(

Ghostkeeper, 2016-12-27:
I still can't reproduce it with the steps I described on December 5th.
Is there something different you're trying? Could you try again with the 2.4 beta from the Ultimaker website?

StefanKi, 2016-12-27:
Since i am using ppa:thopiekar/cura-master i don't think, that using the version from the website would be more up to date. Lately all configuration was lost and i had to redo all confugurations - but the problem persisted.

I will retry it with the latest ppa:thopiekar/cura-master version and post the results the day after tomorrow.

Ghostkeeper, 2016-12-27:
The (beta) version on the website was made last Thursday.

dWoolridge, 2016-12-27:
I'm not sure if this contributes in some way to the problem but here's what I think just happened...
1)  Downloaded and installed Cura 2.4 Beta
2)  Created new Material--I'm not 100% sure, but I think it still existed after closing and re-opening Cura 2.4 beta
3)  Opened Cura 2.4 Beta
4)  Opened Cura 15.02.1 at the same time-to see what my old settings were as I entered them into Cura 2.4
5)  Problems with "materials not saved" re-appeard

dWoolridge, 2016-12-27:
Please disregard my previous post.  Opening Cura 15.02.1 along with 2.4.0 Beta doesn't appear to have anything to do with the problem.  The following steps show that the "<compatible>True</compatible>" line is still added to the file any time a custom material is edited.

1. Uninstalled Cura 2.4.0 Beta
2. Deleted all files from C:\Users\....\AppData\Local\cura\materials
3. Re-installed Cura 2.4.0 Beta (on my E: drive, not the C: drive - otherwise default installation)
4. Opened Cura 2.4.0 Beta
5. Duplicated the "Generic ABS" material
6. Edited the new changing:  Display Name, Brand, Color, Color Description, Diameter to 1.7, Print temp to 220C, Retraction to 3mm, Retraction speed to 25
7. Clicked "Close" on the material editor
8. MATERIAL IS THERE (in Cura's material list)
9. Closed Cura 2.4.0 Beta
10. Open Cura 2.4.0 Beta
11. MATERIAL IS GONE (Doesn't show in Cura's material list)
12. Closed Cura 2.4.0 Beta
13. Removed  "<compatible>True</compatible>" line from material file
14. Open Cura 2.4.0 Beta
15. MATERIAL IS THERE (in Cura's material list)
16. Selected the material
17. Closed Cura 2.4.0 Beta
18. Open Cura 2.4.0 Beta
19. MATERIAL IS THERE (in Cura's material list)
20. Edited material (added a character to the name)
21. Clicked "Close" on the material editor
22. Closed Cura 2.4.0 Beta
23. Checked material file and the "<compatible>True</compatible>" line is back
24. Open Cura 2.4.0 Beta
25. MATERIAL IS GONE (Doesn't show in Cura's material list)26.

StefanKi, 2016-12-29:
Just tried ppa:thopiekar/cura-master 1:2.4.99-master~201612281202~rev1647~pkg130~ubuntu16.04.1
Not working - materials are not saved.

What can I do to help you finding out the problem?
Should I send you my configs? (If yes - which ones?) Maybe the log from the terminal?

StefanKi, 2016-12-31:
I think i have found out something very useful!?

I added an Ultimaker 3 Extended just for fun, and suddenly had options for 2 extruders?
And now for each of this extruders suddenly i could see my custom materials!?!?

Now i can switch between 2 Printers:
1. my custom printer
2. Ultimaker 3 Extended

When switching to my custom printer, all custom materials are gone.
When switching to the Ultimaker 3 Extended, all custom materials are there.

How could that be? I didn't find any option for more than 1 extruder for my custom printer?
Maybe you only have to activate the more than one extruder options for the custom printers and the custom material bug would be gone...?

Ghostkeeper, 2017-01-02:
Did you make that custom printer with that "Custom FDM Printer" menu, or did you create a .def.json file manually?

StefanKi, 2017-01-02:
I did it over the menue.

Is there any docu about the json files with the 2 (or more?) extruders?

Ghostkeeper, 2017-01-03:
Not really. To produce a multi-extrusion printer definition, it's advisable to copy and rename all files related to the Ultimaker 3. Or Punchtec if you don't want to go through the trouble of creating all the quality profiles. But even I still make mistakes when I have to create a new multi-extrusion printer definition. It's quite complicated.

dtrazzi, 2017-01-07:
getting same problem here. Have a printer added as a Custom FDM, and added materials disappear once the program is closed and re-opened. Using Cura 2.3.1

agarthand, 2017-01-08:
Hope I can contribute to this issue...

Facing the same problem of disappearing Material (created via duplicate existing) and restarting Cura.
I'm using a Custom FDM printer and tried this with the Cura 2.4beta.

The interesting thing is that after creating additional printer instance of type "Ultimaker 3" and activate this I can see all the hidden material in (both) extruder material view for this printer. But as soon as I change back to my existing printer it's gone again.

Would love to see this fixed but for the time being I might have to manually customize the compatible line every time. Probably need to write a short script to automatize it ;)

matthock, 2017-01-10:
I'll add another voice on this. I am using 2.4beta, and I am seeing the exact same issue as described above. I tried the same test sequence as @dWoolridge and produced the same results.

Ghostkeeper, 2017-01-22:
We had marked this issue as fixed in 2.4, but I just put it back so that someone will take another look at it for version 2.5.

Ghostkeeper, 2017-01-23:
Our QA testers say they can't reproduce this on Windows or OSX. They tried with UM3 and UM3 Extended, with various materials.

StefanKi, 2017-01-23:
Yes it is working with the UM3.
But it is NOT working with the Custom Printers.
I don't know if the OS is important - I don't have any Windows or OSX to test that.

marekzehra, 2017-01-23:
Just tested OSX version 2.4.0-BETA2. Not working for me.
Steps to reproduce:

1. Add custom printer (In my case Kossel Mini)
2. Select Material->Manage materials...
3. Duplicate Generic PLA (rename, add brand, whatever...)
4. Repeat step 3 to create my set of custom materials (PLA, different colors, sizes, brands)
5. Quit Cura
6. Start Cura again
7. Select Material

Materials gone!

Ghostkeeper, 2017-01-23:
Thanks. The QA testers report that it does reproduce for other printers than UM3 and UM3E.

ntoff, 2017-01-25:
Same issue here on 2.4 beta (mendel90 printer) but if I manually create a material everything is fine, they only don't work if I use cura to generate the material. I've taken to just copy / pasting the material files and manually editing the contents, this results in usable material profiles.

Here's one of my custom materials that *does* work https://gist.github.com/ntoff/495604311240980108815c6bff6a2e54 I started generating random uuid's for them but found it doesn't really matter for my printer / usage (I guess only ultimakers make use of that?)

Chase-san, 2017-02-03:
Same thing here, removing the compatible tag from the file makes it work. For now I am setting the files to read-only, otherwise they revert (and don't show up in the menus).

Ghostkeeper, 2017-02-06:
Our bug tracker says that this is on schedule for fixing in 2.4, 2.5 ánd 2.6! I guess they find it really important... Or Jira is just a bit of a mess. 2.4 is locked, so I'd expect 2.5 then.

StefanKi, 2017-02-28:
For me the problem seems to be fixed in the latest ppa:thopiekar/cura-master.
Maybe otherones can confirm it?

MarcusWolschon, 2017-03-01:
I too can see my "nGen" material after upgrading from Cura 2.3 to 2.4

Ghostkeeper, 2017-03-01:
That's strange, because the issue is still unfixed according to our QA testers.

bubbapizza, 2017-03-01:
This problem still exists.  I haven't used Cura since the numbering scheme was changed and I will guarantee materials do not save properly if you install it on a brand new ubuntu 16.10 box and configure a custom machine.  I just found this thread by accident because I KNOW I configured some materials and they didn't get saved when I closed the app and restarted it.

Sure enough the files are in .local/share/cura/material and they have this line in them:

<setting key="hardware compatible">yes</setting>

bubbapizza, 2017-03-01:
Ohh, wait a sec, there's something wrong here and I think it's because there's no way to create new materials other than choosing "duplicate".  Who designed this anyways?

bubbapizza, 2017-03-01:
Ok, materials management IS SERIOUSLY BROKEN on ubuntu 16.10 and cura 2.4.  Brand new install, never used cura 2.4 before, so matter what changes you make, no materials ever get saved.  It's a mess.

ntoff, 2017-03-02:
@bubbapizza Try manually managing your materials, I agree the inbuilt system is far too broken to bother with, use this as a template: https://gist.github.com/ntoff/495604311240980108815c6bff6a2e54 and manually create your materials with a good text editor (or nano / vi if you prefer command line stuff). I've been making my own manually on windows and it works great.

fieldOfView, 2017-03-02:
> Who designed this anyways?

'sup?

bubbapizza, 2017-03-03:
Why is there no 'add' button for materials?  This is just a common-as-dirt CRUD database.  You need an add/update/delete/next/previous view to show the details of each material, then a browse view where you list the materials like a spreadsheet.  The "duplicate" button should be an option shown AFTER you choose "add".

StefanKi, 2017-03-03:
Feel free to change everything you want. It's open source...

bubbapizza, 2017-03-03:
This is true

Ghostkeeper, 2017-03-06:
I agree that it would be nice to have an "Add" button. We only implemented "Duplicate" because it was easier because you have a GUID and some profiles to start with, and due to time constraints.

However that is a different issue and should not be discussed in here.

LFATCGS, 2017-03-24:
Uggghh jeez, I was wondering why my printer wasn't working. Cura displays myXyzPLA as the material but when I click manage material - there is no myXyzPLA anymore! So it has been printing everything with 2.85mm instead of 1.75mm diameter and who knows how many other settings were lost - probably all of them. Poor form.

phaerus, 2017-03-28:
Can confirm issue still present in Windows 10 Professional, Cura 2.4. Using a custom printer, not Ultimaker. Removing the `<compatible>True</compatible>` line from the material files in `%localappdata%/cura/materials/` lets me view the material in the UI.

nallath, 2017-03-29:
A machine can be compatbile "in general", but a machine can have inter machine differences (in most cases; nozzle size or shape). We call those "variants". 

It can be the case that the machine in general supports it, but a single one of those variants doesn't. In that case you define the compatibility of that variant to false.

Kadah, 2017-04-08:
Ran into this as well. I'm new to Cura and 3d printing in general. I've barely touched Cura 2.x so far as I found the UI very difficult to navigate and use over all the other frontends I've tested.

BagelOrb, 2017-04-17:
Is this issue still not fixed?

fieldOfView, 2017-04-17:
I think it is fixed in at least 2.5b2. At least I could no longer reproduce it when I tested it a week or so ago.

seantapscott, 2017-04-17:
Custom materials work for me.  I can duplicate and edit them. 

I also checked that custom material export/import works (although the open/save dialog has a weirdly set default folder lookup).  Also, strangely, the display name of the imported material is the name of the file, and Cura hangs whenever I edit custom materials (probably a different issue entirely). 

- Using 2.4.0

Ghostkeeper, 2017-04-19:
Still seem to get this bug on Master though when I use the steps described in #1687.

MarcusWolschon, 2017-04-20:
Using 2.4.0 I created a new material "Colorfabb CF20".
Now I have that material still currently selected for the printer "Ultimaker 2ex 0.4mm nozzle" but it and it's category are no longer in "manage materials" and I can not e selected when changing to a different printer such as "Ultimaker 2ex 1.0mm nozzle".
The custom Material "Colorfabb nGen" that I used daily for many month before has completely disapeared from Cura.

MarcusWolschon, 2017-04-21:
After the update to Cura 2.5 all custom materials are still there... but each has 20-30 copies that didn't exist before the update. No keyboard navigation in the list and no way to select multiple  with shift. So I have to delete each one of them and it takes serveral seconds to delete a single entry each time.

nallath, 2017-04-24:
The cost + weight fillament is not a bug; You make duplicates. That means that they share GUID (so they are about the same material). A material has a single cost.

[23 more replies]

closed 2017-11-14 by ChrisTerBeke