Ultimaker/Cura #1105: Cura 2.3.0 100% Infill not showing and not printing.

Underp4ntz, 2016-11-03:
It seems Cura 2.3.0 doesn't want to show the 100% infill and it wont print it either way.
And Cura 2.1.3 does well with showing the infill and printing it.

Cura 2.3.0
[image]

Cura 2.1.3
[image]

Also the Dimensions of the model seems off. 

Dimensions in Cura 2.3.0: 33.5mm x 33.5mm x 9.0mm
Dimensions in Cura 2.1.3: 33.4mm x 33.4mm x 9.0mm

So whats up with that?

nickthetait, 2016-11-03:
Could you upload the model or point to where it was downloaded?

Ghostkeeper, 2016-11-04:
The infill can be explained. It is the Fill Small Gaps feature that has perished in version 2.3. We're planning to bring it back for version 2.4.

The dimensions are indeed different. It might be some change to the scene transformation engine we made for 2.3. I wonder which are the correct dimensions though?

Ghostkeeper, 2016-11-04:
The infill is a duplicate of https://github.com/Ultimaker/Cura/issues/1099

closed 2017-02-24 by Ghostkeeper