Ultimaker/Cura #1106: Font Kerning in End Gcode Text Box

ChristopherBonner, 2016-11-04:
The font kerning in the End Gcode text box is off. Letters are overlapping. Looks like it's mainly the letter "L" (lowercase) that is problematic.
[image]

Ghostkeeper, 2016-11-04:
Doesn't reproduce here on Ubuntu. What operating system are you using, Christopher?

ChristopherBonner, 2016-11-05:
Windows 10 Professional.
Cleartype is on (Windows has this on by default)

Interesting note: this issue seems to be tied very specifically to window size. If I resize the window by one or so pixels in width, the font kerning issue disappears. Will reappear if I resize to another size. Could be an even/odd scaling problem?

awhiemstra, 2016-11-07:
This is most likely an upstream bug. We do not touch the font rendering, this is all handled by Qt. It may very well be related to https://bugreports.qt.io/browse/QTBUG-46030 . If you have some time and a text editor it may be possible to play around with Qt's text rendering hints to get improved rendering.

nallath, 2017-07-27:
Closing this as it's upstream.

closed 2017-07-27 by nallath

fieldOfView, 2017-07-27:
Not only is it upstream, it was fixed by rounding the pixel boundaries, and by switching to another font (non proportional).