Ultimaker/Cura #1107: how about cura

robin1015, 2016-11-04:
Version: 2.1.99-master.20160930
Platform: Windows-7-6.1.7601-SP1
Qt: 5.5.1
PyQt: 5.5.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 61, in <module>
  File "D:\master\build\inst\lib\python3.4\site-packages\UM\Application.py", line 308, in getInstance
  File "D:\master\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 140, in __init__
  File "D:\master\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py", line 94, in __init__
  File "D:\master\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 107, in upgrade
  File "D:\master\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 281, in _upgradeFile
  File "C:\Python34\lib\codecs.py", line 319, in decode
MemoryError

Ghostkeeper, 2016-11-04:
Your computer does not have enough memory to run Cura, apparently. It's crashing in one of the first stages of the start-up process.

The "D" path is actually on our build server that created the executable you're running. In the final executable, there are no paths to the actual .py files any more since everything is packed inside the binaries. So it uses the original paths of the .py files on the build server.

robin1015, 2016-11-04:
[image]
how about it( not have enough memory)?please see pics upload by me.would you help me to solve the problem

fieldOfView, 2016-11-04:
Because none of us read chinese (?), here's the translation looking at my english Windows 7 task monitor:

Physical Memory (MB)
Total    3497
Cached  1534
Available 1854
Free       831

Kernel Memory (MB)
Paged  187
Nonpaed  92

robin1015, 2016-11-05:
yes,you are right.thank very much fieldOfView

robin1015, 2016-11-05:
I completely uninstall and then reinstall the Cura, but still not success

thopiekar, 2016-11-05:
@robin1015 To make a complete uninstallation, you'll need to remove the settings in AppData ($User/AppData/Local/cura) manually. The uninstaller only removes the application not the settings.

robin1015, 2016-11-05:
[image]
[image]

robin1015, 2016-11-05:
[image]

thopiekar, 2016-11-05:
Are you sure that your user account is called "CURRENT_USER"?
There really should be an "Local" directory in there. A lot of applications are using it to store their settings.

robin1015, 2016-11-10:
I'm not sure。
Why uninstall this software is so complex.
And are you sure that's the reason。

Ghostkeeper, 2016-11-11:
It's not *very* complex. The normal installer usually does the job good enough. But to completely remove all traces when uninstalling, we should probably find a way to modify the uninstaller a bit to optionally remove the preferences as well. Lots of uninstallers leave traces in the registry.

We're not sure that is the reason. We're just trying to debug via you, since we can't reproduce the problem on our end.

robin1015, 2016-11-17:
not found
“the settings in AppData ($User/AppData/Local/cura) manually”.try many times。

robin1015, 2016-11-18:
how about python34？would you tell me What is the meaning of the error message? What other possibilities?thank you

robin1015, 2016-11-18:
thopiekar，you are right

bananaslug123, 2016-12-04:
Hi.  I have encountered the same error.  No issues for the last few months, and suddenly a memory error:

Version: 2.3.1
Platform: Windows-7-6.1.7601-SP1
Qt: 5.5.1
PyQt: 5.5.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 61, in <module>
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Application.py", line 310, in getInstance
  File "D:\2.3\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 173, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py", line 94, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 107, in upgrade
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 281, in _upgradeFile
MemoryError

fieldOfView, 2016-12-04:
Please see if deleting cura.log from C:\Users\<your username>\AppData\Local\cura helps

bananaslug123, 2016-12-04:
Thanks for the suggestion.  I just tried and I'm experiencing the same error.

For what it's worth, the error first occurred on 2.1.99.  I installed 2.3.1, and the error is the same.

Ghostkeeper, 2016-12-05:
I recently added a guard around this particular line. It would try and open each file to see if it is a file that should be upgraded to the latest version, but some files, such as the log, can become too big to read. Let's hope the guard helps.

bananaslug123, 2016-12-05:
Hi.  Is the guard you added in v2.3.1?  Also could other files trigger this memory error?  My files are small, and this error occurred after 2-3 months of Cura use.

Thanks.

fieldOfView, 2016-12-05:
No, the safeguard will be in 2.4. Are there no large(ish) files in the AppData\Local\cura folder?

bananaslug123, 2016-12-06:
The largest file in that folder is cura.log which is 21KB.

fieldOfView, 2016-12-06:
@bananaslug123, are you also using a 32 bit version of Cura with 4Gb of memory?

bananaslug123, 2016-12-06:
Correct.

Ghostkeeper, 2016-12-07:
It's plausible that the rest of your applications you have open are using like 3.9Gb of that available memory, leaving Cura with an insufficient amount of memory within the first 4Gb of address space. If you have a 64-bit computer, it is always advisable to use the 64-bit version. If not, I suggest you close a few other applications and try again.

nallath, 2017-07-27:
Closing this, as this doesn't seem to be a Cura issue.

closed 2017-07-27 by nallath