Ultimaker/Cura #1108: Cura have problem with chose right printer

landway, 2016-11-04:
I have 2 printers - UM3 and UM3E. When both printers are ON and on network (both have assigned ip) I start CURA and I choose printer profile **UM3E**. In this moment cura say: Acces to the printer requested.... and on display of printer **UM3** 

show accesing window. Note: in CURA i see camera images from UM3 too.

So in short: When I select a profile for UM3E, CURA want to do access request for UM3 and that is the problem.

On card "printer monitor" I choose UM3E. When i try click on button "Request acess"  - CURA ask UM3 for acces request...

[image]
[image]

thopiekar, 2016-11-06:
Thank you for the report and sorry for the inconvenience.
This is now reported on JIRA..

nallath, 2016-11-07:
Could you post your logs?

Ghostkeeper, 2016-11-07:
When the UM3E is selected, only Ultimaker 3 Extended printers should appear in the printer discovery list where you connect to the printer.

landway, 2016-11-07:
logs are here

[cura.txt](https://github.com/Ultimaker/Cura/files/575965/cura.txt)
[stderr.txt](https://github.com/Ultimaker/Cura/files/575966/stderr.txt)

Ghostkeeper, 2016-11-08:
Would've been nice if we printed the remote version number to the log, so we could have identified which Bonjour service is which printer :stuck_out_tongue:

nallath, 2016-11-21:
As far as I can see from the logs, both printers connect just fine. The only thing I can come up with is that the printer is reporting the wrong printer type.

Could you also try it with 2.3.1 and see if the problem is still there?  Could you also check what {ip}/api/v1/system/variant returns for both machines?

nallath, 2017-07-27:
A number of changes have been made in the meantime. I think this problem no longer occurs, as it's something thats pretty well tested by our system testers. If it does occur, feel free to re-open / comment.

closed 2017-07-27 by nallath