Ultimaker/Cura #1110: After print I can't send another job unless I restart Cura.

partsoven, 2016-11-04:
If I print a job successfully (Ultimaker 3 Extended) and remove it from the build, then I click through the menu on the printer to make sure its ready to go again. However in Cura 2.3.0 it says:

"Unable to start a new print job because the printer is busy. Please check the printer."

If I close Cura and reopen it and reload my files I can then print. Wish I didn't have to close Cura and reopen every time I want to send a job to the printer.

thopiekar, 2016-11-06:
@partsoven Thank you for your report and sorry for the inconvenience.

thopiekar, 2016-11-06:
Now tracked on JIRA..

nallath, 2016-11-07:
Could you add your logs? Without them we can't help you (It seems to work just fine on my setup)

partsoven, 2016-11-07:
This help?

sedwards2009, 2016-11-08:
@partsoven I saw something in the log which suggested that you are using an old beta version of Cura or maybe you did use one and then later upgraded it. Is that possible?

I saw this:

`2016-11-05 10:23:04,677 - DEBUG - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (_backendLog [88]): [Backend] [WARNING] Couldn't load extruder.def.json file for extruder 1. Index out of bounds.`

The reference to `extruder.def.json` looks like the name for an old file which we renamed during development.

Ghostkeeper, 2016-11-08:
Actually, that file name is hardcoded in the warning message in the engine. We also get that error message at every slice in normal operation.

partsoven, 2016-11-14:
Yep. Much of the log is stuff that is preloaded.

nallath, 2016-11-17:
Could you check if this is still the case in cura 2.3.1? I can't find anything in the log to indicate a reason why this is happening.

partsoven, 2016-11-17:
Will do as soon as my UM3 is back up and running. The head shroud filled with material and is all jammed up.

partsoven, 2016-11-22:
Got my UM3 working (kinda)... but still having this issue with the 2.3.1 version

fieldOfView, 2016-11-22:
Just to get something obvious out of the way: do you confirm on the UM3 display that you have cleared the buildplate after printing?

partsoven, 2016-11-22:
Yes.

fieldOfView, 2016-11-22:
I don't have a UM3, so I cannot test or reproduce your issues.

In your logs I see a number of unexpected disconnects and bonjour service interruptions. Directly after uploading the gcode the printer seems to actively say goodbye, which is weird. I hope someone at UM can figure out what is going on.

Wifi or wired? Anything special about your network?

partsoven, 2016-11-23:
thanks. Wired network. Nothing special about my network.

fieldOfView, 2016-11-23:
What happens if after printing you reset the UM3? Does it "return" in Cura? I'm not suggesting this would be a solution, but at least we would know if Cura is still responsive to network events.

Another thing to try is adding another printer to Cura (without connecting it to your UM3), and switching to that and back. This forces a reconnection if Cura can find the printer.

partsoven, 2016-11-23:
If I reboot the machine it returns in Cura yes. Switching machines does not.

partsoven, 2016-11-23:
Deleted the printer and re-added it and reauthorized with Cura 2.3.1 and it seems to be working now...

partsoven, 2016-11-28:
Nevermind this bug is back again...

thopiekar, 2016-11-28:
I'm sorry to hear about this :disappointed: 

Which firmware version are you using, @partsoven ?
Our testers tried to reproduce it approx one week ago and they couldn't reproduce it. So any hint or further info about your setup would be useful , so we could find the reason for this behavior.

Thanks!

partsoven, 2016-11-28:
Cura 2.3.1 
Latest stable firmware. I'm printing now but I will followup after the print completes. Thanks for looking into this.

partsoven, 2016-11-28:
Firmware on printer says: 
3.4.16.20161104

partsoven, 2016-12-01:
Noticed I'm not the only one who is having this issue.  I haven't connected my printer to Wifi and only hardwired it. Could this be causing the issue. I will try after my print finishes to see if Wifi connection cures this.

partsoven, 2017-01-07:
I just realized I hardwired the machine but had not selected "Wired connection" on the UM3. So this may only effect wireless connectivity. Once I did this I am now able (so far) to start another job. I will update again on Monday if this bug persists.

Ghostkeeper, 2017-01-22:
Did this bug persist on the wired connection?

SunnyJames, 2017-05-13:
As far as I can tell this bug still exists in the newest version of Cura v2.5.0.  It doesn't matter for me if it is wired or wireless and I generally cannot predict whether or not Cura will let me print after clearing the print bed.  Switching printers and then switching back clears it up.  Restarting Cura also clears it.  

I'm running Mac OS Sierra 10.12.4

MFHFozzy, 2017-08-01:
Im having this bug. Every. Single. Print.
Wired connections. Cura 2.6.2.
Windows 10 x64.
Looking at the logs, i see this:
2017-08-01 14:14:17,575 - DEBUG - CuraEngineBackend.CuraEngineBackend._onSlicingFinishedMessage [474]: Slicing took 1.7409298419952393 seconds
2017-08-01 14:14:17,577 - DEBUG - UM.Backend.Backend._logSocketState [176]: Socket state changed to Closing
2017-08-01 14:14:17,578 - DEBUG - UM.Backend.Backend._onSocketError [198]: Socket debug: Arcus Error (13): Closing socket because other side requested close.
2017-08-01 14:14:17,579 - DEBUG - UM.Backend.Backend._logSocketState [178]: Socket state changed to Closed
2017-08-01 14:14:17,589 - DEBUG - CuraEngineBackend.CuraEngineBackend._onBackendQuit [585]: Backend quit with return code 0. Resetting process and socket.

tylerecouture, 2017-08-10:
I'm using Cura 2.6.2 AppImage on Mint 17.3 with a U3 Extended printing over Wifi.  After a print is done I click the button on the U3 to confirm print removal, then when I try to "print over network" again I get this: "Unable to start a new print job, printer is busy.  Current printer status is printing". Printer display is on the "Print/Material.../System" menu.

I have to restart Cura to be able to print again.  Not sure where the log file is to show.

Ghostkeeper, 2017-08-15:
I don't think there would be anything in the log, but nevertheless you can find the log with these instructions: https://github.com/Ultimaker/Cura#logging-issues

So far this issue doesn't reproduce here. I'm afraid that for an issue like this we really have to be able to reproduce it locally in order to begin debugging the problem.

MFHFozzy, 2017-08-16:
Are the socket errors that i posted not related?

MFHFozzy, 2017-08-16:
Ive just finished a print, and again, cannot start another one. I have not shut Cura down yet, but will be installing a beta version in a sec to see if this goes away and to test if the disappearing custom materials goes away.

So heres a full bug report for the network issue on 2.6 release as of 2017-08-15
Log file, CFG file and DX Diag attached.
[2017-08015-DxDiag.txt](https://github.com/Ultimaker/Cura/files/1227246/2017-08015-DxDiag.txt)
[cura.cfg.txt](https://github.com/Ultimaker/Cura/files/1227247/cura.cfg.txt)
[cura.zip](https://github.com/Ultimaker/Cura/files/1227248/cura.zip)

Application Version: 2.6

Platform:  Win10 X64 Pro
Qt:  QT5 from help about. No clue how to get more detailed on this for windows.
PyQt: No idea.
Display Driver: Nvidia game ready driver  384.76

Steps to Reproduce:
Connect to printer over wired Ethernet (Gigabit). Load STL file, slice, print. Do not abort.
Prints im using take ~1hour.

Actual Results:
Print finished successfully. Remove print from tray. Press button on front of Ultimaker3.
Load new STL file. Slice. Attempt to print over network. Get error that printer is busy printing.
Check that printer is NOT busy. 

Expected results:
Printer should print.

Additional Information:
Restart Cura and you can print again.

MFHFozzy, 2017-08-16:
Just successfully printed a 2nd print with beta 2.7 without having to close/re-open.
Also noticed that 2.7 actually had a notice for me to remove the print from the bed when it was done, and when i pushed the button the printer, the notice went away.

So there *was*/*is* a bug in 2.6 and its related to recognizing that the "print has been removed" button the ultimaker is pressed.

MFHFozzy, 2017-08-17:
False hope. Problem went away for exactly 1 print. Its back now, on every print.

partsoven, 2017-08-17:
Same here. This problem is such a pain.

nallath, 2017-08-18:
In the beta it should say "Unable to start a new print job, printer is busy. Current printer status is [STATUS]"
What is the status that it gives you?

Ghostkeeper, 2017-08-23:
> Are the socket errors that i posted not related?

No, those are about the connection to CuraEngine, not to the printer.

I'm fearing that this bug only occurs on certain LANs, which could be why it's not reproducing for us at work. Ultimaker's LAN is a bit... unconventional.

MFHFozzy, 2017-08-25:
Im pretty knowledgeable in networks, can i offer some assistance?
Or is there some extra logging i can enable for you?

nallath, 2017-08-28:
As far as logging goes; All logging that we have is enabled by default. But like i said, the message if this happens should give me some more info.

MFHFozzy, 2017-08-30:
Ill get you the exact message next time it happens.

MFHFozzy, 2017-08-31:
Ill get you the exact message next time it happens. Currently Cura is using soo much memory, Windows is killing it before the print is finished.

nallath, 2017-09-06:
I've fixed the memory leak for the next release. So there is that.

MFHFozzy, 2017-09-06:
Looking forward to it too :)

mfrank28, 2017-09-22:
I've been having the same problem for several releases.  I have a hardwired connection & one UM3.  The workaround for me was to re-select the UM3 in the printer drop-down menu.  Cura would take a few seconds to seemingly reconnect, then I would re-slice and be able to print.  With Cura 2.7 & the latest firmware, that workaround no longer works.  A fix would be great.  It's inconvenient to have to restart Cura after every print.  Definitely not the the user experience I'd expect from a polished UM3 / Cura combo.

partsoven, 2017-09-22:
Yeah still having the same issue. I confirm with latest release it is worse. I now can't just select another printer and reselect I also have to either restart cura or restart the printer. Please help this problem is very bad and makes the ultimaker not a worthwhile investment in a production environment.

bckline, 2017-09-22:
I just got an Ultimaker3 and after successfully printing a single item via hardwired network connection I am unable to print anything else. I receive the error "No access to print with this printer. Unable to send print job." every time I press "Print over network". I have restarted Cura and the printer itself multiple times to no avail. I definitely cannot (will not) rely solely on USB, so this issue may be a dealbreaker.

partsoven, 2017-09-22:
I agree bckline. Luckily for me I have several other printers which use Octoprint via Raspberry Pi's that live on the network and are able to print flawlessly. This bug is extremely frustrating especially since the problem is even worse now.

SunnyJames, 2017-09-23:
Agree with others.  This problem has been outstanding for well over a year and counting.  The current release made things significantly worse.

nallath, 2017-09-26:
The issue here is that I'm unable to reproduce this on my system. This might seem like a weak defence, but i need to have the problem in order to fix it (or well, it makes it a whole lot easier!).
Did you guys have the problem with 2.7? If this happens, could you please provide log files (please delete the log files before hand, so it's not cluttered with a gazillion entries)?
I totally agree that this is a major issue. So if you guys can help me, i can return the favour.

fieldOfView, 2017-09-26:
[image]

nallath, 2017-09-26:
I've added a ton of extra logging for the next release in hopes that it might shed some light onto the issue.

partsoven, 2017-09-26:
Thank you nallath. I will do whatever you need. Can you break down a step by step of what you want me to do?

partsoven, 2017-09-26:
Cleared my Cura log and printing some jobs now to recreate. Thanks again nallath.

nallath, 2017-09-27:
When you have the Cura 3.0; delete your log files, use cura as you would normally. If it breaks, send me the log file.

partsoven, 2017-09-27:
[cura.txt.log](https://github.com/Ultimaker/Cura/files/1337935/cura.txt.log)
[image]

nallath, 2017-09-28:
@partsoven; Next time this happens, could you then also look at http://192.168.1.28/api/v1/printer and http://192.168.1.28/api/v1/print_job and share the results with me?

fieldOfView, 2017-09-28:
Why does NetworkPrinterOutputDevice use self._printer_state instead of self._job_state? self._printer_state is only ever used by NetworkPrinterOutputDevice, and self._job_state seems to have the correct information (whereas self._printer_state apparently has the tendency to fluke out).

Note that when switching to self._job_state, check for "ready" instead of "idle".

partsoven, 2017-09-28:
http://192.168.1.28/api/v1/print_job:

{"bed": {"pre_heat": {"active": false}, "temperature": {"current": 22.8, "target": 0.0}}, "beep": {}, "diagnostics": {}, "heads": [{"acceleration": 3000.0, "extruders": [{"active_material": {"GUID": "e256615d-a04e-4f53-b311-114b90560af9", "guid": "e256615d-a04e-4f53-b311-114b90560af9", "length_remaining": -1.0}, "feeder": {"acceleration": 3000.0, "jerk": 5.0, "max_speed": 45.0}, "hotend": {"id": "AA 0.4", "offset": {"state": "valid", "x": 0.0, "y": 0.0, "z": 0.0}, "serial": "422a7d130000", "statistics": {"last_material_guid": "e256615d-a04e-4f53-b311-114b90560af9", "material_extruded": 1236940, "max_temperature_exposed": 261, "time_spent_hot": 9273360}, "temperature": {"current": 27.7, "target": 0.0}}}, {"active_material": {"GUID": "86a89ceb-4159-47f6-ab97-e9953803d70f", "guid": "86a89ceb-4159-47f6-ab97-e9953803d70f", "length_remaining": -1.0}, "feeder": {"acceleration": 3000.0, "jerk": 5.0, "max_speed": 45.0}, "hotend": {"id": "BB 0.4", "offset": {"state": "valid", "x": 0.0, "y": 0.0, "z": 1.6300000000000003}, "serial": "035473140000", "statistics": {"last_material_guid": "86a89ceb-4159-47f6-ab97-e9953803d70f", "material_extruded": 232870, "max_temperature_exposed": 232, "time_spent_hot": 6629280}, "temperature": {"current": 27.6, "target": 0.0}}}], "fan": 0.0, "jerk": {"x": 20.0, "y": 20.0, "z": 0.4}, "max_speed": {"x": 300.0, "y": 300.0, "z": 40.0}, "position": {"x": 0.0, "y": 0.0, "z": 0.0}}], "led": {"blink": {}, "brightness": 100.0, "hue": 0.0, "saturation": 0.0}, "network": {"ethernet": {"connected": true, "enabled": true}, "wifi": {"connected": false, "enabled": false, "ssid": "UM-NO-HOTSPOT-NAME-SET"}, "wifi_networks": []}, "status": "printing"}

nallath, 2017-09-28:
Is this after you pressed the "print removed"? It seems like that the printer didn't register this.

partsoven, 2017-09-28:
Now after pressing print removed and trying to send another job:

http://192.168.1.28/api/v1/print_job:

{"message": "Not found"}

http://192.168.1.28/api/v1/printer:

{"bed": {"pre_heat": {"active": false}, "temperature": {"current": 22.9, "target": 0.0}}, "beep": {}, "diagnostics": {}, "heads": [{"acceleration": 3000.0, "extruders": [{"active_material": {"GUID": "e256615d-a04e-4f53-b311-114b90560af9", "guid": "e256615d-a04e-4f53-b311-114b90560af9", "length_remaining": -1.0}, "feeder": {"acceleration": 3000.0, "jerk": 5.0, "max_speed": 45.0}, "hotend": {"id": "AA 0.4", "offset": {"state": "valid", "x": 0.0, "y": 0.0, "z": 0.0}, "serial": "422a7d130000", "statistics": {"last_material_guid": "e256615d-a04e-4f53-b311-114b90560af9", "material_extruded": 1236940, "max_temperature_exposed": 261, "time_spent_hot": 9273360}, "temperature": {"current": 27.7, "target": 0.0}}}, {"active_material": {"GUID": "86a89ceb-4159-47f6-ab97-e9953803d70f", "guid": "86a89ceb-4159-47f6-ab97-e9953803d70f", "length_remaining": -1.0}, "feeder": {"acceleration": 3000.0, "jerk": 5.0, "max_speed": 45.0}, "hotend": {"id": "BB 0.4", "offset": {"state": "valid", "x": 0.0, "y": 0.0, "z": 1.6300000000000003}, "serial": "035473140000", "statistics": {"last_material_guid": "86a89ceb-4159-47f6-ab97-e9953803d70f", "material_extruded": 232870, "max_temperature_exposed": 232, "time_spent_hot": 6629280}, "temperature": {"current": 27.6, "target": 0.0}}}], "fan": 0.0, "jerk": {"x": 20.0, "y": 20.0, "z": 0.4}, "max_speed": {"x": 300.0, "y": 300.0, "z": 40.0}, "position": {"x": 0.0, "y": 0.0, "z": 0.0}}], "led": {"blink": {}, "brightness": 100.0, "hue": 0.0, "saturation": 0.0}, "network": {"ethernet": {"connected": true, "enabled": true}, "wifi": {"connected": false, "enabled": false, "ssid": "UM-NO-HOTSPOT-NAME-SET"}, "wifi_networks": []}, "status": "idle"}

fieldOfView, 2017-09-28:
Could you try hacking a file?
C:\Program Files\Cura 2.7\plugins\UM3NetworkPrinting\NetworkPrinterOutputDevice.py
https://github.com/Ultimaker/Cura/blob/2.7/plugins/UM3NetworkPrinting/NetworkPrinterOutputDevice.py#L628

Change ```if self._printer_state not in ["idle", ""]:``` into ```if self._job_state not in ["ready", ""]:```. Take care that you do not change the indentation. I would be quite surprised if that does not fix the issue.

partsoven, 2017-09-28:
No problem, I hacked the file while Cura I will reboot Cura and try and run a few jobs now to see if it is fixed. Thanks for your attention on this.

[20 more replies]

closed 2018-01-22 by ChrisTerBeke