Ultimaker/Cura #1111: "Duplicate object" option should allow "multiply object" like previous Cura

partsoven, 2016-11-04:
I noticed in this version 2.3.0 of Cura that the "Multiply object" has been changed to "Duplicate object". If I am making 50 of the same part on the same build this becomes very time consuming to click this over and over. I liked the old version where you could type in exactly how many duplicates you wanted and it would populate them automatically.

nallath, 2016-11-07:
This is not a change in 2.3, this is a change in cura 2.x (2.1 also had a duplicate).

The issue is on our internal tracker, but we have no release assigned yet.

Ghostkeeper, 2016-11-07:
Duplicate (or multiplication?) of https://github.com/Ultimaker/Cura/issues/282

fieldOfView, 2016-11-07:
See https://github.com/Ultimaker/Cura/pull/1116

fieldOfView, 2016-11-09:
This one can be closed

closed 2016-11-09 by nallath