Ultimaker/Cura #1112: install 2.3.0 on debian 8.6 64-bit fail

harry36, 2016-11-04:
Version: 2.3.0
Platform: Linux-3.16.0-4-amd64-x86_64-with-debian-8.6
Qt: 5.5.1
PyQt: 5.4.2

Exception:
Traceback (most recent call last):
  File "/opt/cura/lib/python3/dist-packages/cura/CuraApplication.py", line 530, in event
    return super().event(event)
  File "/opt/cura/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 192, in event
    event._function_event.call()
  File "/opt/cura/lib/python3/dist-packages/UM/Event.py", line 157, in call
    self._function(*self._args, **self._kwargs)
  File "/opt/cura/lib/python3/dist-packages/UM/Signal.py", line 150, in emit
    func(dest, *args, **kwargs)
  File "/opt/cura/lib/python3/dist-packages/cura/BuildVolume.py", line 299, in _onStackChanged
    self._updateDisallowedAreas()
  File "/opt/cura/lib/python3/dist-packages/cura/BuildVolume.py", line 351, in _updateDisallowedAreas
    if ExtruderManager.getInstance().getResolveOrValue("prime_tower_enable") == True:
  File "/opt/cura/lib/python3/dist-packages/cura/Settings/ExtruderManager.py", line 411, in getResolveOrValue
    if user_container.hasProperty(key, "value") or quality_changes_container.hasProperty(key, "value"):
AttributeError: 'NoneType' object has no attribute 'hasProperty'

thopiekar, 2016-11-04:
@harry36 I guess you are using your Debian package?

thopiekar, 2016-11-06:
@harry36 Additionally, your log remembers me of a bug we had during 2.1 to 2.3 upgrade.
Can't remember about an different solution than removing the configs.

nallath, 2017-07-27:
Closing this due to inactivity

closed 2017-07-27 by nallath