Ultimaker/Cura #1114: Cura Slicing Issue. Cura 2.3.0

jnmbhj, 2016-11-06:
I installed a fresh new version of Cura and the program would not slice the model. It is stuck on the "Slicing..." The bar would not move indicating progress. I am using a Custom FDM printer profile. 
[image]

thopiekar, 2016-11-06:
Hey @jnmbhj,
thank you for the report.

Does it work without any changes?

jnmbhj, 2016-11-06:
@thopiekar 

Unfortunately it does not.

thopiekar, 2016-11-06:
Hmm, can't reproduce it here. I added a new printer here using "Custom FDM printer" and sliced an STL just fine.
The following steps might be useful to reproduce the problem here:
1. Make a zip of your current settings at C:\Users\$USERNAME\AppData\Local\cura (the whole content).
2. Remove the current settings.
3. Retry while passing the parameters of your printer.
4. If the problem still exists, make another zip and upload them here.

Thanks and good luck!

jnmbhj, 2016-11-06:
[cura.zip](https://github.com/Ultimaker/Cura/files/574001/cura.zip)

I have tried multiple times to delete the old files and reload cura. I also uninstalled cura, removed settings and did a fresh install after a reboot. No success there either.

thopiekar, 2016-11-06:
Ok, that's strange.. From the logs I see:
```
2016-11-06 12:35:05,731 - ERROR - D:\2.3\build\inst\lib\python3.4\site-packages\UM\Backend\Backend.py (startEngine [55]): Unable to find backend executable: C:\Program Files\Cura 2.3\CuraEngine.exe
```

Could you tell me more about your installation? Are you using a different location then the default to install Cura? Could you try to find CuraEngine.exe at the location where you've installed Cura?

Thank you and sorry for the inconveniences.

jnmbhj, 2016-11-06:
Huh, thats strange. The D drive is a partition of the main drive. I don't know why it will be located there. I choose the C drive for the installation.

thopiekar, 2016-11-06:
Don't mind the ```D:/``` in the path. While building the executable (Cura.exe) it takes the path where we built Cura on our build server. So when executing the Cura.exe after the build process the ```D:/``` still appears in the logs even if it does not exist and Cura is installed into another location 😉

thopiekar, 2016-11-06:
In short nothing to worry about 😄 

But could you confirm that ```C:\Program Files\Cura 2.3\CuraEngine.exe``` exists?

jnmbhj, 2016-11-06:
C:\Program Files\Cura 2.3 

CuraEngine.exe does not exist. 

Cura.exe does

thopiekar, 2016-11-06:
Ok, this is really strange. Is it possible that your anti-virus application accidentally thinks it is an virus?
I personally don't see a reason why CuraEngine is missing here 😕

jnmbhj, 2016-11-06:
Thats what I thought at first, but upon looking at the anti virus logs. Nothing triggered an action. It did scan the downloaded and extracted files and passed.

thopiekar, 2016-11-06:
Hmm, when has the installer been downloaded? Is there a chance you could upload it somewhere?
I could take a deeper look at the installer and maybe I or someone of my colleagues could track why CuraEngine is missing here.

jnmbhj, 2016-11-06:
The installer was downloaded about 5 hours ago. I find it strange I tested this on another computer and it works fine. I just cant get this to work on my computer. I think it might be a possible configuration issue. Do you want my computer setup?

thopiekar, 2016-11-06:
If you mean with "computer setup" more details, like Windows version, etc. then yes please.
But add it to the first post above, so others can find it easily 😉

thopiekar, 2016-11-06:
@jnmbhj I'm sure someone can give you tomorrow feedback on this, eg. whether it is a known problem or not.
Have you tried to disable your anti-virus and other security applications?

thopiekar, 2016-11-06:
Oh, another thing you could try is to unpack the installer with 7zip and see what happens. Eg. whether it gets removed again or not.
Also try to move it to the same location where Cura.exe is. 😉

jnmbhj, 2016-11-06:
Yes I have tried disabling all anti virus software but not success. I also made sure the cura.exe is also in the same folder as curaengine.exe

thopiekar, 2016-11-06:
So now there is CuraEngine and Cura still can't find it?
I'm a bit confused now..

jnmbhj, 2016-11-06:
[image]

thopiekar, 2016-11-06:
What is C:\Users\$USERNAME\AppData\Local\cura\cura.log saying now?

jnmbhj, 2016-11-06:
Its still saying can't find C:\Program Files\Cura 2.3\CuraEngine.exe

jnmbhj, 2016-11-07:
[cura.zip](https://github.com/Ultimaker/Cura/files/574231/cura.zip)

Ghostkeeper, 2016-11-07:
> @jnmbhj I'm sure someone can give you tomorrow feedback on this, eg. whether it is a known problem or not.

Not a known problem as far as I'm aware. I have no idea what is happening with jnmbhj's Cura :(

jnmbhj, 2016-11-07:
I am not sure why it is not executing properly. Can you verify that this CuraEngine.Exe is functional? 
[CuraEngine.zip](https://github.com/Ultimaker/Cura/files/576800/CuraEngine.zip)

jnmbhj, 2016-11-08:
The first time I tried to install Cura. I tried installing it to my E: Drive. It had the same issue, so I deleted it off the E drive scanned for all remaining Cura files and deleted them also. I then proceeded to install it in the C: Drive.

jnmbhj, 2016-11-09:
I updated to newer version 2.3.1. Issue remains, checked anti-virus quarantine, nothing there. Checked file location. Checked the run protocol of  CuraEngineBackend.py in notepad++. Correct. I also tried installing it on another drive. It also did not work. Giving the error | Unable to find backend executable: E:\Program Files\Cura 2.3\CuraEngine.exe | 

But the executable is valid.

Ghostkeeper, 2016-11-11:
I've verified that the CuraEngine executable you supplied previously is identical to the executable produced by the Windows 64 build of 2.3.0. And that executable works for me on Windows 8.1.

closed 2017-07-27 by nallath

Ghostkeeper, 2017-08-10:
We weren't able to reproduce this bug. Has it been solved in the meanwhile, @jnmbhj, or are you still not able to slice?