Ultimaker/Cura #1115: openSUSE 42.1, Cura 2.3.1, Segfaults on start (self build rpm package)

heinervdm, 2016-11-07:
I've tried to package to latest version of cura for openSUSE 42.1 as up to now I could only use the legacy version.

My obs repo can be found here: https://build.opensuse.org/project/show/home:Heinervdm:cura

I've packaged the following versions:
protobuf 3.1.0 [buildlog](https://build.opensuse.org/build/home:Heinervdm:cura/openSUSE_Leap_42.1/x86_64/protobuf/_log)
gmock 1.7.0 [buildlog](https://build.opensuse.org/build/home:Heinervdm:cura/openSUSE_Leap_42.1/x86_64/gmock/_log)
libArcus 2.3.0 (no tag for 2.3.1 available) [buildlog](https://build.opensuse.org/build/home:Heinervdm:cura/openSUSE_Leap_42.1/x86_64/libArcus/_log)
uranium 2.3.1 [buildlog](https://build.opensuse.org/build/home:Heinervdm:cura/openSUSE_Leap_42.1/x86_64/uranium/_log)
cura-engine 2.3.1 [buildlog](https://build.opensuse.org/build/home:Heinervdm:cura/openSUSE_Leap_42.1/x86_64/cura-engine/_log)
cura 2.3.1 [buildlog](https://build.opensuse.org/build/home:Heinervdm:cura/openSUSE_Leap_42.1/x86_64/cura/_log)

I've used:
cmake 3.3
phyton 3.4
gcc 4.8.5 (5.2.1 for cura-engine)

When I run cura I'm only getting the attached backtrace.

I would be thankful for any hint how to fix this.

Thanks

heinervdm, 2016-11-07:
Forgot to attach the backtrace...
[cura-bt.txt](https://github.com/Ultimaker/Cura/files/574879/cura-bt.txt)

awhiemstra, 2016-11-07:
Hmm, I do remember seeing that same backtrace. It is caused by mixing libstdc++ versions. In this case, I suspect it is because libArcus was built with a different libstdc++ and statically links that libstdc++. Please try again with the master version of libArcus, or manually remove line 38-40 from CMakeLists.txt .

Ghostkeeper, 2016-11-07:
I added a tag for libArcus. Must've forgotten that on libArcus' repository, sorry. Should be the same though. Not much has changed.

heinervdm, 2016-11-07:
Updating libArcus to the newly tagged version 2.3.1 solved the issue.
Thank you.

closed 2016-11-07 by heinervdm