Ultimaker/Cura #1117: OpenGL won't initialize

heinervdm, 2016-11-07:
When I try to run Cura 2.3.1 the following exception is repeating forever:

> 2016-11-07 12:20:33,091 - CRITICAL - .../cura/CrashHandler.py (show [24]): An uncaught exception has occurred!
> 2016-11-07 12:20:33,091 - CRITICAL - .../cura/CrashHandler.py (show [24]): Traceback (most recent call last):
> 2016-11-07 12:20:33,091 - CRITICAL - .../cura/CrashHandler.py (show [24]):   File "/usr/lib/python3.4/site-packages/UM/Qt/Bindings/MainWindow.py", line 177, in _render
> 2016-11-07 12:20:33,091 - CRITICAL - .../cura/CrashHandler.py (show [24]):     renderer.beginRendering()
> 2016-11-07 12:20:33,092 - CRITICAL - .../cura/CrashHandler.py (show [24]):   File "/usr/lib/python3.4/site-packages/UM/Qt/QtRenderer.py", line 115, in beginRendering
> 2016-11-07 12:20:33,092 - CRITICAL - .../cura/CrashHandler.py (show [24]):     self._initialize()
> 2016-11-07 12:20:33,092 - CRITICAL - .../cura/CrashHandler.py (show [24]):   File "/usr/lib/python3.4/site-packages/UM/Qt/QtRenderer.py", line 169, in _initialize
> 2016-11-07 12:20:33,092 - CRITICAL - .../cura/CrashHandler.py (show [24]):     OpenGL.setInstance(QtOpenGL())
> 2016-11-07 12:20:33,092 - CRITICAL - .../cura/CrashHandler.py (show [24]):   File "/usr/lib/python3.4/site-packages/UM/Qt/GL/QtOpenGL.py", line 52, in __init__
> 2016-11-07 12:20:33,093 - CRITICAL - .../cura/CrashHandler.py (show [24]):     self._gpu_type = self._gl.glGetString(self._gl.GL_RENDERER)
> 2016-11-07 12:20:33,093 - CRITICAL - .../cura/CrashHandler.py (show [24]): UnicodeDecodeError: 'ascii' codec can't decode byte 0xc2 in position 21: ordinal not in range(128)

My GL_RENDERE is: 
> Mesa DRI Mobile Intel® GM45 Express Chipset
Character 21 is the ®...
Any suggestion how to fix this?

Thanks

thopiekar, 2016-11-07:
What the... ?! I've added the labels and GitHub says it was @heinervdm?!

heinervdm, 2016-11-08:
I cant reproduce this error with a minimal example, therefore I worked around the issue by catching the exception:
`diff -Nurd Uranium-2.3.1.orig/UM/Qt/GL/QtOpenGL.py Uranium-2.3.1/UM/Qt/GL/QtOpenGL.py`
`--- Uranium-2.3.1.orig/UM/Qt/GL/QtOpenGL.py     2016-11-06 11:15:37.000000000 +0100`
`+++ Uranium-2.3.1/UM/Qt/GL/QtOpenGL.py  2016-11-08 09:52:39.426336669 +0100`
`@@ -49,7 +49,11 @@`
`         elif "intel" in vendor_string:`
`             self._gpu_vendor = OpenGL.Vendor.Intel`
` `
`-        self._gpu_type = self._gl.glGetString(self._gl.GL_RENDERER)`
`+        try:`
`+            gpu_type = self._gl.glGetString(self._gl.GL_RENDERER)`
`+        except UnicodeDecodeError:`
`+            gpu_type = "Unkown"`
`+        self._gpu_type = gpu_type`
` `
`         if not self.hasFrameBufferObjects():`
`             Logger.log("w", "No frame buffer support, falling back to texture copies.")`

My minimal example was:
> from OpenGL.GLUT import *
> from OpenGL.GLU import *
> from OpenGL.GL import *
> glutInit()
> glutInitDisplayMode(GLUT_DOUBLE | GLUT_RGB | GLUT_DEPTH)
> glutInitWindowSize(200,200)
> glutCreateWindow("test")
> glGetString(GL_RENDERER)

and gives me:
> b'Mesa DRI Mobile Intel\xc2\xae GM45 Express Chipset '

without conversion error...

awhiemstra, 2016-11-08:
Haha, this is a quite silly one. I certainly never expected special characters in the renderer string.

> I cant reproduce this error with a minimal example,

That's because Python OpenGL treats it as a bytearray whereas PyQt treats it as a const char* which gets converted to a Python str object. This internally tries to convert the bytearray to an unicode string and fails.

sedwards2009, 2016-11-08:
Looks like the byte to string conversion just needs to be told to expect UTF8 instead of 7bit ASCII.

printerguykw, 2016-11-23:
applying @heinervdm 's try/catch code snippet resolved my issue and cura was able to run OpenGL

heinervdm, 2016-11-23:
This try and catch snippet is working, because it just ignores the error and as this information is not needed somewhere else it still works, but it's not the nice way to solve this issue :-)
The correct solution is the one of @sedwards2009, but I don't know how to do this.

Ghostkeeper, 2016-11-23:
In the PyQt source, the `glGetString` function is said to return `const char *`. Basically the only place where it gets mentioned is this:

        const char *glGetString(GLenum name);
    %MethodCode
            sipRes = reinterpret_cast<const char *>(sipCpp->glGetString(a0));
    %End

And then four times (once for each of the OpenGL versions it seems to support). How this C-string gets converted to Python's string by SIP, I don't know. But in any case, this seems to be an upstream bug. Perhaps heinervdm's patch is about the best we can do.

gvJaime, 2016-11-23:
GM45 Intel chipset user here.

@heinervdm allows me to use Cura normally. It starts and renders correctly.

thopiekar, 2016-11-23:
@Ghostkeeper: The only thing I would add to the code is a "TODO:" or "WORKAROUND:" tag including a upstream report.
Otherwise I will get lost whether this code is still needed or not.

closed 2017-01-18 by jackha

heinervdm, 2017-05-17:
This Bug is Back in 2.5.0
It seems that the workaround got lost during the re-factorization...

Ghostkeeper, 2017-05-18:
I have no idea what happened there.

[The original commit that fixed this issue](https://github.com/Ultimaker/Uranium/commit/8a00a24742387fc554cdd7bb16184526da079aa0) says it's on the Master branch, but it's not actually in [the history of that file](https://github.com/Ultimaker/Uranium/commits/master/UM/Qt/GL/QtOpenGL.py)! In any case I re-applied the fix: https://github.com/Ultimaker/Uranium/commit/824faefb6f8ce8736fc8cfe1cc03485b8082a79a

thopiekar, 2017-05-18:
Thanks for killing my commit 😜