Ultimaker/Cura #1119: Z-hop height is hidden

reapingminer, 2016-11-08:
Z-hop height does not show under advanced print settings, even if the checkbox is ticked in the cura settings menu.

reapingminer, 2016-11-08:
Strange, it just appeared after changing some other settings. (Not related to Z-hop)

Ghostkeeper, 2016-11-09:
The `Z Hop Height` setting should only be visible if `Z Hop when Retracted` is on and `Enable Retraction` is on. Also, `Z Hop when Retracted` is visible only if `Enable Retraction` is on.

If any of that is not the case, that's a bug.

reapingminer, 2016-11-09:
I am fairly sure Z-hop when retracted was ticked. I ticked/unticked it a couple of times to see if i could get the Z hop height to show but i couldnt. I just noticed the option was there after i finished setting all other settings as my final check of the settings about 2 minutes later.

From: Ghostkeeper 
Sent: Wednesday, November 09, 2016 10:47 AM
To: Ultimaker/Cura 
Cc: reapingminer ; Author 
Subject: Re: [Ultimaker/Cura] Z-hop height is hidden (#1119)

The Z Hop Height setting should only be visible if Z Hop when Retracted is on and Enable Retraction is on. Also, Z Hop when Retracted is visible only if Enable Retraction is on.

If any of that is not the case, that's a bug.

Ghostkeeper, 2016-11-10:
I can't reproduce it. Do you remember what machine/material/quality you were using?

uwe999, 2017-04-11:
I ran into the same issue today (cura 2.4).
I enabled "z hop only over printed parts" and "z hop height" but it wouldn't show up.
I had to enable "z hop when retracted" before those options would show.
The "Settings visibility" box should indent those 2 options to make it obvious that they are only available when the main "z hop when retracted" has been checked.

closed 2018-02-27 by ianpaschal