Ultimaker/Cura #1120: Frozen on "Slicing... because CuraEngine.exe is a virus, apparently.

sanworks, 2016-11-08:
Hi Cura team

Panda Free Antivirus seems to think CuraEngine.exe is a virus, as of Cura 2.3.1
I'm using Cura on Win7, 64 bit.
On install, CuraEngine.exe was quarantined. I didn't think much of it, and ran Cura. It opened, and I could change settings and load .stl objects - but the software would never get past "Slicing...", with no progress shown on the progress indicator, and no visible error message.
I un-quarantined CuraEngine, restarted Cura and slicing was fixed.
Earlier versions of Cura install without issue on my setup.
Anyhow, it may be a good idea to figure out what part of the new code is creeping out the virus scanners!

Cheers
Josh

Ghostkeeper, 2016-11-09:
Perhaps we can test this by uploading CuraEngine.exe to https://www.virustotal.com/ or some similar online service. That allows us to test for various virus scanners without needing to install the antivirus software ourselves.

There are no code changes in CuraEngine between version 2.3.0 and 2.3.1. There was one commit made, but it was reverted. So any change in the behaviour on the part of your virus scanner is either:
* Cura's front-end behaves differently, and Panda identifies the wrong executable as being the cause, or
* There is an actual virus in CuraEngine that was injected in the binary after compilation. I verified that the binary that ultimaker.com serves is the same binary as what the Cura team delivers and I trust the Cura team, so I would suspect that the problem is on your side, that something on your local computer is injecting assembly, or
* It is simply a false positive.

I'd say that the last possibility is the most likely one, and in second place the first possibility.

FlameRunner, 2016-11-09:
Avast also was super cautious.  The file was sent for analysis and stopped from running. Today results came back - clean.

IIRC, Comodo right away assumed the file to be safe.

Maybe the AV makers just need a day or two?

Ghostkeeper, 2017-08-10:
This is still a problem; it gets reported on Ultimaker's forum from time to time.

ChrisTerBeke, 2018-01-22:
Might the new binary signing for Windows fix this?

Ghostkeeper, 2018-02-08:
It is still getting reported a lot in the 3.2 forum thread: https://community.ultimaker.com/topic/21382-welcome-to-ultimaker-cura-32-stable/

closed 2018-11-26 by Appesteijn