Ultimaker/Cura #1121: Missing Z lift on new layer

Goldi80r, 2016-11-09:
Want to print a small ring and on every layer i need to remove nozzle up for Z 200mm every layer

regards

Ghostkeeper, 2016-11-09:
You wish to move the nozzle up 20cm after every layer, and then back down again? There's a few ways you could approach this:
* Under the Cooling category, set the Minimum Layer Time and Minimum Speed higher, and then enable the "Lift Head" setting. It's not 200mm, but might be what you intend. It moves the nozzle away a bit to allow the layer to cool down.
* Write your own post-processing script that adds the Z-moves at every layer. It requires some technical know-how though, and you need to be careful not to go outside the bounds of your printer volume.

May I ask why you want something like this?

Goldi80r, 2016-11-09:
Is very simply you can make small prints just need to move from part every
slice
Pls put it on new cura .printstudio slice3r simply3d have it but I like
cura.

My nozzle is 0.1 25mm conical I don't heat pla but is steal hot to much I
used cura 15.4 some plugins to retract Z. Making highest level prints fdm
is testing but I'm close 100mm/s is speed
Whit retract Z I can test my SLA laser

Is very good to have

closed 2018-02-27 by ianpaschal