Ultimaker/Cura #1124: Pause - Post Processing Plugin caused error

doncrush, 2016-11-09:
I got a message that the print head tried to move outside of acceptable range or something to that effect.

I suspect it has to do with the 190 mm default entry for Park Print Head X and Y?

thopiekar, 2016-11-09:
Can you tell us something about your printer? Cura version? Operating system?

Thanks!

Ghostkeeper, 2016-11-10:
Perhaps we can park the print head by default on coordinate 0,0 or else just `machine_width - 10` or something.

doncrush, 2016-11-14:
UM2Go, latest Cura 2.3.1 on Win10.

fieldOfView, 2016-11-14:
machine_width-10 would be better than hitting the endstops at 0,0. Say the axes have slipped a fraction of a mm; homing the head halfway through the print would then leave a visible line in the print.

Ghostkeeper, 2016-11-14:
Or `10,10` then.

ianpaschal, 2018-05-26:
Very old, never got required reproducibility info.

closed 2018-05-26 by ianpaschal