Ultimaker/Cura #1128: feature request "non"-Material for disabling extruder train on dual setup

Drayson80, 2016-11-10:
Requirement: 
A "non"-material shall be introduced to disable not used extruder trains in an dual/multi extruder setup

Explaination:
When printing just with one extruder train on a dual-setup printer (in this case UMO), the second nozzle also get heat up and primed, independend if in use/selected for a part or not. 
This gives a bad print quality as it dips material as  it never sets the unused extruder train to standby temperature or cools down - as it never get used...
Only workaround is currently to add an additional printer with an singe extruder train.
This feature would also enable the selection of the used extruder train in case different materials are loaded...

Ghostkeeper, 2016-11-10:
It should not be heating up the second nozzle if the nozzle is unused. I propose that we fix this bug at the root: In the engine.

Ghostkeeper, 2016-11-10:
I could reproduce this using the UMO Dual Extrusion. We should have no `M104 T1 S205` at the g-code header there.

Drayson80, 2016-11-10:
Thanks for confirming. Hope it is easy to fix - so if you need a tester keep me informed :-)

Drayson80, 2016-11-10:
@ Ghostkleeper, just to understand it correctly, "we should have no M104 T1 S205 at...." would also mean that there have to be different machine_start_gcode / machine_end_gcode for single/dual setup similar to Cura 15.04 - right?
Else (with just the start/end from the dual setup) it would switch to a cold extruder and trying to prime it...

fieldOfView, 2016-11-10:
There is no M104 in the UMO dual definition start gcode; it gets added automatically by the engine. It has to be added, because priming the extruders is done in the start gcode. The start gcode is used regardless of whether the print uses both extruders or not.

Drayson80, 2016-11-11:
Sorry, but there is T1 in... 
You're right, the initial heatup code is added by the engine, but there is the need of priming.
Pls. ref. to line 67 in ultimaker_original_dual.def.json, machine_start_code, there is written (I made it a bit more readable):
[...]
\nT1 ;Switch to the 2nd extruder
\nG92 E0 ;zero the extruded length
\nG1 F200 E6 ;extrude 6 mm of feed stock
\nG92 E0 ;zero the extruded length again
\nG1 F200 E-{switch_extruder_retraction_amount}
\nT0 ;Switch to the 1st extruder
\nG92 E0 ;zero the extruded length
\nG1 F200 E6 ;extrude 6 mm of feed stock
\nG92 E0 ;zero the extruded length again
\nG1 F9000\n;Put printing message on LCD screen\
nM117 Printing..."
[...]

And - b.t.w. - at the end code there is also a retract for T1 (but not for T0..?)

Ghostkeeper, 2016-11-11:
It would also need to make sure not to prime the unused extruder...

Drayson80, 2016-11-13:
Basically, it has to have the same selective functionality like legacy cura - if single extruder, then start/end code 1, if dual, start/end code 2 - best case with a transition code so that intermediate actions could be added too - e.g. move to x/y, move servo, push nozzle lift activator lever... or whatever... :-)

Drayson80, 2017-02-21:
@Ghostkeeper @fieldOfView 
Any news on this bug/issue??

fieldOfView, 2017-02-21:
AFAIK, Cura 2.4 (just released) should be more intelligent about the extruders that are actually being used.

Drayson80, 2017-02-21:
Downloaded, installed and tried... 
2.4 has the same issue...
Heat&primes both trains, does not cool down the unused one (even not to standby temp), and shuts down both at the end...

Drayson80, 2017-02-22:
@Ghostkeeper @BagelOrb, is there any way to get any indication if this bug is on the fixing list or even if it will be addressed... or when...?

BagelOrb, 2017-02-22:
Whow long topic!

Afaik this can only occur in 2.4 if you have support enabled on the second extruder but only so small a support structure is generated that it's too small to print.

If you are using a different printer than the UM3 than the problem might be in the machine definition.

If your problem is with an UM3 and it's not the support issue then please attach your .curaproject.3mf file (rename it to .Text for github to accept it)

Drayson80, 2017-02-22:
@BagelOrb: Sorry that I forgot to mention that it's in my case an UMO with Dual Extrusion.
Support has been disabled and prior set to first extruder, but the outcome is the same...
Here the first lines (of course, from G21 it´s the start.gcode...)

;FLAVOR:RepRap
;TIME:765
;Filament used: 0.133397m
;Layer height: 0.1
;Generated with Cura_SteamEngine 2.4.0
M190 S60
M104 S205
M104 T1 S240
M109 S205
M109 T1 S240
G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off

Regarding start.gcode it´s another issue that there is no possibility to have a prime code extruder-dependend (similar to 15.04.)

Ghostkeeper, 2017-02-23:
I reproduced it earlier. It indeed heats up unused extruders for the UMO Dual Extrusion, where the same set-up doesn't have the issue with the UM3.

The bug is filed (CURA-2950), and it's been put in the Some Future Release thing that basically means we won't fix this anytime soon.

fieldOfView, 2017-02-23:
Isn't the reason obvious? The UMO dual definition has a start gcode that heats up both hotends. For the UM3 there is no start gcode because the firmware takes care of preparing the hotends.

willyt, 2017-02-23:
No the M104 and M109 gcode instructions that heat the hot ends are added by cura engine before the start gcode, unless your start gcode already contains the M104 or M109 instructions. If a nozzle is not used Cura engine should set the temperature of the unused nozzle to 0 straight after the start.gcode. Or there should be some way to insert conditional code into start.gcode to make it possible to ignore the unused nozzle. i.e. don't omit heating and priming instructions for the unused nozzle.

Drayson80, 2017-02-24:
@Ghostkeeper "The bug is filed (CURA-2950), and it's been put in the Some Future Release thing that basically means we won't fix this anytime soon."

So basically Cura 2.4 and later is not usable for UMO dual or every other 3rd party printer having more than one nozzle when using in single print mode...

Ghostkeeper, 2017-02-24:
> So basically Cura 2.4 and later is not usable for UMO dual or every other 3rd party printer having more than one nozzle when using in single print mode...

Yes, and there's nothing I can really do about that :disappointed: I dislike that Ultimaker only wants to fix bugs that apply to their own printers, since it reduces the feeling of free software. Though of course it is understandable that they want to maintain an edge over other printers via Cura. UMO Dual Extrusion is not sold any more and other printers don't add to Ultimaker's profit margin either. That's the reason. I'm just communicating the decision back to Github as well in an effort to make the process more open.

fieldOfView, 2017-02-24:
Here's to hoping that it will be fixed with a community contribution. Perhaps it could even be done with a postprocessing plugin (though it should really be fixed in CuraEngine I guess).

Drayson80, 2017-02-24:
Agree... unfortunaltely my programming skills are not existing so also the postprocessing is no option to go for me - but I would be a available as a tester :-)

Drayson80, 2017-02-24:
Ah.. regarding the postprocessor: I already had the idea to modify the "LayerView" plugin to put "T1" in instead of the layer number to have at least the extruder train selectable in a single-train-machine-workaround. Nevertheless, I have nobody around be able to program python... :-(

BagelOrb, 2017-02-28:
> So basically Cura 2.4 and later is not usable for UMO dual or every other 3rd party printer having more than one nozzle when using in single print mode...

I disagree. It works, but it heats the unused nozzle. That's not really nice, but it shouldn't make your print fail.

Drayson80, 2017-02-28:
au contraire, mon amis...
if the second nozzle is oozing, it makes the print fail... and worst case clog the unused nozzle because of burned filament (e.g. PVA)
But anyway... letz call it a feature - it safeguards that the unused nozzle is always ready to go in case a fast material change is necessary :-)

BagelOrb, 2017-02-28:
Alright.

PS: Note that for the UMO dual extrusion you can always select a normal UMO
if you are doing single extrusion prints ;)

justinmrkva, 2017-04-17:
I'm having the same issue on an Ultimaker 3 with Cura 2.4; the Enable Support option is *disabled* and the second extruder still gets primed. Every time. The firmware is on the latest version.

People are saying that this problem doesn't occur on an Ultimaker, but it definitely does.

BagelOrb, 2017-04-18:
I think this is fixed in Cura 2.5 beta 2. Could you check out that version?

Maybe make a backup of your config folder if you want to go back to 2.4...

maukcc, 2017-05-01:
Also if this is the start of the gcode:

M190 S60
M104 S205
M104 T1 S240
M109 S205
M109 T1 S240

And your active extruder is T1(from an earlier print) then T0 does not get heated at all, and the print will definitely fail.
It needs to be :
M190 S60
M104 T0 S205
M104 T1 S240
M109 T0 S205
M109 T1 S240

Also at the end of the file when cooling down

Drayson80, 2017-05-02:
I also recognized this. When printing something with T1, and the next print shall be done with T0, it will not switch automatically back to T0, but heats up the last used train (T1). So I put a T0 at the very end of my standard end code just to be safe. But that can´t be a standard....

maukcc, 2017-05-02:
> I put a T0 at the very end of my standard end code just to be safe

I did this as well, but make sure you turn your heaters of in the end gcode as well. If you let Cura handle it, and your last used extruder is T1 then you get

> T0 
> M104 S0 

cooling T0 down and leaving T1 hot.

This is the reason why it is good practice to always specify an extruder in heating codes.

lorenmcconnell, 2017-09-17:
@BagelOrb said this is fixed in Cura 2.5 b2.  I'm using Cura 2.7.0 and its still an issue.

Ghostkeeper, 2017-09-24:
Indeed, I can still reproduce this.

The front-end correctly finds that the extruder is unused. The back-end puts the heating codes there though.

Drayson80, 2017-09-25:
Oh... Do you expect it will be fixed there soon?

maukcc, 2017-09-25:
When addressing this issue, please regard #1518 as well.

ayildirim, 2017-11-28:
I am still experiencing this issue on Cura 3. Besides I am getting extra 2 lines of brim from second extruder in combination after the actual brim is completed by the first extruder.

Even more, when going onto the second layer, after that extra brim by second extruder, head goes to home position waits and comes back to a shifted position. As if the position of 2nd extruder was over compensated. Although I picked to use 1st extruder only all along, both for print, support and brim.

stuartpb, 2018-01-01:
There's another bug using extruders that aren't T0 where Cura assumes they are: Ultimaker/CuraEngine#676

closed 2018-11-26 by Appesteijn

milch, 2021-09-07:
In case anyone stumbles on this issue like me while googling how to prevent Cura from heating the second extruder if it's not needed, I wrote a post processing script to do this. I use Klipper with a `START_PRINT` macro which takes care of the heating, so my post processing script adjusts the START_PRINT macro accordingly. 

```python
from ..Script import Script
import inspect

class RemoveUnusedToolHeating(Script):
    """Removes the heating of tools if they are not currently being used
    """

    def getSettingDataString(self):
        return """{
            "name": "Remove Unused Tool Heating",
            "key": "RemoveUnusedToolHeating",
            "metadata": {},
            "version": 2,
            "settings":
            {
            }
        }"""

    def execute(self, data):
        startup = data[1].split("\n")
        t0_used = len(list(filter(lambda s: s.__contains__("T0"), data[1:-1]))) > 0
        t1_used = len(list(filter(lambda s: s.__contains__("T1"), data[1:-1]))) > 0
        start_print_idx = [i for i, value in enumerate(startup) if value.startswith("START_PRINT")][0]
        start_print_directive = startup[start_print_idx]

        start_print, bed_temp, left_extruder_temp, right_extruder_temp = start_print_directive.split(" ")

        new_directive = [start_print, bed_temp]
        if t0_used:
            new_directive.append(left_extruder_temp)
        if t1_used:
            new_directive.append(right_extruder_temp)
        updated_directive = " ".join(new_directive)

        startup[start_print_idx] = updated_directive
        data[1] = "\n".join(startup)

        messages = inspect.cleandoc(f"""
        ; Disabled unused heaters in this file
        ;
        ; Found startup script: {startup}
        ; T0 is used: {t0_used}
        ; T1 is used: {t1_used}
        ; Original start print code: {start_print_directive}
        ; Updated start print code: {updated_directive}
        """)
        data.insert(1, messages+"\n")

        return data
```

Cura always seems to insert T0 even if that extruder isn't used for printing, so effectively it always either results in `START_PRINT ... LEFT_EXTRUDER_TEMP=temp` or `START_PRINT ... LEFT_EXTRUDER_TEMP=temp RIGHT_EXTRUDER_TEMP=temp`. I haven't actively used python in years and I just threw this together in minimal time, so the code may look a bit wonky, but it works for my use-case.