Ultimaker/Cura #1129: "Custom start" gcode adding M commands before my custom start code

theflyingbean, 2016-11-10:
Cura 2.3.1
Windows 10
Custom printer (Lulzbot Mini)

I have opted to use custom start and end gcodes so that I can utilize the nozzle cleaning and auto bed tramming features of my printer. I have atached two photos showing the start code I have put in, vs what actually gets put on my sd card.

When starting a print, this is how the printer behaves:
1: Heat bed to 60c
2: Heat extruder to 200c
3: Home all
4: Sets extruder tem to 140c
5: Runs through nozzle cleaning and auto tramming just fine
6: Then sets desired bed and extruder temp to 0c and waits indefinitely (I'm not sure why this occurs)

Desired start code
[image]

Actual start code from sd card
[image]

Last few lines of start gcode if needed
[image]

fieldOfView, 2016-11-10:
Use {material_bed_temperarure} instead of {print_bed_temperature} and {material_print_temperature} instead of {print_temperature} and it should work.

Ghostkeeper, 2016-11-11:
It actually detects whether you have `{material_print_temperature}` in the start g-code, and if you do it won't put the extruder temperature at the front of the g-code. Similar with `{material_bed_temperature}`.

theflyingbean, 2016-11-11:
Thank you! :) It works as expected now

closed 2016-11-11 by theflyingbean

uwezi, 2017-06-04:
Is there a list of supported g-code macro replacements like "material_print_temperature" somewhere? I used to have a personal start/end-code in Cura 15.04 which does not work in 2.6 anymore since at least F{travel_speed} is not supported anymore...

Ghostkeeper, 2017-06-05:
It's all settings: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json

`travel_speed` was renamed to `speed_travel`.

Additionally, `time`, `date` and `day` are also allowed: https://github.com/Ultimaker/Cura/blob/master/plugins/CuraEngineBackend/StartSliceJob.py#L244