Ultimaker/Cura #1130: Printer variant not saved

RedTrain, 2016-11-10:
Cura 2.3.0, W10 pro, custom printer

I add a 2nd custom printer and adjust some settings.
When I close and reopen Cura, the printer I added has disappeared.

Log: cura1852.txt : adding printer and closing cura
[cura1852.txt](https://github.com/Ultimaker/Cura/files/584135/cura1852.txt)
Log: cura1853.txt : reopening cura, printer disappeared
[cura1853.txt](https://github.com/Ultimaker/Cura/files/584136/cura1853.txt)

Ghostkeeper, 2016-11-11:
Exception: When trying to deserialize Custom FDM printer #2, we received an unknown ID (Custom+FDM+printer+%232_variant.inst) for container

For some reason, it thinks the ID of this variant has the `.inst` extension...? Weird.

We fixed a few things with profiles and the way they are saved for version 2.3.1. Perhaps you could try [upgrading](https://software.ultimaker.com/)? It's a patch release without any big changes.

RedTrain, 2016-11-19:
Upgraded to 2.3.1 - a first look suggests that custom printer settings are now retained correctly.
Thank you.

closed 2016-11-19 by RedTrain